meerk40t/meerk40t #1: Kerf Correction

tatarize, 2019-07-21:
The svgpathtools class has the offset curve ability and I natively use the correct type of curves. It's not hard to nudge offset the curve by half a beam length in one direction or another.

c2v4, 2022-06-07:
I was searching for a solution for a week, and found nothing ;( Basically I need a bit more than an offset that Inkscape provides out of the box (Yes, that solves kerf dimensional accuracy both for inner and outer cuts). What is the problem with it, is that it does not apply correction to a corner—it's a simple tool. Basically, whenever you have kerf, you cannot cut a sharp corner which is needed to create interlocking parts—there will be always some material left if you correct for kerf. The solution for that is adding a dogbone that would remove additional material in the corners. In my opinion, this is easily doable by checking the angle between path segments and adding a cubic Bézier whenever the angle is greater than some value. Boxes.py are dealing with such problem in a clean way: https://florianfesti.github.io/boxes/html/api_burn.html

tatarize, 2022-06-11:
There's a few ways to do this math. Usually there's approximations and these usually involve a couple different steps. You solve for the normal and perpendicular vectors for each particular section and curve and then you solve for the parts that suddenly overlap each other and then use that to adjust the the shape. I hear that going from a quadratic bezier curve can and does sometimes yield a 10th dimensional bezier curve to do the *correct* offset curve so you're going to have to bite a bullet somewhere and do some approximations of the resulting curve that are close enough that it won't matter that much.

Svgpathtools, I think has some pretty solid code to do tangent vectors and thus could do normal vectors and offset a path. Some corner regions are going to look weird, and mathwise I think you can adjust some of these things like finding self intersecting corners and adjust them. It doesn't need to go too far with laser control software since we're talking like 0.1 mm or so move half a beam outward.

Sophist-UK, 2023-04-25:
Presumably only applies to cut operations and only to closed paths that are either the outside of a piece or an inside cut with nothing inside. (What happens if you are cutting e.g. several separate pieces nested inside one another with scrap material between?)

So MK will need to determine which are outside / inside closed cut paths and use odd/even to determine which way to offset the curve. We already have functionality to determine what burns are inside a cut path, so we would only need to use this to find other closed cut paths inside and then use odd/even to offset in the right direction.

closed 2023-10-19 by tatarize