meerk40t/meerk40t #5: Start from middle.

tatarize, 2019-08-23:
It's likely a typical thing that the execution of laser project will break at some point. If we still have the correct sequence we should be able to move the head to some point within the design and start from that point. So a sort of resume without knowing the current position feature. So rather than start at cut 0, start at cut 2449 moving the head to relevant position.

tatarize, 2019-09-07:
The system is entirely able to track the position and update the gui on the fly so it wouldn't be too hard to give a progress bar that would actually give feedback as to where the project stopped. If we know we're about 80% of the way through the job, we could just skip 80% of the project then move the laser head to that location. Iterate through the different positions near there to find the last position and restart the job.

Since the head position changes a bit, we could nudge it slightly to find the exact position. This also might be useful if the project goes off register for any other reason.

tatarize, 2019-09-28:
An easier way to manage this might be to place the allow the job to be micro-managed. If we allow, not just break paths into subpaths. But, also, break paths into individual elements. Move laser to start position (which might be useful for other reasons and move object to laser position.) Then somebody could take a job. Find the part they were executing, find the subpart within that that was being run. Then only spool everything below that point. Thereby only running the job from the point they dictated.

Sophist-UK, 2021-05-19:
To resume you would need to:

A. Know for certain that the second set of instructions were identical to the first;
B. Store a persistent state (on disk) of the ACTUAL instructions executed (not what has been sent to the laser, but what has actually executed)

I am not sure that either of these are possible.

tatarize, 2021-05-20:
You'd need to know the actual state of the laser. Which you should. If you went through 20% of the cutcode and you had the remaining 80% the project would delete the first 20% of cutcode and then go ahead and run the remaining 80% without issue. Assuming you don't optimize it different or do something weird that'll leave you with the cuts you didn't already do. You can home the laser etc and you'll be mostly fine. 

One assumes the end user know where it was when the project failed, and could go ahead and find that point and have it perform the remaining cuts. Since the state needed to perform the cut is known the state of the laser would be set to the one needed to do that cut.

Sophist-UK, 2021-05-20:
If optimisation is guaranteed repeatable, great. But if the burn order is different then restart is not possible.

If user has to guess where to restart, they will not be precise and either some bits won't be burned at all, or some bits will be doubly burned. May not be an issue for a Cut, where cutting right through is the same as cutting right through twice, but for other operations it will be.

So, you need to record the position persistently (across power outages or O/S crashes). And that means storing it to disc frequently. If it is a HDD, it will be too slow. If it is a SSD, then frequent writes will shorten life considerably.

BUT ... if you are prepared to accept a small amount of double burning (perhaps so small that the eye won't notice it), then you can reduce the frequency of persisting the burn state to the point that it might make sense.

Sophist-UK, 2021-05-20:
BUT ... if you only want to be able to click cancel, fix something and then restart from the point you cancelled, then you don't need to persist the position on disc, just hold it in the memory for the MeerK40t process.

tatarize, 2021-05-21:
You can save the cutcode file on start. If you have the cut order you're running. You don't need to save the exact position. The user would have a record of the position on their lasered material. You need only move to the relevant cut in particular. You'd either lose or double a little bit, but if you had to hard kill your laser and your material is either unmoved or has a deterministic position the job wouldn't be lost.

There are clearly going to be some cases this is essential for like the use of client material where you can't just get a different piece. The other way to do this in a more low-key way is to re-run the same project again. Turning the laser off, then turning the laser back on. This would just be a version of that where you skip a bunch of the early stuff and just go to about the right cut.

Sophist-UK, 2021-06-01:
Yes - you may be right that a slight repeat burn - image raster or vector might not be an issue compared to having to start again.

So let's suppose that:

1. The material has not moved.
2. The Cutcode is identical (because it was saved and we are reusing it).

Then what we need is a wizard to help the user. Here is how I imagine it might work:

a. you display the Cutcode in a simulation style window. 
b. user clicks where the previous burn got to as far as they can tell. Multi-passes may be an complication to deal with (but if 2nd and subsequent passes are only cuts, then less of a problem).
c. MK goes to approx point. Tells user to set low power on laser. Then user can step forward - and backwards if they have gone too far - (say) 1mm at a time, and user can ask for a v. short laser burst to show where it is. 
d. When user has got close, then you click restart and it gets going again.

tatarize, 2021-06-02:
Though the easiest would be find the position in the cutcode you want to start from. It's literally just taking simulator and applying a line of code on the cutcode to throwaway everything before the marked point. Assuming you have the same cutcode or the cutcode is deterministic (it should be), then delete everything before this position in the simulator would be an easy menubar item.

Sophist-UK, 2021-06-02:
You need fine controls to step forward and backwards, by vector path segment or raster line, and a control to do a blip on the laser to show where the head actually is. Then when you are at the right point, you click restart and it does what you just suggested - lose the cut code prior to that point (including the early part of the raster which is probably a bit more complicated than starting a vector from the beginning.

There might be complications around multiple passes and user determining where it got to. BUT also, whilst it is not possible to record exactly where it got to because disk storage is either not fast enough (HDD) or not robust enough to survive it (SSD), it **is** possible to persist a less frequent status like which pass or operation it has reached or perhaps persist a status every 10s so you only have 10s worth of cut code to step through.

Sophist-UK, 2023-04-25:
It should be possible to permanently record at a minimum the "passes" (I use the term loosely because depending on merges set and special operations they may not really be passes). Saving progress every (say) 15 seconds would not create too much I/O either.

Either of these would enable you to get close and then the user can tweak restart point by eye as discussed above.

closed 2023-10-19 by tatarize