meerk40t/meerk40t #18: Phase 2: Full installer.

tatarize, 2019-09-07:
When beyond phase 1. It needs a proper installer. Shouldn't require anything from the user.

Feedback and people are a must for this project so once beyond initial phase 1. Phase 2 should be getting a proper installer.

zoul0813, 2019-09-24:
I may be able to help out with a Drag/Drop Mac binary release.  Let me know.

tatarize, 2019-09-24:
The idea here was to make sure the libusb driver mostly installed. Was apparently a headache for Scorch and he switched over to making installers that integrated the libusb stuff in them. And it seemed like the amount of questions he got about that stuff dropped. I'm thinking maybe this needs to be expanded to providing solid binary installers for the major OSes.

tatarize, 2020-02-06:
This is actually fixed circa 0.4.0 since the new release has #124 finished.

closed 2020-02-06 by tatarize