meerk40t/meerk40t #24: Multi-shade raster

tatarize, 2019-09-08:
Raster the same image several times at different speeds for different cut times. There's no need to raster the same pixels repeatedly at the same speed. If we calculate the pixel time to burn amount based on the heat impact, we can actually get pretty significant detail overall in a couple passes.

tatarize, 2020-02-24:
Some really skilled laser folks with the M2 nano. Do different passes at different speeds in order to get different depths. While it's entirely reasonable to run a series of  images to try to match a grey scale. It could also be the case that other pixel information could be used to store this same data.

We could, for example, raster all the red pixels at one speed and all the blue pixels at a different speed. Cutting different slices out of the image. This could also be done based on shade, so we run a raster that divides up the image based on the image shade. So we run all the pixels with a darkness of 127 to 200 in value, at one speed and all the pixels of 0-127 at a different speed.

This could be done like the zraster where the image processing is done beforehand and cut into slices. Since those parts are pretty reasonable.

tatarize, 2020-08-24:
With RasterWizard and other modifications like Zrasters there's some clear need for this feature to be its own object. It would store the original image and several processed sub-images which would be rastered in order. It would need to provide the ability to flip through these different passes, and have each of them rendered in different ways with some different properties. But, they would all move and act as a group. And functionally allow access to each frame within the multi-pass object.

* This would also allow different backstroke objects, for unidirectional rastering where the backstroke is utilized to also provide information.
* This would allow z-raster objects to be a real and understandable thing. Rather than a bunch of images that you literally cannot properly manipulate. Since they must overlap and you can't see the ones below the other ones. You could also do a couple passes to pretty exactly match PWM rastering. Mostly because you could modulate the pixel level heat impact at each point.
* These would store the original image, so you could apply more RasterWizard to other objects without breaking the original.
* These would allow hybrid single images that are cropped/sliced into multiple images, so you could have an image-series that is several images but these would be processed in series rather than in parallel.

tatarize, 2022-01-07:
There's some need to allow slice and slash to define an image-cluster thing so they can be moved as one even while they are multiple different image elements. So 1 image set in 2 parts and their positions relative to each other is paramount. Though this might just be defined a bit as group.

Sophist-UK, 2022-01-07:
I think that there is some real merit here - but the scene UI would be ... um ... messy if we are not careful and I am unclear what the UI would look like to create this stuff (just console commands?). If we could do variable speed to handle grey scale in some automated fashion, that would be awesome, especially if the results are way way better - however I am sceptical that a picture (which has extremely variable greyscale) could be rendered with varying the speeds that quickly.

HOWEVER...

This does make me wonder whether there is scope for e.g. wood carving (in the same way that 3d printers do lithophanes) by automatically varying speed and number of burns based on grey scale.

Sophist-UK, 2022-02-20:
Assuming a perfect material which evaporates away at a rate entirely proportional to the laser energy it receives (so you can raster as slowly as you want without e.,g. it bursting into flames), and zero spread of heat to surrounding areas (so that you don't need to use multiple passes in order to allow surrounding material to cool down between passes), then is there really any difference between this proposed approach and a much, much slower single raster using greyscale PPI to vary the depth of material burned away?

Of course, wood, leather, acrylic etc. are **not** perfect materials. Leather and wood char from material either side of the cut overheating. and acrylic melts locally instead. But I do wonder how much of this can be done with one pass at slower speed and variable use of PPI to adjust the depth of burn.

tatarize, 2022-02-21:
The UI object would need to be different than other types. Like a group where everything is anchored in place. And It's not clear how we iterate through the different layers. But, you if we could specify that a layered object it might be useful. There's some big gaps in implementation after that idea though, because even if we modify the other routines in various ways I don't really see how the UI works. I guess we flip through them like a book or something?

closed 2024-10-04 by jpirnay