meerk40t/meerk40t #36: Design colors should be predefined settings.

tatarize, 2019-09-23:
Should be able to define particular colors and have them refer to predefined known settings. To quickly allow these to be used.

tatarize, 2019-09-24:
When a particular stroke color is loaded the matching stroke would be assigned that setting rather than the default setting. Needs a dialog to assign default settings, and default settings to particular colors. It would be kinda like Whisperer's color = settings but, with all strokes considered and fallback settings etc. Setting the colors for a particular path would just assign those settings.

matthewflinchbaugh, 2019-11-25:
Would also be a good idea to color code the paths in the nav tree to represent their assigned color without having to click each path to identify where it is in the drawing,

tatarize, 2019-11-25:
Certainly should be done. Perhaps even draw out an icon for it that looks a bit like the correct path. Shouldn't be too hard to render something like that. And if they are all the same color you'll still be clicking around looking for the correct path without seeing something that looks about right.

tatarize, 2019-11-25:
Yeah, that's certainly something that needs to be done. I'm more actively working on this and the color code vs. operations issues are a bit more complex than I had considered at first brush. This along with #89 have some different points. Cuts differ from vector engraves in that cuts are preprocessed and optimized to ensure they are cut inside to out. This needs to be done more automatically. But, I also need to account for a bunch of other operations like #88 and #87. Some of these operations should be triggered with a different color. But, some settings like speed would apply to all but certainly vector speed, cut speed and raster speed are pretty different and I've not yet sorted out how it should entirely work.

But, at a minimum you should be able to basically load up an svg from inkscape etc, and merely hit the go button and assuming the settings in Meerk are right, it should be able to perform any operations needed even if these operations are not currently known as such. 

Though, I've not really figured out how to define these colors or where. And when I switch one type to another type do all the settings switch over or does each color have default settings. I'm thinking it's the latter. Each color gets a default operation, and each operation uses it's own settings. Though if you modify one setting it might need to modify all those settings or just the setting for that one path/text/image/rectangle etc you tweaked.

This should allow for correct actions properly dividing cut and vector engrave where there's a different optimization added to the cut. Or permit other optimizations like minimizing distances between otherwise equal elements. Or if you're making a 3x5 grid of identical rasters you could raster across all the objects or raster each one individually, and there's currently no methods for that kind of fine grain control. And to be maximally hacky I sort of need to have that stuff sorted.

Haleybelinda, 2019-11-25:
Hi, When will it be packaged to be a successful full download. Thanks

tatarize, 2019-11-25:
I fell a bit behind schedule given the weirdness of that problem, so I was focusing more on getting inkscape to work for all contingencies. This lead to making sure my images from inkscape lined up correctly, which was going to require I permit arbitrary image matrices, and greatly expanding the svg project into a something more robust. As far as I can tell there's a number of things that need to be done before the next version.

1. Get the transformations for shapes to fully line up.
2. Get the arbitrary matrices for images to work properly.
3. Add in the ability to apply the matrix of the image into a modified image.
4. Color Define needs to be redesigned according to the more solved methodology.
5. Colored settings need to be applied correctly.
6. Full audit of all features since the switch to svg.elements is notably drastic.

There's a bunch more minor things that also matter but they could be pushed to a future version.  Like cuts properly optimizing inside out. And some of the methods like 3dzraster and cross-rastering.  But, the code is kinda apart right now and needs to be better put together. Which has lead to a pretty long delay already.

I'll try to focus on it and less on rarer issues like a quirky svg parsing issue for arcs that nobody would ever use.

tatarize, 2020-02-22:
The underlying problem behind this was resolved in 0.5.0 with the #132 change. Colors can be used to help classify elements into operations. But, operations are the first order citizens of the lasering landscape. Elements merely have colors and some display properties. These do not get specially classified by color with different properties. Rather they are two different objects and the entire thing was backwards which is why it was confusing. You engrave operations at a 1:many pointing to elements which merely exist as data.

closed 2020-02-22 by tatarize