meerk40t/meerk40t #41: Dxf support option

tatarize, 2019-09-25:
Support Dxf files.

https://en.wikipedia.org/wiki/AutoCAD_DXF

tatarize, 2019-09-30:
https://github.com/mozman/ezdxf is a solid project MIT licensed that does the DXF work. This should work well for our ends.

tatarize, 2019-09-30:
https://ezdxf.mozman.at

Haleybelinda, 2019-10-08:
Any chance of bitmap also, please? (GSW)

tatarize, 2019-10-10:
Bitmaps would be .bmp and are covered as a graphics format. You can already drag and drop them into meerk40t.

Haleybelinda, 2019-10-10:
That is brill. Gsw is a form of bitmap from a programme I use.  It was not displaying as a valid file type. I will try drag and drop. Forgive my ignorance, where do I put the laser reference to get my machine to talk to the meerk40t?

tatarize, 2019-10-11:
I dunno if it's compatible with standard .bmp. If it is a valid recognized image format it should be but GSW isn't any format I can find any references to so it actually seems doubtful.

Haleybelinda, 2019-10-11:
Hi,
Thank you for getting back to me, where do I put the device I'd. To get
your programme to communicate with my laser. I have an m2 board so it
should work, I would like to try it. The software GSW uses either for
photos.
Thanks
Haley

tatarize, 2019-10-11:
I'd save it to one of the more common graphics formats, like jpeg or png and just use it that way. The laser communicates through libusb. If whisperer works, then this would work. But, I don't really understand the problem or the question. There are instructions for getting it to work in the readme as well as some premade binaries for windows in releases.

Haleybelinda, 2019-10-11:
Hi tatarize,
I am still using laserdraw,
I seem to have the program open it just cant find the usb. The libusbo is there. I think I need to tell the program which USB its at and cannot tell which is which. Any way thank you for your help.
Haley

tatarize, 2019-10-11:
Yeah, you'll have to switch out your driver. Which means currently I don't have an install to do that. I'd daresay your easiest bet is to install K40Whisperer and install the driver. With the install. My setup uses the same driver methodology.

https://www.scorchworks.com/K40whisperer/k40whisperer.html so if Whisperer works so will meerk40t.

tatarize, 2020-03-06:
0.5.2 has most of this included. It still has some bugs but exists as a thing. @joerlane You said you found some attribute errors, can you send me some specific files to trigger those?

closed 2020-03-11 by tatarize