meerk40t/meerk40t #51: SVG does not size properly after import

dietervansteenwegen, 2019-09-27:
Attached file (zipped because .svg is not allowed) does not size properly after import. Works in k40 Whisperer V0.27.
Had this issue with a number of other files as well.
Exported from Inkscape, document 
[back.zip](https://github.com/meerk40t/meerk40t/files/3661677/back.zip)
Inkscape document settings:
[image]
Result on import (note sizing):
[image]

tatarize, 2019-09-27:
Yeah, the parser uses the native numbers but they don't actually correspond to anything. I'm doing an audit of the SVG now. Since this will likely be the bulk of how people import things and I should get it right.

tatarize, 2019-09-28:
Okay, implemented SVG section 7.2 to convert the scale info into a matrix and apply it. Testing now, but it seems pretty solid. Though I applied it to the matrix so when it decides the units are mm because it used mm, it scales it correctly but then says scaled 39... X. I'm going to have to reify the matrix on the path data. Gotta make the paths really say they are that big. Besides somebody requested a reset matrix option that would otherwise kill the svg applied matrix. But, so far it looks solid.

dietervansteenwegen, 2019-09-28:
Unfortunately the file in attachment still scales much too small (height = 0.14cm in Meerk40t).
K40Whisperer does size correctly.
[front complete.zip](https://github.com/meerk40t/meerk40t/files/3665232/front.complete.zip)

tatarize, 2019-09-28:
I fully implemented the section and added it to the latest version. I'll leave this open for a bit but it should be fixed in 0.1.6 as all my tests came back correctly and it's actually done quite robustly. I even fully implemented the preserveAspectRatio stuff.

closed 2019-10-02 by tatarize