meerk40t/meerk40t #53: Mouse over hints on some buttons would really help

dietervansteenwegen, 2019-09-27:
Quite a lot of buttons would benefit from a mouse over text.
Might help people find out features for themselves and lessen the amount of questions you get.

dietervansteenwegen, 2019-09-27:
Although, I must say I like your buttons icons...

tatarize, 2019-09-28:
They are Icon8 icons. They have some nice ones and fit with the same theme once you pick them.  I haven't actually deleted the icons directory, since I might still use some but for a number of technical reasons icons are best distributed in python with converting them to text. Hence the icons.py file. It properly bundles like the rest of the file but things like pyinstaller get much easier when you don't need to include actual files other than python.

But, it does remind me that, I should make an about dialog.

"DESKTOP APPS
macOS and Windows applications should have the link to https://icons8.com in the About dialog."

Oh, and probably more mouseover hits for buttons. And other elements. :)

tatarize, 2019-10-10:
- [ ] Alignment
- [ ] BufferView
- [ ] ColorDefine
- [ ] Controller
- [ ] ElementProperty
- [ ] JobInfo
- [ ] JobSpooler
- [ ] Keymap
- [ ] Navigation
- [ ] Preferences
- [ ] Usbconnect

tatarize, 2020-02-06:
The unhinted windows have been marked.

closed 2020-05-31 by tatarize