meerk40t/meerk40t #58: Laser trace does not update/redraw when zooming

dietervansteenwegen, 2019-09-27:
Not sure if this is worth looking into.
When a job is done, if you zoom in to the path that the laser followed you get something like this:
[image]
Or maybe that is really the path because the imported SVG gets scaled to about 1/20th of the real size.

tatarize, 2019-09-27:
Yeah, the trace lines there are anywhere you laser head was. It's actually a bit of a bad design. Basically it tracks the last 500 locations (maybe 50) of the laser head and just draws those lines between them. They stay there because those lines still exist. I could trash them altogether or change them to mean something else or figure out a better way of doing that altogether. Basically I have an updated location for the laser head, and wanted something to indicate where it was.

tatarize, 2020-02-23:
This problem has gotten worse as the trace lines are signaled and as such don't even connect anymore. Running a project just leaves some pointless crap in the scene.

tatarize, 2020-07-24:
This is somewhat fixed. It now deletes when you click on a different item to be selected. Also, 178 can be the main objection here.

closed 2020-07-24 by tatarize