meerk40t/meerk40t #61: SVG text should be accounted for.

tatarize, 2019-09-28:
Even without actually rendering the text, it would be nice and correct make a text object for svg text.

tatarize, 2019-09-28:
https://catherineh.github.io/programming/2018/02/01/text-to-svg-paths

I use a different earlier fork of the methods behind svgpathtools, given that this isn't too hard to do.  And will provide internal ability to do this glyph editing on the fly. So you could have text paths that are editable inside the program or flow a series of them etc.

tatarize, 2019-09-28:
https://github.com/rougier/freetype-py

Seems perfect for this task.

tatarize, 2019-09-28:
Added the Parsing and the Element. I did not add the actual conversion to paths or the execution of those paths. And it says much more obviously that this is NOT IMPLEMENTED in that you had text in your SVG and that text got eaten because I don't deal with it yet.

tatarize, 2020-01-17:
Briefly added an attempt to parse the text with freetype-py but gave it up when it became a bit of a time suck. Also, not sure how to access system fonts directly and it might be reasonable to use access from https://fonts.google.com/ as then it would basically have all the fonts you needed and could locate most of the families even gave it a font from a different system that wasn't supported in the system set to run the font.

tatarize, 2020-01-19:
Though for now, permit cheating. Allow text to path functionality, if the user provides a path the inkscape.

Better working functionality is great, but no sense letting the perfect prevent the good.

tatarize, 2020-06-23:
You can actually make text and directly raster the objects.

jpirnay, 2023-04-04:
Obsolete

closed 2023-04-04 by jpirnay