meerk40t/meerk40t #63: Reload design function

dietervansteenwegen, 2019-09-28:
In Whisperer, I often position my design to match unused positions on my source material.
There's the option to click "reload" to load changes in the design file.
It seems that the way to reload in Meerk40t is to load a "new" design.

tatarize, 2019-09-28:
As I'm not intended to do massive amounts of graphic editing and I do actually sort of love that method I'll certainly look into adding that. It'll require saving the file that first gave rise that group, but I should likely do that anyway. If for no other reason than to give the base group element a name other than "group". `Butterfly.svg`, `mother.png` and let you reload from there, and have a clue what is what when you have more than one. Which is kinda needed for the ordering etc.

I love it.

tatarize, 2019-10-02:
This is added in 0.1.8

closed 2019-10-07 by tatarize