meerk40t/meerk40t #68: Not enough values to unpack on loading SVG

dietervansteenwegen, 2019-09-28:
Seems like I manage to keep coming up with SVG import issues...

MeerK40t crash log. Version: 0.1.6Please report to: https://github.com/meerk40t/meerk40t/issues
Traceback (most recent call last):
  File "site-packages\wx\core.py", line 3240, in <lambda>
  File "LaserSceneView.py", line 137, in update_buffer
  File "LaserSceneView.py", line 491, in on_draw_scene
  File "LaserProject.py", line 254, in draw
  File "LaserProject.py", line 439, in generate
  File "path.py", line 2085, in bbox
ValueError: not enough values to unpack (expected 4, got 0)

Do you want the SVG?

tatarize, 2019-09-29:
Path.py line 2085:
bbs = [seg.bbox() for seg in self._segments]

It's expecting the bbox() to return 4 elements, and something returned zero. The only thing that could do this is Close which could maybe not know what it's closing. If it's repeatable the svg might help, though here I think Close should rather overtly return None and then properly deal with it where it matters for that edge case. Since a closing segment might not actually have that information depending on how it was created.

tatarize, 2019-10-07:
Fixed in 0.1.9

closed 2019-10-07 by tatarize