meerk40t/meerk40t #70: Clipping Paths With Images.

tatarize, 2019-09-29:
While I thought it was likely pointless for my uses, there is a clipping path built into SVG and masking off a shape with an image is certainly doable and somebody in the Facebook K40 group wants to do text with a pattern inside it, and it would be much easier to do that in Illustrator or something and just import it here. While I figured doing that and saving it as a raster would be fine, There are some patterns that would simply be remarkably hard to duplicate without the native functionality. Such as in the topic of adding a pattern inside the lettering to help the paint stick to the inside of those letters.

tatarize, 2020-09-02:
There are also things that have cropped up. Like attempting to using a clip path to bound an image and MeerK40t loaded the image up without any clip-path since it ignored it. Further there are some utilities where a raw clip path would be highly useful. Like if you wanted to select a section of an image and raster just that area a second time to make it pop cut out some paint or several other operations. A clip path that takes up the view of the objects below it would absolutely allow such things correctly.

Much like some features for typesetting images into shapes there's clearly a need to sort of permit this. And the code would be mostly render based. Basically find the bounds of this image and any pixel located inside this closed shape is moved to a secondary raster and shown.

tatarize, 2020-09-02:
https://developer.mozilla.org/en-US/docs/Web/SVG/Element/clipPath

Sophist-UK, 2023-04-25:
Just to point out that you can apply a clip-path to other elements and even groups in SVG not just Images.

If you apply it to a group it is dynamic as you move the elements that are inside the group.

Sophist-UK, 2023-04-25:
#1295 and later fixes implemented a (permanent) image-clipping-by-path functionality.

jpirnay, 2024-10-04:
We have a keyhole functionality since 0.9.5.b8, we will not do more

closed 2024-10-04 by jpirnay

Sophist-UK, 2024-10-04:
If I have a path clipped image or vectors in InkScape, does that load into MK as a keyhole?

Sophist-UK, 2024-10-04:
P.S. Wanna bet that at some point you will do more (i.e. a minor enhancement to lock keyholes to a linked group or element?