meerk40t/meerk40t #71: Error opening Raster & Vector svg

Cyber-phobia, 2019-09-30:
Opening Raster only images - works (albeit with scaling issues)
Opening Vector only images - works (albeit with scaling issues)

Opening Raster & Vector SVG files - No response. Nothing is opened, no new group is added.

tatarize, 2019-09-30:
Hm. I'll double check that. I think the vector data inside the SVG should actually work. The Raster data within the SVG however isn't working yet. It needs to process local files and xlink data images. And currently doesn't. I had not tried combining them but the assumption is it should load the vector data and ignore the raster data.

tatarize, 2019-09-30:
Also, what do you see as the scaling issues with rasters alone and vectors alone?

tatarize, 2019-12-27:
This likely is fixed circa 0.2.0 since I basically fixed all svg loading issues with the new library.

joerlane, 2019-12-31:
I tested this briefly on OS X last night and it appears to work. I had tried the Vector => Convert to Raster for some paths, but was annoyed that they ran raster one-by-one on the laser cutter (despite being on the same horizontal row). Seems like it would require some kind of grouping by speed (automatically?) to resolve that problem in MeerK40t. 

As such; I decided to rasterize the row of multiple vector graphics in Inkscape and saved the combined vector/raster SVG. The embedded rasters were converted to 1000ppi PNGs (inside Inkscape) and re-imported to the file. 

I got a strange message about actualizing the graphics files in MeerK40t, but it all worked fine when I sent it to the laser hardware. I have a few more of those type of files to run tonight; so I'll update if I find any further issues.

tatarize, 2020-01-02:
The strange message will go away in the next version as sending it from jobs to spooler will require you run the needed operations. Basically the code requires actual pixels that actually exist, whereas sometimes you have graphics like that what you have are stretched out, scaled, or rotated pixels. That message is telling you that those images needed to be turned into actual images and it changed things in the scene. It should look identical but the pixels are actual rather than merely transformed. 

The speed of rasters is often increased by doing each raster individually. Keep in mind that all the blank space between two images is basically always lost. If these two bits are really far apart you'll basically tank your speed when taking them one at a time would transverse that distance 1 time but combined they would transverse that distance as many times as there are raster-lines.

Actually solving for the fastest path is certainly doable. You'd figure all the lines you need rastered and where they are. Then you'd put them into a little graph internally and apply 2-opt optimization on the Eulerian path. To solve for a pareto-optimal solution to the travelling salesman problem. I'm actually planning on doing that at some point since I'm also going to, at some point, adapt some of this code to Machine Embroidery. There, you want the thread to require no cuts and to fill up vectorized shapes. Here it's kinda of the same thing except you're not going to take a hit for turning the laser off. Sometimes the solution is a boustrophedon serpentine path, but other times it could be deeply odd. For example a lot of text might be best to render each character and starting in odd positions so you end in closer positions to the next glyph.  

However if you want them all as one thing you need to convert them to raster as a single operation. Which is to say apply the `Make Raster Image` operation to the group containing all the elements you want to be part of that raster image. You can just drag them into the same group if you loaded multiple copies of relevant vector.

tatarize, 2020-01-02:
https://www.youtube.com/watch?v=v0831M0H5Ro&feature=youtu.be Gives basic difference on how you convert these items to raster and how it makes multiple or a single image. So you can convert them to raster as a group and they will go all the way across as you wanted.

joerlane, 2020-01-02:
Thanks for the link, I just tried dragging them into the same group... Didn't realize that was an option in this case. I can't seem to re-order paths in the SVG that way (which is probably why I didn't think to try), but can move converted raster images into a single group. What confuses me from a technical perspective is; what is the point of that image group if they run as separate raster events? Is there a future reasoning for that?

Anyway; that's not an issue after watching the video. I never thought to enclose multiple SVGs in the same tree, or rasterize the entire file (not individual paths). Both of those would produce the single image for a single image and single raster event that I wanted. The only reason I cared is because the objects I was rasterizing are a few mm each. The laser head flies a few inches left to right wasting quite a bit of time on those tiny things. Looks good; I can certainly work with this.

tatarize, 2020-01-05:
Currently, the way that UI works is kinda wrong. When you right click and say make raster, it makes an image file from the relevant path data. And adds that to the group as such. The problem with this is a couple things, mostly since you made the image from the paths and added it, you'd also need to delete all the paths if you *don't* want those being caught in the execution. And the image you make perfectly covers up the paths you're using to make said image, so it's a bit confusing that way too.

What I'm currently thinking, and feel free to point out any problems with it, is that the operation you do on the rasters or path data should be able to be switched without needed to perform weird selected operations in the paths. You should be able to select the paths you'd want to raster and just tell MeerK40t that that is a raster, and then in the Jobs Dialog it should do the conversion there. So before jobs are sent to the spooler they should be processed into the correct types of operations.

The problem I'm trying to address is there's a lot of different operations which can be done on the data. Like crosshatching over the same fill area or doing z-depth rastering, or rastering a filled vector path. And I sort of need a space to include all of these. The way it currently works is not actually helpful here. The spooler will clearly need a rastered path, but it doesn't need to create it in the scene when it can create it in the jobs dialog. Now it will likely need some methods to visualize this since it won't be altering the data in the scene, but rather copying and doing some operations on them, and sending that to the spooler for the writer to deal with. 

The mere ability to convert paths into something that is kind of raster doesn't address the need to perform a lot of different operations on the same data. Such as running a raster on some data and then a vector engrave over that. Doing that currently would require selecting the paths, making a raster, and running them in the correct order. This isn't bad, but after adding more things you can do, this will become kind of excessively limiting. 

It's not really worked out currently, the reason it works the way it works, isn't a well thought out masterpiece, but rather something that should work. It makes doing most things possible even if these things are done easily.  

You can move individual paths into the group currently holding that path and it will reorder them in a manner of speaking. It'll make that path last, which you could use to make them in whatever order you want.

joerlane, 2020-01-06:
I'm not sure this is fixed yet; though I haven't gone back and tried every test case individually to isolate the problem further. I still have to experiment with various file formats to see if it's one or more of those achieve different results. Ended up switching back to whisperer again for a few days to get some work done.  

I had done some early testing with the laser tube turned off; so it looked like combined SVG files worked as the head was scanning. Then I ran something real and had a fire. At the time I thought it was me; maybe I left it turned up too high from a previous job... I only later figured out the raster is at least firing very hot; the output always looks like a filled rectangle of the shape of the PNG raster placed in the SVG. 

Is is possible I have a setting that's trying to laser cut an alpha channel, or at extreme brightness or contrast of the embedded raster image? These are PNGs in this case generated through Inkscape; but MeerK40t does display them correctly on screen.

Paths converted to a raster inside MeerK40t do actually modulate the laser as intended.

Sorry for the bad image; but the upper rectangle I thought should have simply etched tiny shapes  actually burned entirely. I then rode down the laser power to the point it surface etched, and don't see anything indicating image detail; just a plain filled rectangle the shape of the entire PNG image component. 

[image]

I fully agree with your assessment that the dialog should allow a raster operation on SVG path data. I don't know the core source code details of whisperer; but that's how I send my files to it. I've been trying to convert them to raster these other ways as a method of utilizing MeerK40t; but it appears some of that is still problematic. 

My main use case is running a raster operation on filled vector paths in these latest projects. But I've also played with Z-depth rastering via multiple files, or multiple passes.

I think having multiple selection ability in the tree would clear up a lot of the arduous parts of working with complex files. It might pay to add some kind of descriptor to the tree items though. I can see the stroke and fill colors in the dialog when I double click; but a tiny shape glyph would be even better next to the Path in the tree. In some cases it might be nice to have control over tree view width, which doesn't seem to be possible (though I might be able to figure that out; it does adjust when I resize the window). 

I just re-read that bit about changing order. I see now that I can drop the path on the top level and it will drop to the end of that list. That's certainly helpful. I did it with every part of my SVG file and ended up losing the group, and getting the path back that I had earlier converted to an image (but keeping the image).

joerlane, 2020-01-06:
I started running the same SVG with the embedded image in different formats. 

The same exact PNG with no alpha channel included; does modulate the laser correctly. 

A TIFF format image with alpha channel rasters a completely filled rectangle. Would you guess? The TIFF with no alpha channel saved does work correctly.

JPEG of the same image data also modulates the laser.

tatarize, 2020-01-06:
Best guess is the step was still set to 1. Whisperer, defaults the step which I think Scorch calls scanline gap or something, is set to .002 (step 2) by default and can be set higher as needed. MeerK40t's step defaults to 1. The problem here is a couple things, first, `step 1` is maximum density so that's a lot of laser firing in a small area. When you increase the step it will scale the size up in two dimensions and the gap between the raster lines in one dimension. So if you go from 1 to 2. It's 4 times bigger and fires the laser twice as much so it's half the previous power, over the same area. You can easily at full size burn the wood, especially at lower speeds. You can raise the speed. If twice as fast it should half the amount of laser fired over the same area. So it should maintain darkness if you use step 1 rather than step 2 but run it at twice the speed.

The second reason for this is that I set all these properties at on a per-element basis. Because of this you'd need to set the step each time. The intent here is that there should be default properties which affect any unknown raster or path. And then other derived properties based on the stroke color. So red lines can be default flagged as a `cut` operation and unless stated otherwise all images can have a raster step of 2, with an overriding default value being able to be set differently elsewhere. There's other things like `overscan` which also have a purely per-item setting. Though with that you can only actually invoke it, by setting it directly in svg.

```
    <image
       xlink:href="silimage.png"
       width="53.181252"
       height="57.414585"
       preserveAspectRatio="none"
       id="image5077"
       x="5.0081787"
       y="122.3939"

	  overscan="200"
       raster_step="4"
       power="500"/>

```

I didn't make a point of saying it anywhere but if you set those kinds of properties in the svg file directly they should totally work. Overscan is the extra amount at the edge beyond the rastering the lasering uses. I haven't made any sorts of dialogs for it.

And finally, if you run it at good speeds, and with the knob turned down and dial that in, you can do better resolution with step 1 rasters. At least that's my impression. Obviously you can also burn things but, that's likely the result of a hot enough laser to start a not moving enough etc. You can do 1000 dpi rasterings and defaulting to 500 dpi (step 2) seems like it might be in error.
 
---

The embedded stuff should work fairly well. It's part of the svgelements library and I did a pretty deep dive making sure everything worked out correctly. And the actual image manipulation stuff is done with Pillow, which actually does a pretty spectacular job. ( https://pillow.readthedocs.io/en/5.1.x/handbook/image-file-formats.html ). There might be some quirks with it but for the most part it should deal with the images quite effectively.

joerlane, 2020-01-06:
That's some excellent information about step; and especially embedding things in the SVG. 

I think there is still an issue with alpha channels for some reason. I have to save my PNG specially without to make it work. The defaults when converting in Inkscape didn't work; and I'm guessing other programs as well. Without an alpha channel included, the same image will burn the raster graphics as I would expect.

tatarize, 2020-01-06:
Hm. Alpha is at times weird and could certainly be an issue. I'll raise one. Might be a fluke where it goes 100% on or something. That channel is going to be loaded and might certainly be different than the other formats which will alway have alpha at max.

closed 2020-01-19 by tatarize