meerk40t/meerk40t #73: Update checker

tatarize, 2019-10-01:
Would require server checks etc, but it was suggested. Might as well leave the issue here and see if anybody comes out strongly in support or anything else.

tatarize, 2020-02-06:
Could also be used to help deal with modules. Where is there is a dxf reader but it has a requirement that you don't need that module. However the pre-packaged stand alone builds would kinda need them all to work.

tatarize, 2021-09-29:
https://api.github.com/repos/meerk40t/meerk40t/releases/latest

Would be fairly doable. Just grab and parse the json there.

Sophist-UK, 2021-09-29:
That is the plan - or part of a slightly more sophisticated plan anyway.

Sophist-UK, 2023-04-25:
Update checking now has a UI. I had in mind a download and replace capability as well, but there are several complications with current executable location being writeable and on windows the current executable being locked.

ALso some users like me put the version number in the filename in order to keep multiple versions available, and (of course) you probably want to provide backup/backout capability in case new version is broken.

closed 2023-08-26 by tatarize