meerk40t/meerk40t #84: Constant crash on Raspberry pi

Leoavilacr, 2019-10-30:
Crash log
wxglade_tmp_menu.Append(wx.ID_ANY, "Transform", wxglade_tmp_menu_sub, "")
MeerK40t.py: Fatal IO error 11 (Resource temporarily unavailable) on X server :10.0.
Segmentation fault

/meerk40t-0.1.10/MeerK40t.py:185: DeprecationWarning: Menu.Append() is deprecated
  wxglade_tmp_menu.Append(wx.ID_ANY, "Transform", wxglade_tmp_menu_sub, "")
munmap_chunk(): invalid pointer
Aborted

tatarize, 2019-10-31:
I'm pretty sure this is on the level and consistent. I tried with my Pi at one point and it didn't work. But, I figured I'd get around to it later. Well, it's later now. I'm spending a while to spin up an emulator on my development machine. Though the crashes in question are pretty hard crashes within wxpython itself rather than my code in particular.

I'm not particularly optimistic about this across the board. But, it's worth it to check.

tatarize, 2019-11-01:
https://github.com/meerk40t/meerk40t/wiki/Installing-MeerK40t-on-a-Raspberry-Pi.

This set of commands at least launches it with the latest stuff and gets an interface. I'll have to launch it on my RP to fully check things.  It at least launches with those instructions. Still seems to spit out some errors but it might be with regard to my emulation.

tatarize, 2019-11-01:
> (MeerK40t.py:1709): Gdk-ERROR **: 19:52:38.678: The program 'MeerK40t.py' received an X Window System error.
> This probably reflects a bug in the program.
> The error was 'BadIDChoice (invalid resource ID chosen for this connection)'.
>   (Details: serial 9512 error_code 14 request_code 53 (core protocol) minor_code 0)
>   (Note to programmers: normally, X errors are reported asynchronously;
>    that is, you will receive the error a while after causing it.
>    To debug your program, run it with the GDK_SYNCHRONIZE environment
>    variable to change this behavior. You can then get a meaningful
>    backtrace from your debugger if you break on the gdk_x_error() function.)
> Trace/breakpoint trap

Got it to launch but it broke pretty fast from an error.

tatarize, 2019-11-01:
Also, background is entirely white and can't be seen as different from the field. Panning with middle mouse button does not work. 

Also:

> xcb] Unknown request in queue while dequeuing
> [xcb] Most likely this is a multi-threaded client and XInitThreads has not been called
> [xcb] Aborting, sorry about that.
> python3: ../../src/xcb_io.c:165: dequeue_pending_request: Assertion `!xcb_xlib_unknown_req_in_deq' failed.
> Aborted
> 
>

tatarize, 2019-11-01:
Scaling matrix for images in Pi doesn't work. It draws at the same location regardless of the locations etc. It look as if the matrix doesn't work at all.

The location tracking for the head and the last position in the blue are not synced.

[errors.zip](https://github.com/meerk40t/meerk40t/files/3796711/errors.zip)

tatarize, 2019-11-01:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/wx/core.py", line 3259, in <lambda>
    lambda event: event.callable(*event.args, **event.kw) )
  File "/home/pi/meerk40t-master/JobSpooler.py", line 249, in post_update_on_gui_thread
    self.gauge_controller.SetValue(self.buffer_size)
wx._core.wxAssertionError: C++ assertion "pos <= m_rangeMax" failed at ../src/gtk/gauge.cpp(95) in SetValue(): invalid value in wxGauge::SetValue()

tatarize, 2019-11-01:
#65

tatarize, 2019-11-01:
Apparently after Buster we can get do ``sudo apt-get install python3-wxgtk4.0`` and get it to work from there. However there are apparently a few crashes and flaws in the Linux implementation that cause it to not work. In theory if these are corrected for should work. I'll make a pendrive linux and see if I can get it fully working in that. Then, in theory, the bugs in Pi should be much reduced.

tatarize, 2019-12-30:
This needs to be rechecked with the Mac Updates Especially if there's some way to run wx.Python 4.1 on the Pi.

joerlane, 2019-12-31:
I pulled down a copy of the Dec 21st wxPython 4.1 source and began the build last night. It halted a few times due to the Raspberry Pi apparently overheating (mine is in a case with an LCD); the build continued after being restarted a few times. I finally got a small fan to cool the Pi and ran into a repeatable hard crash in the early part of the compile around [16/905]. At that point I increased the default swap from 100 meg to 1024 and was able to continue the compile. I might have been able to take some memory from the GPU instead; but didn't attempt that yet. My GPU memory is currently set to 64 meg so it's questionable if that could have been enough in this case. 

This is a Raspberry Pi 3 B+ running Raspbian 9 (stretch), though it's probably not completely up to date. It's been sitting in a drawer for a while. 

Compile is running again, and will likely be a few more hours at the very least (could be a full day)... Pulled MeerK40t last night and the only error I saw so far was the lack of wx so far. Just figured I would mention I began this so as not to duplicate too many efforts. 

[image]

joerlane, 2020-01-01:
Today; or maybe tomorrow...

[image]

tatarize, 2020-01-01:
I was pretty sure the process was recompiling all of wxPython so you won't get any competition from me. My [instructions for install](https://github.com/meerk40t/meerk40t/wiki/Installing-MeerK40t-on-a-Raspberry-Pi) will get 4.0.7 which will not draw anything. If Raspberry Pi actually works, and trying is already be way above and beyond, could you use pyinstaller and make an executable?

Because unless I make around highly useful fallback method for viewing the scene in wx4.0, or 4.1 gets a proper full release, your compile for basically ever method might be the only way and an install working. And the `--onefile` installers wrap up wxPython and python in them. So it would bypass that (1-2 day step) and run. It might require Stretch to run, but Pi would have an option that works.

Tho, I'm fine with just having to code around it at a later date.

joerlane, 2020-01-01:
The wxPython 4.1 compile took at least 8 hours; though I stopped paying attention after a while. Full build might be closer to 12 hours on the 3 B+.

So far it looks good. The SVG paths are similarly thin on the display pane (and hard to see); but I can probably change my stroke width to fix that. I can jog the red circle with the keyboard, middle click and drag to pan, scroll wheel zoom, select paths, etc. It was always my plan to drive the laser with a Raspberry Pi so I'll work towards that goal and see how it goes. 

[image]

tatarize, 2020-01-02:
That entirely sounds like success. The arrow keys jog the laser head around. And seems like it seemed on regular linux. There might be some issues with image processing. Like sometimes the images get really large. It creates the micro-gl data on the fly so it should be reasonable in that department. And I don't really see any issues beyond that. There might be an issue where queued events basically don't process and keeps adding more and more attempts to the queue, and it might be more pressing on the RPi, but likely needs to be doublechecked/fixed anyway.  12 hours for a Pi3 isn't too bad. I could likely do the same on my Pi4. Since really it should only need to be done once for wx and just pull updated versions as needed beyond that.

Great to know it works on the relevant oses. The whole Phase 2 goal of packaging it for those OSes seems kind of reasonably easy compared to the Phase 1 goal of fixing most people's workflows. Especially because those packages would build in the right version of WxPython. 

Thank you again.

joerlane, 2020-01-02:
I haven't found any issues with the image and path manipulation on the Pi software build. Unfortunately I plugged it into my laser and the problems are so far inconsistent. There are a few potential issues here; but first and foremost, I need to find an adequate power supply. I'm also using the supplied 6+ foot long USB cable; so that is also a possible compounding issue.

I was able to reboot the Pi and load a drawing with a few SVG paths for the laser, a few of which I convert into rasters in MeerK40t. It successfully cuts any paths that are first, and then the head freezes. It doesn't seem to attempt the included raster at all; but the spooler will finish sooner or later. Further attempts to cut paths or rasters looks like it's working in software (they queue up and clear the spooler); but doesn't actually home or try to drive the laser head again until I reboot the Pi.  I can't seem to get the controller to reset at this point successfully in software. I'm not toggling the power on the laser cutter though; which might work to reset it.

I've done this a few times; and once I saw this included crash, though I don't know what to make of it. It may be unrelated. Has this been tested in any other Linux variants (SVG with one path, one raster)? I can say that it does vector cuts and raster images in OS X without failing in the middle. Again; this could be simply a power or USB cable issue so I won't stress it until I can prove otherwise.
[image]

tatarize, 2020-01-02:
It does look like the connection is screwing up. You can check in the USB bit for the log, it should provide a good amount of information of what is going on. The idea is to make things much easier to troubleshoot where the problem is. There's also some chance the refresh queue is getting bogged down. Which may be a quirk with how I changed the messaging to the UI thread. I might need to try it myself to figure out where the problem is. 

A lot of people think USB cables cause a bunch of problems but they mostly don't. You can basically view that stuff in realtime from the controller and USB and when it's stuttering it's actually bottoming out the cards queue before more data gets sent generally because the commands don't take enough time to complete. A bunch of commands of UTaDTa over and over again, doesn't take long enough to complete since move over a mill and fire the laser takes basically no time, especially at higher speeds. 

Testing this from my linux thumbdrive seems like an easy thing. If the rastering problems go as squirrely as you say it might start messing up there too. I think it seemed a bit strange in Windows at some point while I was doing the test raster and sucked down more memory than it should have and the time between view updates got prohibitively longer. It might need a timed refresh scheme to shoot for an FPS rather than demand a new frame on each change and hope it get to a point where that can be done.

In theory if it got really bogged down it might just lose the connection and have no usb to write to.

joerlane, 2020-01-02:
I'm pretty sure that's all true. Just trying to provide the best detail for future inquiry. I didn't mention; the WiFi knocks off when it hits a raster in the spooler queue. Pretty sure this is a power issue to begin with; need a solid 5V for the Pi board and peripherals.

This is easy to ignore when you're trying to make things work on a Pi. I like to experiment on the edge anyway, so I'm not shocked to see the lighting bolt on the screen. Some older Pi boards can't even connect to WiFi with this level of power (from a USB hub). 

I have some 3A power supplies around here somewhere. Pretty sure powering the board correctly is going to help. At the very least.

tatarize, 2020-01-03:
Could be just taxing the memory and cpu. But, that shouldn't be that taxing. So in theory, I should be able to find what's doing that and give it a much gentler footprint. The biggest thing currently is refreshing the screen while it's running. It'll drop frames here and there as it starts getting taxed try it with refreshing simply off. 

It was the last commit I made on the manipulate branch.
https://github.com/meerk40t/meerk40t/tree/tatarize-manipulate

It'll have some nice widget size control stuff for the scene that does not yet work. But, also `View` will have a `Do Not Refresh` option.  Try turning that off after it just starts or before it starts the raster. Should basically turn the GUI updating off for the main scene.

joerlane, 2020-01-06:
I checked out the branch and even with the refresh turned off I got the same group of problems. Then I tried a (production) test file to be sure; and that one actually works. The embedded graphics are rastering just fine. So an SVG with both paths and graphics embedded that have no alpha channel works. 

So I took that new work file; and tried each of the specified functions as separate options, and that all does work. Converts paths to rasters, runs vectors cuts, etc. My previous Pi test file was three rectangles; and converting one or more of those to rasters froze it every time. I'm not sure how it could be corrupted; but I can't reproduce it any other way so far. It's possible each of the paths is missing a segment or something... I've seen that in Inkscape before and I used the Pi to create it.

I'm not sure how to get it to auto-home at the end of a job (option might be cut off because of my small display resolution?); but otherwise it appears to be running fine within the constraints I'm aware of. 

The only addition I can see adding is some kind of load status. There are times I think it is broken because something is going on. For instance I clicked start job, and the button doesn't change to signal it's acknowledged. It appears frozen for close to 30 seconds (while loading the spooler dialog/window). The CPU runs up around 30%, and it feels frozen; but it's not. Clicking the button repeatedly doesn't make any difference.  Kind of typical Pi behavior, but might stress people out if it were grafted into a K40 as the controller. Just food for thought; I think simply changing the color the "Start Job" button on click, and maybe the text would be good enough in a pinch; sans some kind of hourglass, or progress bar. 

At this point it looks like I can put a MeerK40t/Pi controller into production. I've been meaning to make an exclosure for a larger screen; but this 3.5" touch works well enough to load files, and fire the laser (with a little pixel doubling trickery).

[image]

[image]

joerlane, 2020-01-06:
Figured I would post a satisfaction update here. I'm not sure how to reproduce those waiting bugs, but they do happen from time to time. Otherwise the interface is snappy on repeated cuts; even when changing out files. Maybe it's simply temperature.

I ran the Pi as the controller until it apparently overheated. I would consider that promising and it shouldn't be difficult to keep it cool. I also didn't bother turning off the screen drawing; so it might be possible to use full time with passive cooling and some care. 

I was able to reproduce a similar controller hang that I was getting earlier with the Pi and rasters; this time in OS X in the middle of a job. Unfortunately; since I can't save the project file there is no way to absolutely reproduce it. I started watching the controller window, and haven't had another hang in about a dozen variable multi-pass raster jobs. I've seen whisperer burp and pick back up a few moments later. This seems a lot like that; but it's not picking back up again, and I can't do anything but power cycle and start fresh.

I'm not sure how that works in whisperer. I get the impression that the controller doesn't talk back. I did a quick browse of whisperer code and it appears there is a timer running in the EVG send code to make sure it's still slurping data?

tatarize, 2020-01-06:
To reproduce the wait thing, you'd need to have an image which is zoomed away from 1:1 pixels so that the program must actualize the pixels. Try zoom by a little bit, then it should stop there. Unless you're speccing the next version which will have that wait forced to happen when you hit the Operations button in the dialog. It's not strictly speaking a bug but is clearly something that can, at times, take time and some nice confirmation that it's doing something would certainly be nice.

I can likely run it in a different thread and give some sort of waiting dialog.

--

In theory if it dies and doesn't come back, it should. And there's likely something in USB saying why it died. However, it's also the case that it should be possible to resume a job 40% through or whereever you already are since all the commands are knowable, even a power-outage should be recoverable (if the gui and code are there). Since all you want is to run this sort of job that you just ran but most of the way through it, and start at some given position or such.

The whisperer code has the controller kind of talk back. It doesn't have multi threads so, sometimes every once in a while while running, it calls a function which updates the UI from the controllerish code. This lets it seem like it's updating. But, it ties the controller very heavily to the GUI where basically the processing of packets will stop briefly to tell the UI how to update the status bar and give messages, which is wildly coupled. In theory I should be able to take my K40Controller code and make it run from the command line and it won't care because the update signals will get ignored since they aren't necessarily tied to anything. In fact a lot of stuff doesn't do anything if you don't have a dialog open. Like that controller data doesn't go anywhere unless the controller dialog is up. It means I can have dialogs that can read whatever sorts of updated data it needs with some pretty basic hooks that won't  do anything unless the stuff is there to be updating. 

There's actually a lot of things below the hood that makes MeerK40t much more expandable. You can basically make your own dialog to do basically anything and it won't affect how any of the other stuff works. So I can do things like slap together things like the Alignment dialog or any other potential tools people might come up with and it should work pretty damned nicely.

BTW. I likely will be incommunicado until the Jan 10th. But I'll be reply to any issues you have at that time.

joerlane, 2020-01-06:
No worry about being incommunicado. At this point it appears the core functions are all working for me. I can focus on getting a Pi running on a 7" touch screen with active cooling. I also have some Pi zeros I can load up to test an even cheaper, more compact option. 

I'll do my best to establish testing methods for any minor issues I run into. The tip about zooming the window to re-actualize the images is probably correct in terms of the Spooler loading delay. I simply hadn't run enough different jobs to see a pattern emerge. I got about a dozen files through the Pi though before it overheated. They all included pretty heavy manipulation on the tree. Deleting paths (I use them as guides in development, but not cutting), converting paths to rasters, re-ordering, setting speeds and number of passes. 

I'll look more closely the next time I get a controller hang. I thought I looked last time and didn't see anything following USB Connection Successful. But the navigation panel wouldn't work, and jobs would clear the spooler without ever moving the head. Best I can tell the board inside the K40 is very confused at that point. I didn't wait a full hour to see if it would recover; but I gave it plenty of time more than once. Running with the controller dialog open so it was updating elements more often may have made it work better (not worse). Hard to say without something known to repeat the problem. I'll keep you posted; thanks again.

joerlane, 2020-01-08:
I got a start on a Pi Zero W setup. Since it doesn't have a DSI connector; I'm going to do yet another install on an original Pi with the 7" touch screen. The main differences with the Zero build are; lower power consumption, and less RAM. I'm guessing (hoping?) this board isn't as likely to overheat with full time operation. I also have another 3.5" LCD touch screen I can try the Zero with since that style runs off the GPIO header (which also leaves the HDMI open).

Decided to start the Zero build fresh; on a barebones install of OSMC, for a bit of an extra challenge. Got the build-essential installed and a few other odds and ends; and now the wxPython source build is running. I'll start the same wxPython build on my original Pi later, since those will probably take a bit longer than it did on the Pi3. 

Don't expect any real issues with MeerK40t, but figured it might help someone else down the road if there are idiosyncrasies with less RAM or slower CPUs. Cheers.

joerlane, 2020-01-09:
Needed to install python3-setuptools as well in the middle of the wxPython build process, and pip for good measure. The file compile at [15/905] sip/cpp/sip_corewxAccessible.cpp ran for many hours but didn't appear to crash like I saw on my Raspbian build with a Pi3-B+.

Unfortunately it had crashed; and without any kind of warning. The SSH session remained open; it appeared to be running; the light on the Pi Zero was even blinking like it was "working. I only figured out when I was unable to log in to the Pi with another SSH session. Had to force reboot; install a swap system (dphys-swapfile), configure (nano /etc/dphys-swapfile), and reboot again, before continuing the build.

Based on previous experience; it appears it's smooth sailing from here on. Don't say I didn't warn that this process takes forever... Probably exceeds 24 hours easily; even if it runs smoothly start to finish.

joerlane, 2020-01-10:
Looks like I might have underestimated the time for wxPython 4.1a to compile. It appears to be about linear with available cores in this case, so I'm now expecting about 96 hours total. Might as well start packaging this up for my own use at the very least.

I already did pack up an app bundle for Mac OS X. Tried py2app but the output only crashes so far. On the other hand, pyinstaller --windowed works just fine; and is noticeably faster than launching through python on the command line. The app is ~123 Meg and should be portable. I'm not yet sure how this would fare without windowed mode. It should be possible to produce a headless command line application too, by simply omitting the --windowed switch (did that by accident the first time). Also; not sure off-hand what is necessary to accept command line arguments in either type of application, is that already functional?

In terms of future feature requests. It might be nice to think about peripheral support on a Pi based setup. One thing I can think of off the bat is monitoring the water temperature. I currently use an Inkbird thermostat controller for that purpose; but recently noticed a variation in laser output across a small variation in water temperature with a fixed laser power setting. This is of course an extreme use case; but there are many possibilities with the GPIO included on a Pi board. 

I bring this up; because I gather one of the major benefit's of MeerK40t, is that you can keep the stock controller. This kind of thought experiment is less of an issue for people that already swapped out the K40's controller of course. There appears to be enough room to mount a full 7" Pi touch screen in the K40 case. I'll probably wait until I create an arm/Pi package before I begin testing that specific setup. 

Attached a pic of the MeerK40t Mac OS X app running (with MeerK40t icon). Cheers.

[image]
[image]

tatarize, 2020-01-10:
> In terms of future feature requests. It might be nice to think about peripheral support on a Pi based setup. One thing I can think of off the bat is monitoring the water temperature. I currently use [...]

Much of the hackable nature is intended to allow people to write these sort of things. It's, for example, entirely possible to stop the laser, by halting the data sending at the controller. In theory that would let the water cool down. So if you had temperature data you could, as a matter of course halt the laser when the temperature gets too high. Or you could, pause the laser and use some of the GPIO to change the Z-level control assuming you had a movable bed that could be controlled with a GPIO. 

There's also some chance that you could just use the GPIOs directly to control the stepper motors. And fire the laser. Though that's a bit more extreme.

--

Also, as a thing, I do need a command line process, but moreover one that doesn't require any GUI. Assuming you did everything right and you know what you're doing. You can make an svg that is perfectly reasonable to run, you can put all the lesser options directly in the SVG and simply execute that. Then the GUI wouldn't be needed at all, you could just execute the SVG and have it run the laser. It would also be plausible to give it a console like setup, or at a minimum pause and unpause.

joerlane, 2020-01-11:
I was pretty sure that command line process had come up in conversation before; and it's not at all difficult to achieve on the Mac side. I've seen them run both ways in the past. Where you reference the executable inside the .app bundle; as well as a separate command line only executable that can be installed in the system path, or simply called from the home directory (some Mac users especially are scared to install things in /usr/local/bin or /opt/whatever). 

I did some more research on the temperature sensing, and it seems like it would be arduous to support community wide. There are simply too many different GPIO libraries, and too many ADC and temperature sensor options to provide any kind of logical support path. That said; it is something I want to do for my own needs. Indeed; a high temp shutdown would be ideal. I do not run my laser water above 20C; and I keep my fine detail work below about 17.5C. I might even prefer to have a low temp hold as well. So the laser job won't start until temperature is higher than the low point, and will pause if the temperature exceeds the high set point.

Great point about the Z-height. I haven't built an adjustable table yet; but it's on the list. It sure would be nice to have that feature. The Pi should have no problem driving a stepper from any number of small driver boards. Still using a pair of manual adjust lab jacks here.

Pi Zero wx compile continues. lol. Thought it may well finish around midnight at this rate. I'll package it up and transfer it to my original Pi for testing as soon as it does.

tatarize, 2020-01-11:
Most of the GPIO specialty ideas aren't really things I would support community wide. The optimal way to support that is feed MeerK40t the data, ignoring how it's acquired, and write a control panel display or autoturn off thing as its own thing. There's some internal methods where different, fairly independent modules talk to each other, so the controller code talks to the Controller dialog and to the Spooler dialog. But, the controller code doesn't have any idea if anything is getting the commands it sends out. And, as a point of fact if nothing is picking them up they just go away. 

Basically you'd be able to listen to project('temperature') calls whether or not there was a thermostat telling you anything, as long as you deal with the returning None case which would be the default without a thermostat. Then if somebody wrote a thermostat and the dialog for it, you could always reuse the dialog since it isn't linked at all to the code or driver for that particular thermostat.

The modular signal system is core to the operations. Its main job is to make sure communication channels exist without requiring one part be completely linked to another. I knew from the get-go that the system needed to be threaded and largely decoupled, so I could have things interacting with the controller without the controller needing to know what those things are. Things like listening to the position the writer is currently at is important to make a fairly agnostic set of dialogs. That can listen to those calls. (This is also so I can change to a different controller type for a completely different laser and keep the rest of the software.)

Temperature would work the same way. There'd be a module that gives out temperature readings and another that reads those readings and does all the gui interactions. While it's true each different temperature sensing thing would need its own code, it wouldn't actually need anything more than reading the temperature and feeding that into the messages. Since other code would do the rest. You could easily imagine setting up a bunch of different readings and dials and such without any data coming in from physical devices they'd do nothing. But, if there were dials you could have a flowmeter, thermometer, door ajar sensor, air assist on/off checker, or anything else that it might be nice to stop the device if they happened or give an invalid value.

The support paths for all that stuff though would simply be out of scope of the main project. I don't know how your various hacked together sensors work. If you feed a raspberry Pi a copy of the end stop signal, it would be perfectly reasonable to stop and notify the user that you struck the wall, figuring out how that data should come to MeerK40t isn't as much of a concern.

joerlane, 2020-01-11:
The wxPython compile finished, and I build an executable bundle for the Pi (under Debian Stretch). It appears to be functional and be portable. The Mac OS X application bundle is also working well, and has a few more hours of testing on it. I'm not sure what else you have on your list before making binaries available for multiple platforms...

Thanks for the tips on modularization. And you're right; the more I looked into it, the more I realized it might not be a concern for very many people. I will indeed work towards my own solution for a few of those things though. It certainly does make sense to add a Z adjustable bed; and I've actually had MeerK40t try to drive my gantry into the end stops over and over (the stock controller doesn't monitor them except on home action?)... I've got a water flow sensor inline (not connected electrically), air assist, etc.... 

So; does it make sense for me to fork off and start a Pi specific set of GUI widgets and data collection? It's not my highest priority at the moment; but I've got most everything you mentioned hardware-wise. Hardware power control is also on my list; though I'm guessing that's just confusing considering the work you've done on pulse modulation.

tatarize, 2020-01-11:
I tossed you an invite with write access. That should allow you to edit releases and add those binaries to the releases. I know if I wanted to try it on my Pi that compiling it myself is plausible but honestly not going to be the best methodology.

It also occurs to me that you have understanding that I lack for writing a good article on how to get it to work on a mac. 

This doesn't actually give proper step by step instructions by any stretch.
https://github.com/meerk40t/meerk40t/wiki/Installing-Running-MeerK40t-on-OSX

Or the specifics for how you're compiling the Pi version, which the current article would have people install 4.0.7.
https://github.com/meerk40t/meerk40t/wiki/Installing-MeerK40t-on-a-Raspberry-Pi

---

As for the GUI widgets there are:
https://github.com/meerk40t/gui-files-meerk40t

Has the current wxGlade files for most of the dialogs currently being used. Though some got some other edits and changes since it was easier to do from the source than from the gui builder. A lot of different software supports both PPI and PWM changes as such, it's a bit scary since I just leave my dial alone. 

For my efforts most of the considerations I need to make are with regard to how to process this stuff. If I assume you have a custom rig that you've setup with your own highly nicely hacked stuff. For example you have a z-bed that you can control from your version of MeerK40t, what does that look like for how MeerK40t runs? Obviously, you'd need to issue that as a function that gets called in the spooler. That's how the auto-home and auto-beep are going to work. So an addon like that would need to permit a custom command to be added in, and it would need to allow the spooler the ability to halt for a bit, and wait for a blocking command to finish executing. Which it kind of already has since if you put a function in the spooler it will just call that function.  --- So if such a thing comes up, I'd need to automatically query for some specialty functions to add to the job and let people add them. Which could just be a custom .py file somebody adds, for permit these to be inserted into the spooler, or try to add this function automatically after each job, or call this function on the job itself and let it modify the data. Since that's most of the ways that stuff could be called or modified.

---

The controller gives you the same 6 bytes the controller is feeding back. Only byte 1 is known and is giving the control code which tended to be guessed at. If it gave information about the end stop it would likely change those six bytes but I think I tested it and I've never seen those numbers change. Their actual meaning other than Byte 1 is unknown. Part of the controller functionality is to maybe figure out what that stuff means, like if something causes the other numbers to change for some reason, that would be knowable. Maybe it'll feed data from the board or something but, if it provides that data, it must be in those 6 bytes and I've never seen it. What those numbers do or mean though is currently unknown. But, currently as far as I know, the board doesn't give any information on whether the endstop was hit. I made that information clear and public in case somebody figures it out and it does something really useful or something.

tatarize, 2020-01-12:
If you made a reasonable GUI that just listens to things that might not exist, and has some gauges or whatever, that'd be fine to include. Like if you read 'temperature' and assume that the data will either return None or the temperature in celsius. And if it returns a temperature outside the acceptable range it would broadcast a signal to pause the operations. That temperature gauge thing could be added prima facie to the main code since it's mostly going to do nothing for obvious reasons. But, to get it to work, it would need a few things.

1) Some code that isn't automatically on in the vanilla version that will interface with some temperature either through GPIO, usb, or maybe wifi, maybe read a file that keeps getting saved with the latest temperature reading, or some other unknown means and provide updates on the temperature. This code would need to run as a background thread in MeerK40t rather than as part of a particular dialog GUI. In theory then, the GUI would just interface with this background process.
2) Broadcast code to pause operations. (Pretty sure it currently only permits setting pause in the controller directly  and not through a signal broadcast).
3) Maybe some internal redesign to make sure this sort of add-on would work without the GUI itself. So LaserProject which currently does the signaling would also be responsible for tracking background processes or maybe finding them and launching them. 

In theory to have that possible without actually implementing, but also making sure everything else can integrate properly it would need some background threads to be able to be launched and interact with the ecosystem itself. So you could maybe have a background thread that does logging and have it write all signals to a file. Or another thread that listens to a telnet connection and provides information and statistics on the job, or something that rather than having a GUI at all, works like GBRL and translates gcode and sends it to the spooler and does that whole query system thing, but interacts with the rest of the ecosystem as such.

Keeping the options opens means that it might be more correct depend more heavily on the messaging system to deal with the interactions between the parts. So that the writer could be switched out for something that drives a Ruida controller, or the GUI could be switched out for a command line or terminal, or a background thread could check temperature through some means, and a different thread could read temperatures and maybe halt the laser if beyond the allowed ranges. And maybe a gui that could change those allowed ranges and watch the realtime temperature data.

In general even if it's something that's likely unique to you, making the system account for such things and easily adapt to addons as such, is going to help everybody else who also wants to do their own thing. Even without knowing how you're getting the temperatures, it would be possible to write in vanilla MeerK40t a background thread to check temperature and halts operations. And a GUI that adjusts those acceptable ranges, and displays the real time data. Even if nobody actually has the code to read the temperatures and feed it into MeerK40t.

joerlane, 2020-01-12:
Quite a bit to chew on here. Yes; I hadn't fully internalized that the wxPython install could be that current version 4.0.7 on the Pi. I probably wasted a bit of time compiling the newer 4.1.0a; but it does give me consistency across Mac and Pi builds. I'll take the time to read through requirements.txt and see how it compares to my bare metal install.

I did a bit of GUI modification last night by hand; but ended up with a Seg Fault 11. After sleeping on it; I recognized I should probably have been using wxGlade as was mentioned in the comments of those files. So; I will head in that direction and work towards something that isn't obtrusive to the rest of the application. From a pure simplicity standpoint; it might make sense to have a single checkbox in the preferences to enable Pi Hardware features. Though; what's to stop someone with Windows or OS X from wanting the same with an Arduino over a serial connection? Should it really be called Pi Hardware out of the gate? Maybe something more generic makes sense? Hardware I/O perhaps? Out of the gate Temp and/or Flow would be an Input, and Z would be an Output. It could activate either a Tab in the preferences, or enable a new button and dedicated configuration window. 

Or; maybe it makes sense to make it even more generic. And literally add support for custom.py; and let all else happen externally from that link? I'll leave some of that thought up to you; but I do agree that adding a few hooks could spur development interest across the board. 

Started testing Mac OS X build across versions, and have a bit more work to do. It appears I'll need to use a few old operating system versions and compile for OS version groups. Looks like 10.7 might be the earliest, and then another build around 10.12. Slightly annoying; but this isn't my first rodeo. I've started preparing a partitioned USB boot drive so that I can test each version to ferret out what's needed explicitly. 

Also doing the same for the Pi; I've got a Jessie build loaded to test backwards compatibility there.

I'll poke around for a bit and update/upload when I know what the compatibility limits are. Cheers.

tatarize, 2020-01-12:
No, I mean my instructions for the Pi would install wxPython 4.0.7 version but really that won't work. It would quite likely use GTK and thus fail the transformations until 4.1. It would have to be compiled to use 4.1 unless somebody has an already compiled copy somewhere. But, I don't, as a matter of course, have instructions that direct how to compile wxPython 4.1.0a+ yourself or currently have or direct anybody to a precompiled version. You would tend to have that information and with the invite could edit the wiki for that info, or edit the last release to include a working bit of compiled for Pi stuff.

---

wxGlade is the way to go, since hand modifying that stuff is too hard and you can save out the project files and upload them to the gui project ( https://github.com/meerk40t/gui-files-meerk40t ). You can just upload them directly to master since that data isn't going to ever be mission critical. I should write up the methodology for making icons though. It's a bit weird to make pyinstaller directly package the icon files so they are properly converted to text and saved in the icons.py file, it's a bit cryptic. And I need to make instructions on the typical icon styles and how to add some new icons to the icons files. GUI design takes a skillset I somewhat lack since it's more graphic design than code.

---

I certainly think a custom.py sort of thing is the way to go. Since there's no predicting what somebody's workflow is going to be. I could not possibly know every different way a person is measuring temperature, or lid up input or anything else. You can just make the gui listen to some particular values and update whatever gauges or whatever it has. But, the gui code will only really run while the dialog is up and running. So it actually needs an ability to run things in a background thread that do directly connect to the UI. So you could have one code section that will pause a job if the temperature gets too high, and a different code section that will check the temperature. If you don't have pause if beyond the acceptable temperature range code running, you might still want the check temperature code running to feed into the gui that shows the user the data. Or have a bit to do logging and save all the different things going on in the program. 

But, most users aren't going to have a temperature gauge and so they won't need either code section. I do think the GUI stuff should always be included but simply detect that `temperature` is returning `None` and go ahead and say that in the dialog.

As another example, I looked into getting a camera working and there's some dependencies you could use, but these might shift based on the various hardware and different setups and different OSes. But, it might be perfectly easy to have code that will set a background image if a `background` signal is sent and display that in the scene. Then if you were actually so inclined you could code up your own methods for making these signals that change a background image or background color in the scene. Then, later, if a GUI was made for cameras in general, you could give it an "update overlay" button that would just capture the camera image, apply the matrix or grid transformation to the image, and send that as a `background` signal. Or if you wanted you could code up your own GUI to just set the background color. It gives a lot of versatility. Or use some background thread that merely gets an image from somewhere else and sends it to the scene. So it doesn't need to use MeerK40t for that, because maybe you have a better way of doing that.

All it would need in the short term is the ability to trigger custom background threads in MeerK40t and for the scene to listen for the signal `background` and apply it if has an acceptable type.

With a system monitor gui or whatever, you could listen to a bunch of static hardware information streams, even if those streams don't have any methods to be populated with data and would require later custom code to get working. So with a Pi you'd need a little custom code to read the GPIO pins for your thermometer, but everything else would know `temperature` is a usable signal that will usually just return None. 

There are also some functions that aren't trivial and might require some other python dependencies that I wouldn't want to support out of the box. For example potrace ( http://potrace.sourceforge.net/ ) is some very nice software for converting images into rasters. But I dunno if I'd like to add it as a dependency to ally somebody to run that as a function. But, I could certainly see including that as an addon. But, it would need to show up as a function for images but only if the custom code existed. 

Or perhaps my future text code sucks and you'd rather do what Whisperer does and simply use inkscape to do that action. You could save a temp svg, then call inkscape with `--export-text-to-path` add the paths to the scene. I wouldn't necessarily want that function in MeerK40t proper since it depends on having inkscape, but I would want that custom bit of code to be easily added to MeerK40t without too much trouble, or needing to hack it in directly. It would simply need to take 'text' elements and give them a custom operation. There's likely a bunch of custom packages and addons people might like to use, but would not need to clutter up MeerK40t's basecode since maybe that function or whatever is only available on Linux, or is proprietary, or GPL code (which co [... cut ...]
I might have to look deeper into how Inkscape does its addon system and borrow some ideas from that. Especially since there's clear advantages to not directly requiring python code. I wrote a pretty extensive Embroidery IO library (pyembroidery) (obviously not a library that should be a direct requirement) and maybe somebody might want to open a .DST file for some reason, or maybe somebody wants to load lightburn files (.lbrn), though really that'd be fine to add directly into MeerK40t.

I'll open on issue on an addon system. Got a bit away from me there. And it's an outsourced problem anyhow and rest of the code base isn't there yet anyhow.

joerlane, 2020-01-12:
Boy am I glad you mentioned that. I've spent a few hours messing around with the old wxPython 4.0.7 on the Pi; thinking it was going to work with requirements.txt. It appears to me that pip is trying to compile it anyway; which can be painful on a single core. I might have otherwise tried to let this go another four full days if you didn't mention it... Only to find out it doesn't draw.

Anyway; there are a few idiosyncrasies with backwards compatibility. In this case it was also the python version dropping back to 3.4 (or 2.7). Is there a reason to support, or not support binaries based on python 2.7? How is your .exe built? 

I had been aiming for the best version of python 3 supported; but that specific dot version appears to be a limiting factor. I need to seek out lowest common denominators for best backwards compatibility. I could probably avoid a recompile of wxPython if I downgrade my python3 install to meet older requirements for the Pi. Mac OS X has similar limitations; on top of some other pesky OS specific libraries. 

I got wxGlade installed; and downloaded the GUI files. Looks quite a bit easier than what I was attempting earlier. Do you have a larger size MeerK40t icon? The one I found through GitHub was 100x100 and 1024x1024 is supported.

I don't see a way to add binaries to the existing release. I'm not sure if it's a permissions issue or not. I'll work on both easy and hard instructions for Pi, and OS X as soon as each one develops. Going to take me a bit longer to get a (way) backwards compatibly binary built for the Pi.

tatarize, 2020-01-12:
The exe is built with Python 3.7. The reason to avoid Python 2.7 is that it's no longer supported as of 1/1/2020 it reached End-Of-Life. Mind you MeerK40t is currently still completely compatible but that stuff can go away, I was also planning on using ezdxf which dropped it's support for Python 2 a while back and is far enough along to be non-compatible.

However, that said for now everything is compatible, and part of the helpfulness of packaging them into binaries is they include their own version of Python. So if you have any version of python that works, (even if it's not the same across all the different packages) it will work for anybody with that OS.

The adding binaries things requires you to edit that stuff. Somewhere in the email you registered your account with there'll be an invite to MeerKittens group which will give you the permissions you'll need. Or here, dunno where it notifies you. ( https://github.com/orgs/meerk40t/teams/meerkittens ).

Less cropped:
[image]

Though the original picture was taken from: https://commons.wikimedia.org/wiki/File:Meerkat_At_the_zoo_Novosibirsk_Siberia.jpg as it had an acceptable license. And I think I edited it in Microsoft Paint. There's a reason for Issue #30. It's a stand-in for a better more brandworthy icon. It was something I needed generally before starting to publishing the code. It did not need to be good.

tatarize, 2020-01-12:
The windows binaries also don't automatically find the libusb dll. So I always compile twice, and edit in a fix to force it to include the binary. If something similar happens in another OS it might be kind of annoying. You should test this by running the binary on the OS and making sure you're not set for Mock USB and trying to send some command to the laser. It should say the USB is Not Found. It should not say "No Driver". If it does say "No Driver" that version is broken.

https://github.com/meerk40t/meerk40t/wiki/How-to-generate-MeerK40t.exe

joerlane, 2020-01-13:
Cool; I was aware of the EOL of Python 2.7, and I assumed you were packaging with 3.something. Spent all night messing around with python versions in Debian. Apt-get doesn't have packages beyond 3.4 available for this build; but I can install newer from source. Tried Python 3.5, and I still couldn't install wxPython; so it's up to 3.7 to match your .exe build. It appears OS X should be able to mirror that Python 3.7 version; but it too requires compile from source (to build the binary apps).

I have not seen any issues with the USB drivers on linux or OS X. I can double check how all that works, but I think the only requirement is pyusb which is packaged up with Meerk40t itself in both cases. I remember reading about permissions issues for either USB or USB/serial on Pi builds way back in the day. But this latest build is bone stock, so it should ferret itself out if it is.

I've got a friendly tester for OS X 10.15 (which I don't run because I don't want to be bleeding edge). I've found almost every single OS X installer back to 10.7 as well; so it's just a matter of waiting out a bunch of those as I have time. I'm pretty sure I know where the limits are, but I want to be sure. 

I tested the Mock USB on the OS X side and it does indeed say Not Found; appears to be looping commands through the spooler/controller indefinitely. I assume that's the expected behavior, and this is a known good build (for this OS X Version).

I had simply ignored that "join" email; I didn't know it was a "mother may I" conversation, so I never even opened it. I assumed it had already been applied to my user. Now that I clicked join; I see that I can add binaries to the release, etc. I'm going to post the OS X binary and have it tested in 10.15 immediately. If it works; I'll leave it up and work towards builds that have significantly more backwards compatibility. Same for the Pi.

tatarize, 2020-01-14:
Mirroring the version shouldn't matter that much. It should package into a single app and never see the python version outside the package. So any version that works will work. 

The Mock USB just doesn't force any initialization of the real USB device. If the USB gets Not Found it means it had the driver and the tried the connection. Which means it's likely in good shape.

---

There might be some issues in future versions with the language translations. Somebody elsewhere mentioned them, so I  pushed the timetable up. And they are currently working in the current Pull Request (#102). However, they require including a .mo file which might not get automatically included. Though, if they don't, it'd be fine. It just turns off the languages without them. Deleting the `locale` directory gets rid of the languages menu altogether.  And I got distracted making the tree menu thing look better and be more helpful.

[image]

Really terrible partial French translation is likely to get added next version, if only to serve as an example of how translations work. I should likely feature freeze and push out a version. Though pushing out various compiled versions for releases will be nice, it's also compounds the work needed to push a version. I'll also likely have to start rerunning the text extractions for the .mo translation files to maintain the needed keys as such.

joerlane, 2020-01-15:
I'm holding out on 0.2.4 binaries; but not because of you. I see you've made a few more improvements as well. I will indeed launch some binaries very soon; even if it takes some extra effort on my end. Yeah; there are tiny compatibility issues that are giving me a headache in terms of minor versions. Once I distill all of that I can release binaries that have windows of restriction in terms of compatibility. The nuances would bore even the most hardcore among us. Just trying to help out as I am able. I just don't want to upload before I've tested explicitly.

tatarize, 2020-01-15:
I think you mean 0.2.3. And no sweat.  Besides once that stuff is mapped out, it should be more or less the same going forward assuming nothing strange gets introduced, though entirely plausible. And frankly the more robust binary versions with installers and whatnot are still set as Phase 2. So your work is putting that nicely ahead of schedule. 

I had intended to feature freeze, but since then I've ripped out most of the guts of the code and figured a more general set of methods to allow addons and to kind of modularize the stuff that's already a part of the main code base. And since I altered so many different files in kinda fundamental ways I'll need to retest everything pretty robustly.

I will likely just disable most of the newer changed stuff, test and publish (disable the JobInfo menus items, and remove the rectangles from the selection widget). Though a lot of dialogs are pretty heavily turned off already with features that are suggested but not implemented, more of that is fine I guess.

joerlane, 2020-01-16:
No; I didn't mean 0.2.3, but that's no sweat give or take. I've been running newer code for a while; but still trying to get used to git's nuances (especially when building binaries at home). 

I think we can close this Pi crashing bug very soon. I'll spin up a few of those boards to double check 0.2.3; but I've already been digging into 0.2.4 issues. I see binaries being released for both about the same time. Seems like a good split anyway; 0.2.3 was nice and stable besides the random controller spank crash. I think my binary build works in new Pis; but still do want to ask a friend...

joerlane, 2020-01-16:
Oh yeah; I'm not rolling installers, just like you're making an exe. That's a key distinction for some people that want to try this kind of software. You mentioned rolling your driver inside for Windows? Is your .exe build 100% double clickable without any other installs? That's been my goal here in terms of packaging; which isn't always a breeze, when you start iterating versions  across the spectrum of Linux and pals.

tatarize, 2020-01-16:
In theory the driver might be required inside a windows installer. I haven't messed with it enough to be sure what exactly is required. Whisperer wraps their copy of libusb install inside their program installer and lets you install the driver. In MeerK40t the driver is included in the .exe with pyusb as a dependency etc. But, I haven't bothered to check the limits on this. Does MeerK40t including the .dll mean it doesn't need anything at all? Like it might just work on the OS without needing any modifications to the system. Or maybe there's an easier modification that is possible since I'm not installing a .dll to the system folder or anything weird. --- The plan was for installers, and maybe quicky, lemme put this in the right folder and run it installers wouldn't be an issue at all, or package the code in one of those many different methods modern linux flavors use. The obvious benefits of clicking the linux package installer finding the package in the database and clicking install would be pretty big, especially once the software develops a bit more and I do some stuff like make it run ruida or whatever.

But, since I know very little about the subject, and there's still stuff to do that I know a lot about that I'd be really hard to get done without me, I'm working on that stuff instead. Getting it working in any fashion on other OSes is clearly a big first step. People might be fine with whatever. I'm just not focusing on that at all for a long while because it sounds like you'd need a bunch of OSes and trying to install it a bunch of different times in a bunch of different ways and messing with install making utilities etc, and uploading installs to various different places. It sounds much harder to me than writing code that would save a project out as an SVG file, or getting a good API that'll let arbitrary parts that don't know if other parts exist work in a coherent fashion. Or even going back and getting the implementation for arcs to work exactly for Zingl Plotting algorithms. Basically priority wise it's pretty low for me, even though functionally it's one of the more important parts of the entire project.

joerlane, 2020-01-18:
I think we are done with the constant crash on the Raspberry Pi. I have a Pi3 MeerK40t binary build that likely works on the Pi4. Looks like I can create a .desktop file to make an exe-like build for the Pi. It's going to be about a week (or more) to see if the Pi1-Pi2 build even works...

Willing to test this binary tatarize? I think you mentioned you had a Pi4... That's the one I'm missing at the moment.

tatarize, 2020-01-19:
Went through and harvested the ideas worth making issues out of. So yeah, this is closable, and likely should be closed. Yeah, I'll totally check a binary of that.

Be nice to let my Pi4 run MeerK40t.

tatarize, 2020-01-19:
I guess I'll test it on my Pi before closing this issue. Though the initial problem is long since finished.

tatarize, 2020-01-19:
Test went fine #120 has my list of minor and more serious issues. But, the compile was fine.

closed 2020-01-19 by tatarize