meerk40t/meerk40t #86: Allow mirror image and home point in upper right corner

loulouki, 2019-11-10:
My laser is based on M2 but has a bigger bed and homing point is upper right corner (X axis is mirrored compared to the usual k40). By default designed are printed mirrored with Meerk40t, would be nice to have an option for people with machines like mine.

thanks

tatarize, 2019-11-11:
Yeah, okay. I'd guess just flip the X-axis and change the home position to the upper right. And I should likely think through doing lower right and lower left with the also relevant flips available as options.

tatarize, 2019-12-30:
I added the stub-options for this in the last GUI update. In 0.2.2 still needs to be hooked up and implemented though. Flipping the X and Y axis is pretty easy since I'll just change the character to go right or go left. And it'll just know the home location and set it to the value of the upper right corner.

tatarize, 2020-02-22:
@loulouki This was corrected in 0.5.1.

The option that were added in 0.2.2 were hooked up in 0.5.1.

closed 2020-02-22 by tatarize