meerk40t/meerk40t #88: Custom (multipath) Cross-engraving angle

SandUhrGucker, 2019-11-11:
Please add a checkbox (or parameter-field) for an angle to move/route when doing raster-engraving.
I want to engrave molds for Tin casting. First experiments seems not bad, but the visible scanlines are disturbing.
Material: Finpappe/Machinepaper/100%wood(fine grade):
[image]
coin with visible scanlines
[image]

tatarize, 2019-11-13:
Multi-directional rastering like that is certainly doable. But, the board itself has a raster mode that doesn't seem to like y-axis engraving. It seems like the ability is sort-of there. Might require me doing some more research. The alternative is to do lines of non-raster mode engraving. Which is fine and also quite doable, but lacks some of the elegance and would limit the max speed to about 35 mm/s since above that the machine starts messing up beyond that in raster mode. It would do clicks when it moves down to the next line which sound significantly different from x-axis rastering.

tatarize, 2020-02-19:
This is flagged as a potential option for 0.5.1, and added to RasterProperty. Should have an second pass option at the end. Still disabled though.

tatarize, 2020-07-16:
This is corrected in 0.6.1, now live. Crosshatch is added and implemented. Tested and Functional.

closed 2020-07-16 by tatarize