meerk40t/meerk40t #89: Cut order

matthewflinchbaugh, 2019-11-25:
Currently cuts are only available by either selecting the individual paths in the order you want to cut executing, starting without autostart, then select next and repeat. This causes paths to be cut that are not optimal, and sometimes the outer cut then inner cuts which creates issue when the part drops free. 
Would be a good option to assign color to cut ordering.

tatarize, 2019-12-31:
Previously there was no location for the cuts to be optimized at. You'd need to perform this action at in the main gui if it existed. In the next spec'ed version ( #102 ) there's actually a place where this feature belongs. It's currently being used for actualize images but, there's no reason it couldn't add a command to find all elements flagged with operation "Cut" and make sure they occur last and cut from innermost to outermost.

tatarize, 2020-02-19:
This is partially fixed. While it does not do the required inner first optimization it does as a matter of course put the cuts last when adding them to a job.

tatarize, 2020-07-16:
This is corrected in 0.6.1, now live. Colored cuts and inner first optimization.

closed 2020-07-16 by tatarize

camrbuss, 2020-11-03:
Instead of opening a new issue, I thought I would add here.

From what is stated above, it makes it easy enough to put any of your cuts/engraves that you want to happen before the final cut in an operation the precedes the final cut. I usually do this just by setting a RGB color in inkscape. The issue that I run into is when I am cutting out a rectangle with curved edges. The curved edges and lines are represented by individual path elements with M or m attributes for d. It could be left up to the user to get these elements in the correct order so the laser cuts the top line, the right line, the bottom line, and the finally the left line, but often the path is parsed so it does the top line from right the left, then it does the right line from bottom to top. This causes an incredible amount of wasted time even with rapid moves and affects cut quality with interruptions. 

I believe that it would be possible to determine if a series of paths are in a "closed" loop by guessing if the endpoints of svgelements::Arc or svgelements::Line are within a certain tolerance of each other. Once you determine if a series of paths is closed, then proceed in re-arranging the path elements in the right order to allow for one contiguous cut around a closed loop. Maybe the Inkscpae Gcode tools extension has something that could be borrowed?

I am happy to help contribute, but am still finding my way around the code base. Would this live in svgelements or in something after Classify so that it can only be applied to Cut operations?

Would an issue be better suited on the svgelements repository?

tatarize, 2020-11-03:
Opened #256 reply will be there.