meerk40t/meerk40t #91: Speed limited to 240

matthewflinchbaugh, 2019-12-14:
If you try to increase speed by more than 240 it defaults back to 240 this should not be simply because of different setups people are running, may be capable of fast speeds.

tatarize, 2019-12-16:
For the laserspeed.py code I loaded up the Chinese software and a macro thing and spat out all the speed codes, for various files. I produced all the codes since it hadn't actually be correctly figured out by Scorch. I spent the better part of a couple weeks on it and since my goal at the time was to get Scorch to use it for Whisperer (which he did). I provided a giant test library that aptly tested all speeds for all devices and displayed the code's ability to accurately duplicate the Chinese software speedcodes exactly (including the bugs). 

The reason the speed maxes out at 240 is because the Chinese software did. Actually when you put the speed above 240 the software actually gives you a speed of 19.05mm/s. So I might have maxed it within MeerK40t too, since that was a bit pointless.

Also, for whatever reason the speed codes are off by 9%. Any speed you give it will physically run 9% slower than you set it to run, across the board. A fact that was figured out by actually timing the machine itself. ( https://github.com/meerk40t/meerk40t/wiki/Physical-Speed-Measurements )

When I was coding this up, I opted to use the Chinese software codes rather than even my corrected codes, since I fiddled with it a bit, and realized I'd have to map out some more stuff, when you try some of these disallowed codes they actually do crazy things and sound horrible. So rather than risk it, I just left the codes along.

Removing the max speed is likely sort of fine. Here's the code that does that. Just comment it out if you want to try it. It's in LaserSpeed.py. 

```
if mm_per_second > 240 and raster_step == 0:
        mm_per_second = 19.05  # Arbitrary default speed for out range value.
```

When you enter in some crazy codes it'll try to accelerate from 0 to 300 instantly and throw some problems I can't even get it above 50mm/s without it sounding like it wants to rip a hole in things. I duplicated the functionality of the Chinese software, and that software had these training wheels on it. So I duplicated them too.

Another thing that could be done is manually set the gearing. While I call it gearing, it might be something else that I just don't understanding like the microstepping or something. In any event, in the speed code there's a number from 1 to 4 in it, that controls something. The chinese software moves these numbers up according to the speed. So 1 is for speeds 0-25.4, 2 for 25.4-60, 3 for 60-127 and 4 for greater than 127. These shift at different times for rasters. 

You can actually run the same speed encoded in multiple different ways. Some of them sound like they are trying to murder people.

    > from LaserSpeed import *
    > LaserSpeed.get_code_from_speed(25, gear=1)
    'CV1911931026000113'
    > LaserSpeed.get_code_from_speed(25, gear=2)
    'CV1911932026000113'
    > LaserSpeed.get_code_from_speed(25, gear=3)
    'CV1891933026000113'
    > LaserSpeed.get_code_from_speed(25, gear=4)
    'CV1871934026000113'

The 3rd gearing has a different equation so it changed more than just the 7th digit. I'm thinking maybe it would be best to open this stuff up for users, but simply default to the kid-gloves version.

tatarize, 2019-12-30:
I added the stub-options for this in the last GUI update. In 0.2.2 still needs to be hooked up and implemented though. Also, might be useful to force limit them slower than the speed code maxes just as a general point. Because really I can't vector at 240 regardless whether that's the hardcoded max or not.

tatarize, 2020-01-20:
I misunderstood this. #118 will fix this.

tatarize, 2020-01-21:
This was fixed in version 0.3.2. The added options I have there could be removed as not-requested.

closed 2020-01-21 by tatarize