meerk40t/meerk40t #92: Invalid wx.Gauge range in JobSpooler.post_update_on_gui_thread

jkramarz, 2019-12-22:
https://github.com/meerk40t/meerk40t/blob/57c4c3c957bece764ea27cdab91ed1f0a511a759/JobSpooler.py#L242-L251

I'm getting `C++ assertion "pos <= m_rangeMax" failed at ../src/gtk/gauge.cpp(95) in SetValue(): invalid value in wxGauge::SetValue()` because gauge_controller's range is set to 100 and SetValue is called with self.buffer_size=161.

After reordering SetRange and SetValue calls, range is getting set to 600 and exception is thrown when calling SetValue with self.buffer_size=1601.

Should it use GetValue() or rather GetMax()?

jkramarz, 2019-12-22:
OK, it's already mentioned in #65 and #84.

tatarize, 2019-12-23:
Yeah, I think I've gotten around to fixing it linux in the new version but there were a number of other smaller issues for linux that I didn't correct. The bug doesn't affect the windows versions the best I could tell but would crash linux and I don't run mac to see if that crashes or what quirks it develops.

I'm going to need to make sure major and apparent linux crashes are dealt with at a minimum. But, I'm holding off until the current version gets the kinks worked out and is at least on par.

tatarize, 2019-12-27:
This specific issue should be fixed as of 0.2.0.

While there are larger Linux and other OS errors, this is no longer one.

closed 2020-01-11 by tatarize