meerk40t/meerk40t #94: Problem with dimensions in SVG file

jkramarz, 2019-12-22:
When importing [simple SVG](https://pastebin.com/ZVWG6Pck) file saved from Inkscape:
[image]
I'm getting quite unexpected result:
[image]

tatarize, 2019-12-23:
There were a number of small but pressing bugs in the svg reading. I've spent the last month and some change getting all of them fixed. There's a couple minor deficits most of this stuff should be done and work for basically all svg.elements now. See Pull Request #90 for the version working with the new and corrected code. Here's your result in the new code.

[image]
[image]

You'll see it hits correctly on length and height. And should be fine on the x and y too. There's a bunch of different ways svgs specify that stuff and I went through and all of them. To the point of passing most all the W3C_SVG_11_TestSuite rather nicely.

Basically there was a serious need to get this right so that I can copy whisperer's workflow where people will very often do all their alignment work for images in inkscape and then reload the svg in the laser cutting program so I needed all that alignment stuff to correctly work.

tatarize, 2019-12-27:
This should be fixed as of 0.2.0.

tatarize, 2019-12-27:
#90

closed 2020-01-11 by tatarize