meerk40t/meerk40t #95: Mac OS X interface issues

joerlane, 2019-12-25:
I noticed another bug reported about SVG dimensions. Currently running OS X 10.14.5 with both python 2.7 and 3.7. With either version of python; the path tree on the left side is unresponsive. I get no response clicking on the file name, or any individual path. As such I can't make any changes to the settings for each component, or the entire file. I am able to scroll the list, but nothing else.

Furthermore, the loaded file does not try to display in the grid window until after the Execute Job button is pressed. When it does try to show the paths, I see long lines headed out to an infinite distance. Interestingly enough, when running the job; the laser cutter will cut the actual and correct paths. I simply can't visualize any of that process in the software.

Attached are screen shots showing the views from both python 2.7 and 3.7 which are only slightly different. I did zoom out quite a ways in one case hoping to find the actual SVG paths drawn, but the blue lines never move. I experience the same problems with any SVG I've created in Inkscape (for Mac OS X); simplicity or complexity does not seem to have any bearing.

[image]
[image]

joerlane, 2019-12-25:
Just wanted to confirm ahead of time. The driving of the laser cutter hardware works fine with the manual power adjustment. I simply can not adjust any speed or power settings in software because of UI issues. 

I am able to delete single jobs from the spooler; or all as needed. No way to select any of the individual settings this way, so it's defaults every time for every path. This is two copies of the same file. A simple 3 rectangle SVG. Set smaller than the size of the bed this time just to be sure. Should I try switching units to inches or otherwise for fun?
[image]

tatarize, 2019-12-26:
There's a couple different methods to access the needed UI. Double clicking on the path would do it in either the window (which might be hard because it's weird) or in the tree. Also, right clicking which might require holding a key down in Mac to emulate (but doesn't give you access to the properties). Also, if you launched it from command line, it'll show you errors if they occur. If it threw anything that could be tracked down pretty easily.

The problem here are likely rather small, like change when an update happens or the display commands, but since I didn't have a copy of Mac OS to test it against small irregularities become insurmountable problems. I can launch it with Linux and I'll try to fix the panning bug there once I get the new version online which hopefully should be soon but I changed enough that I basically have to reproof everything. And maybe getting that fully functional will inadvertently fix the other bugs in Mac.

I am however much enthused to know it actually does work on Mac save the UI issues. Getting beyond these will likely take somebody with MacOS and some knowhow, though I might be able to embezzle a Mac or build a Hackintosh, and I'd be leaning towards the former. Also the newer version will, for a number of reasons let you set the power within the svg. So <path d="..." power="500" stroke="green"> would be read properly. So even without fixing the problem circa next version you could certainly test that feature.

joerlane, 2019-12-26:
In response; clicking or double clicking isn't working in the tree. Those lines to infinity don't respond to mouse clicks either. I use a USB mouse that does control clicks (right click) as well as a scroll click (mouse 3). If I try to load a file with many more paths; I still can't click on any of the lines to infinity. Also can't see, or select the objects, or any of the parts.

These problems are indeed rather small. I would be happy to explore with any kind if debugging. I would also be happy to install any linux flavor in attempts to resolve these problems. I've got a few testing environments loaded with Debian if that helps.

Happy to help with any code issues once I understand any of the rationale. As you mentioned; the core driving code (hardware control) works fine. It's simply the UI that's lacking on my Macs. This isn't uncommon at all with Mac OS X, so you shouldn't worry too much about your quality of code. I was simply curious if those lines to infinity were telling in your investigations of SVG files. Those huge numbers in the path tree were strange in my mind...

tatarize, 2019-12-26:
I'm thinking the side panel there just isn't accepting clicks. I can run linux versions and should be able to get that done. The numbers in the path are the hashes of those paths. They don't actually mean anything they are just unique ids. I think the blue lines aren't to infinity, I think they are on the interface. Like they are somehow part of the guides at the 0 point and somehow there's a bug at that that causes it go weird.

It seems like you should be able to pan which would quickly reveal that as being the case. However, the Linux version doesn't pan and maybe the mac version doesn't either. There's also clearly a bug that I also saw on the Linux version where the background for the view is only supposed to be white within the working field. So at a minimum several of these bugs are shared between mac and linux. So fixing one should fix the other. And I can just bootup my thumbdrive linux and squash the latter bugs.

[image]

I might just push out PR #90 with a couple bugs. Since the version has changed a bunch with a couple months of work on the SVGs. But, it still has some critical elements that need fixing. Most everything else seems to work. Then try to figure out the couple bugs that show up in Linux too. And hopefully make some methods that work for macOS even if I can't figure out what causes the other element.

tatarize, 2019-12-27:
Yeah, doing this with Linux, I also loaded an item and can confirm that most all of it looks like you described. I believe the problem to be the bindings have no effect when they are bound to the sub panels. So things that are done through the binding like mouse and keyboard entry or some refreshes as such are simply not done. The other dialogs don't have subpanels.

tatarize, 2019-12-29:
Mostly finished with the fix. One main issue though is that the transformation matrix doesn't actually work outside Windows. Some other bits said it would work for some versions of wx.python with GTK3+ and OSX. I checked whether it was supported and held off on the rendering the scene if it wasn't. The svgelements code has a fully fleshed out matrix class with all the bells and whistles so I'll need to apply the math and emulate a viewport myself. But it'll be pushed to next version, at the minimum. The linux branch has some somewhat consistent stuff, still needs to be better designed since it's drawing on the frame rather than in the panel and not keeping everything laid out correctly.

Next version will not display the scene for you but will give you control over what's happening and properties etc.

joerlane, 2019-12-29:
I wrote this whole post af while ago. I'll happily test any update you send down. You seem to understand well how difficult it is to fix these weird glitches.

I greatly appreciate the updates on this issue. I spent a few hours reading MeerK40t.py, and a few other files end to end. While I don't have a lot of experience with wx; it comes up often enough in cross platform open source software I might as well work towards a passing understanding of it.

I ran MeerK40t.py with each python 2.7 and 3.7 from the command line and got no errors. In verbose mode; there wasn't any kind of response when I clicked around the main window's subpanels. I am not able to pan explicitly in the bed/graphics view; but I can change the zoom and move around that way. Unfortunately I'm not at all sure it's updating the loaded drawing when I do that; so can't be sure if the object is actually drawn down lower right where I can't see. Every time I open (add) a new file it re-zooms the view window (and probably re-draws?).

Similarly I can scroll in the tree pane if I have even path elements, but not successfully click on anything there either. It appears to me that both left and right mouse are ignored in both of those two subpanels. Furthermore; I think there is a burn button included in the tree's panel which doesn't function for me; though the one at the top next to file open does work. 

Not sure if this is helpful; but I tried to drag and drop a file, and that doesn't seem to work in OS X either. From my basic understanding of the source, it seems like it should work, as it's part of the main frame's code. I did find someone else mention having problems with global 

I might start off with a specific OS X app bundle, and also a wxglade demo app while I'm feeling bored. These are things for me; and not for you. I am so very inspired by your progress. I'll try what I am able on my end. Thanks again!

tatarize, 2019-12-29:
Finished the version that *sort of* works in Linux. Try the latest version. 

I changed the frame to stop delegating to sub-elements since that wasn't cooperating very well. One of the chief errors is that the wx.DC doesn't permit matrices in Linux. So my nice code to change the matrix to alter the view doesn't apply and it shows what basically amounts to junk. I now check whether that's allowed and simply display an alternative message if it's not. The other major one was the depth ordering so, I just moved everything into the Frame level. 

From the documentation the DC should actually work with GTK and OSX starting around 4.1 for wx.Python. So in theory https://wxpython.org/Phoenix/snapshot-builds/ might provide a build that would have the transform ability, return True, and display the scene correctly. It should still likely correct a lot of the other issues, even if it fails to draw the scene.

Feedback on 0.2.1 is certainly requested. If you're fancy enough to try a wxPython 4.1 snapshot that would also be nice. There's some chance that that with that snapshot it would totally work with OSX. 

The alternative is to do the scaling myself and just pretend you're zoomed out and just shrink everything. It's weird sounding but, the math is totally the same. Also, there's an extra advantage in that methodology, mostly it would makes wx.Python less necessary and might make a gui in TK far easier. Or REMI ( https://github.com/dddomodossola/remi ) (web GUI).

joerlane, 2019-12-29:
I haven't tested every feature with the laser hardware yet since I can't explicitly tell which path is which in the tree (without trial and error). Is there a way to fall back to the older style drawing in the view window instead of displaying a failure message?

Major progress none the less; I can now access the tree items and context menus. I will work on testing the speed and power features on some simple files. I'm going out on a limb here; but I'm guessing the paths in the SVG are written in the order they are drawn, so I should be able to figure that out in the short term.

I'll work on getting wxPython 4.1. Looks like my python 3 is currently using wxPython-4.0.7.

[image]
[image]

joerlane, 2019-12-30:
Upgraded to 4.1 via the .whl file. Removed the right align defined in the attached error which was on line 588 of MeerK40t.py. After that I was able to get the grid to display. Still does not appear to be displaying any of the SVG path objects though. I did try running through a fake burn to see if it would load then (the old version wouldn't try to display without doing that); but that didn't work either. 

Do let me know if there is anything else I should try, or try to reproduce. 

I loaded a .jpg file and it DID show up on the display. Only the file that was roughly 600x900 pixels shows up as about 15mmx30mm on the view display. Haven't tried engraving a raster yet, so I don't know what size it really is.
[image]

[image]
[MeerK40t-2019-12-29_19_34_47.txt](https://github.com/meerk40t/meerk40t/files/4008754/MeerK40t-2019-12-29_19_34_47.txt)

joerlane, 2019-12-30:
Just wanted to offer a quick update so you don't waste any time. It appears that the SVG drawing display might be working after all. I'm not sure if it's the line weight I use, or a normal issue; but I can barely see the objects that are simple outlines. I took a screenshot of a few of those for reference. I can really only see them on the LCD screen if I look at them way off axis, far outside of normal viewing angles.

At this point; I can't find any more issues at all in OS X, I can now access all the menus and features that I was aware of. I'll do a full test with the k40 hardware itself just to be sure.

[image]

joerlane, 2019-12-30:
Looks like I spoke a hair too soon. I am still working with files created specifically for the k40 whiperer software. In this case, I have a bunch of paths that are black in color that I usually run as rasters. 

When I right click the path, and select Vector->Make Raster Image I get this error. 

[image]

tatarize, 2019-12-30:
The last bug there isn't indicative of Mac OS. The PC version will undoubtedly do that too. I haven't even checked but there is no LaserNode object `.bounds`. I removed it. Which actually means that that routine just didn't get properly tested and is a bug. 

I'll integrate the newer settings dialogs I cooked up earlier today, and fix that bug. I'm adding options for the suggestions made in the issues even if I'm not hooking them up right away. 

The routine to raster vector objects is a bit screwy. Like that is not a highly useful workflow. To force the user to click raster for all the images. But, I likely will have to change the Jobs dialog to support a variety of required changes like Actualizing Images, and Making Rasters of vectors that are expected to raster. And I'll need an operation to apply the rotary scaling, optimizing the cuts, and apply a few other pre-spooler operations.

tatarize, 2019-12-30:
> Is there a way to fall back to the older style drawing in the view window instead of displaying a failure message?

There is, but, it won't help. The window drawing is a failure. Without the transformation window all you see is a couple grid lines, a screen full of white background and everything interesting is outside the scope of that view and without a matrix, it cannot make a viewbox to see the interesting things. I think it's better off with the failure message. I can't properly draw anything without a working matrix. I do have a different methodology but it would take a bit of a rewrite. I'd just scale everything I draw down to the right size to fit in the window as it exists. But, it's not trivial and things like line weights would not be trivial. And from the 4.1 example, there might be functional versions coming down the pike anyway.

> I loaded a .jpg file and it DID show up on the display. Only the file that was roughly 600x900 pixels shows up as about 15mmx30mm on the view display. Haven't tried engraving a raster yet, so I don't know what size it really is.

It's really that size. The default for image files is to display them at native resolution. Since the stepper motors do 1 mil for 1 step that means the actual core native resolution of an engrave is 1 mil per pixel. Which means that 600x900 image 600 mils by 900 mils or 0.6 inches by 0.9 inches. You can increase the step size to like 3 or so and it'll be more reasonable. But, honestly keeping it at the native resolution is for the best and gives really nice results.

Some information is a bit old but you can check: https://github.com/meerk40t/meerk40t/wiki/Notes-on-Rastering by default it gives you the best resolution, even if rendering at that size might be a bit odd. You're better off using good graphic editors to plot your rasters as a deliberate action.

I added the more typical include the image in an SVG and let the user manipulate them with affine transformations so it's less restrictive and people can use the same workflow they used with Whisperer. Though it required an operation to convert a transformed image into ones with real pixels rather than transformed pixels. But, the image default for loading an image, is still the older best-resolution method.

tatarize, 2019-12-30:
Changing lines  47 in LaserRender to:
```
       xmin, ymin, xmax, ymax = group.scene_bounds
```

And line 1491 in MeerK40t to:
```
       xmin, ymin, xmax, ymax = group.scene_bounds
```

Will fix that other bug.

joerlane, 2019-12-30:
Thanks for the tips on raster images.

I tried the group.scene_bounds modification earlier while trying to hack it working; but kept running into this problem. I even did a fresh pull and made the changes to make sure it wasn't something I had messed up. Probably something simple I'm missing; but I haven't figured this one out yet.

[image]

tatarize, 2019-12-30:
Okay, I updated the version to 0.2.2 it should correct those issues for you. Both crashes should be properly corrected (I'd hope). Did the update before seeing that error there.

But, it will also implement the PenWidth / SVG StrokeWidth connection. Since the line width does nothing I never bothered to implement it. But, you could barely see the lines so I figure it's an issue with MacOS widths there being really tiny. So I've added it properly. Hopefully you should be able to see it now.

Not really sure why the group.scene_bounds is none. Maybe email me the svg you're using at tatarize at my gmail account. It really should have a size there. I'll see if I can get mine to crash.

tatarize, 2019-12-30:
Next version will read:
```
    def make_raster(self, group):
        flat_elements = list(group.flat_elements(types='path'))
        bounds = group.scene_bounds
        if bounds is None:
            self.project.validate()
            bounds = group.scene_bounds
        xmin, ymin, xmax, ymax = bounds
...
```

If scene_bounds is blank, then validate and get scene_bounds. Though I'll likely have to look if that problem will crop up elsewhere, or simply force the validation on load. If this None value for scene_bounds going to be a common issue.

joerlane, 2019-12-30:
Pulled the latest, and then updated the LaserRender.py as you mentioned. I can now rasterize the paths without error.

I checked into the stroke width. Turns out it's not an OS X issue that I know of. It's that I have been using a stroke width of 0.035mm in many of my SVG documents. Looks ok in Inkscape; but it appears those were carrying over to be almost invisible here. I set a more reasonable stroke width and those show up normally in MeerK40t.

tatarize, 2019-12-30:
Before 0.2.2 these widths should have not been read at all. Version 0.2.1 and before treat all line widths as pointless and do nothing with them. But, now the widths should matter. Just hadn't bothered to hook it up since the width of a line does nothing in the program. All lines move the laser, nothing else. And wideness isn't something the laser can care about.

tatarize, 2019-12-30:
Updated to 0.2.3 to fix that bug. Should work on a clean pull now. Though won't change anything from 2.2 with that correction.

tatarize, 2019-12-30:
From the bottom of my heart though, thank you. I honestly had no way to check OSX for functionality. It was basically a complete nonstarter without you.

joerlane, 2019-12-30:
Glad to help out; and I will continue using MeerK40t as much as I can. It's really nice not to have to maintain an old Windows box to run a K40, and even nicer still that I don't have to rely on the included software since it's such utter trash. 

I often do a few passes at different speeds and/or with manual power adjustments. Being able to automate even parts of those experimental procedures will be a great time savings. Also; the granularity of being able to delete paths individually, and spool multiple files and jobs together is rather handy. Much appreciated!

closed 2019-12-30 by joerlane