meerk40t/meerk40t #98: Use Rotary Settings - seems to be ignored

matthewflinchbaugh, 2019-12-29:
When trying to calibrate for rotary on the y axis, checking the box to use rotary settings does not seem to actually be utilizing the scale factor entered. Seems to just completely ignore it.

tatarize, 2019-12-29:
Yeah. That's accurate. I didn't actually hook that to anything. Since I never really came up with the sort of preprocessing it would require. In theory the entire scene should have a matrix which is applied and realized to the shapes and the counter matrix would be applied to the view. So unmodified images would require being actualized since they'd need to be scaled. For the shapes it'd be fine though. There's also need to change the home position or the directly flip the x or y axises.

tatarize, 2019-12-29:
Actually with the modern version I could likely just apply a scale to it as it's sent to the spooler. The tracking location would be a bit weird since that's filtered back from from the writer. But, everything else would be fine. Some of the generic sent jobs wouldn't be affected. I would need to make the break from the scene to the spooler a bit more real. Rather than actualize transformed images, I'd need to copy them and then actualize those modified forms within the spooler. Which is likely a better way to do such things anyway.

Whisperer implements that feature basically the same way. Modified the image width according to the scale and scales the points vector points and rounds them to their newer location (which isn't going to cause more than a mill of error). It very well used to be a bit harder of a problem, now it's just a feature that'll require applying a matrix before sending to the backend.

tatarize, 2019-12-30:
Version 0.2.2 adds a stub rotary dialog. Still nothing is hooked up but it looks cooler. And has more not-hooked up options. Really I'll need to add in the processing routines between the Job and Spooler to perform those actions. Also added a bunch of other options in settings that are also not hooked up.

joerlane, 2020-01-17:
So; assuming someone had hardware on hand. What would be the best way to hook up something like this? Would a 4PDT switch, or 4PDT relay be a reasonably assumption when using a rotary axis on a multi-purpose machine? 

The stepper motors themselves would be switched in and out to the controller port; and from my understanding the holding power to the motor should be explicitly cut during the time of switching (unlock rail command?). 

I think it stands to reason that dedicated users would "set it and forget it" in terms of wiring and preference switching. But as a casual user; I would like the option to try round things....

tatarize, 2020-01-17:
There's some nice arrangements but usually people unplug their y-axis and use that signal to control the rollers as such. Then everything tends to work correctly, except that there's no longer a 1:1 ratio with the y-steps and the steps on the surface of the thing being engraved. Which is why there's a setting for that. Though it doesn't currently have any effect on the encoding. Software wise I just need to stretch the shapes out in a preprocessing step.

joerlane, 2020-01-18:
I'll put my hat in the ring here. I'm not sure how long it will take for assembly; but I do need a small rotary encoder for some specialized work. I would prefer it to remain small enough, that I could leave it in place full time and just switch the motor as needed. I know it's overkill; but a quad relay bank is a viable option for that considering (for me), those jobs will be short, few; and far between. 

I want a simpler design actually; but I found this as a jumping off point at least. It has some good resources to start from, including gears and frame parts. https://geeksmithing.com/rotary/

I was also considering ways to make it even more on the fly; and I suppose once could clip to the existing Y belt and do some mechanical translation for simple objects. My goal is small marks on the exterior of a rings in this case, so I have broad design latitude; in that, I won't require 360 degree rotation, or laying down large cylinders.

I could also manually scale my designs in the vertical axis to account for skew in the short term. Don't even really need functional settings to get started and experiment. No rush on this for me, but I will report back when I have something connected for testing purposes.

joerlane, 2020-01-18:
I started collecting available hardware to build a rotary adapter. I also did a google image search and looked for unique designs specifically. I found a total of four that I considered to be unique from each other in some way.

What I gathered from that was there are two or three basic styles, depending on one's perspective. A simple direct drive lathe style which has a 1:1 rotation of the object to the stepper motor's output. That requires an object diameter setting; which is independent of the rotary system. Then there are numerous belt drive systems. Those have both a tooth count ratio; and drive roller diameter to contend with. But the object drive distance becomes absolute, and the object can be any diameter as long as it's large enough (otherwise it will simply overprint). There is also a semi-direct drive modification of the double roller design which again requires an object diameter setting (for each object type or dimensions).

So yes; it could always be distilled into a single scalar, but it could also be calculated based on the specific design. Some lathe style units have a gear ratio too; so there are variations among even those. In total, I believe it's: gear ratio, drive roller diameter, object diameter, stepper motor steps per rotation (and micro-steps?)? Some of those would be hardware specific, optional; or job (project, or, svg?) specific. Might be wrong or missing something; as I'm still in the design phase. Feel free to interject anywhere.

Being a work in progress, I also don't need any code for this to function on my end. I can do the math; and then work out the ideal set of constraints. I used to take apart inkjet printers for fun; and saved all the "cool stuff". This should be a walk in the park...

[image]

tatarize, 2020-01-19:
Yeah, the RotarySettings dialog stole the parameters from Lightburn basically because that's pretty useful for people. While you can do the math yourself you shouldn't have to. Most the parts are the same, so you'll have the same size wheel for example, so to calculate out the correct size you'd use the same parameter as before. You'd just give a different value for the one other thing. 

It will, in the end, come down to one scale value, but the math to figure that out might as well be done by the computer based on all the other math that is just going to be the exact same thing.

Basically you have chuck and roller versions, and the chuck is a direct 1:1 spin=device spin. So you need to set how many stepper motor ticks causes one revolution. In the roller version. The ticks control the roller that rolls the thing being engraved. So you have ticks to cause a roller to make a revolution same as the chuck. But the thing being engraved on is it's own gear and you need the direct diameter of that thing so you calculate the distance per tick.

Also, when you do that the shape of the field is not a bed. You don't have x and y. You have range of x and infinite range of y which loops back on itself, after a given distance. If you are actually using that, mathematically 1 turn +y is the same as 0 turns.  If you're engraving something it might be nice to have the project reflect this point. I have a foolish idea of actually visualizing this for the user by creating an infinite field. I likely would never get around to it, but the math is such that, you might have better optimized paths if you take into account that the whole thing is really looped. Take the example of engraving a single line on a cup that spirals around the cup in an unbroken line. Topologically that could literally just be a straight line. If the field is such that I let you draw a line and engrave, it would spin the cup while moving the laser head to the right and finish the project nice and easy. However if you didn't change the topology of the scene to show this you'd missing some stuff. And when items are put in the looped scene they'd be show in the correct spot at turn 0 even if they are located in turn 2. Properly duplicating the views (assuming the user's viewbox needs to see it) would let you see the spiral around the cup even though you only drew a single line.

Better visualization would make the math easier and loops are part of the charm. But for now you get a scale value that gets applied to the scene, and nothing else.

tatarize, 2020-01-19:
Also, there's a easier way to do the math. Empirically, okay here's some controls to spin the thing. Hit these buttons, when you've made 1 whole turn exactly. Then, hit this button.  It now knows exactly how many ticks equals 1 turn. And then you can give it the surface area and it'll throw that ratio into the mix.

tatarize, 2020-02-05:
This was fixed in 0.3.1

closed 2020-02-05 by tatarize