meerk40t/meerk40t #99: Backend API, Command Line

tatarize, 2019-12-30:
The front end GUI should be properly isolated from the backend writer/controller such that a command line interface as well as other GUI interfaces would be entirely permitted.

The K40Controller.py should work completely independent of other code and should include a default main functionality that would permit directly connecting and sending data to the K40. Such that the controller will work independently of any other modules.

The LhymicroWriter should work completely independently of all GUI code. Requiring the `K40Controller` for data sending. `LaserSpeed` for laser speedcode generation. ThreadConstant and `svgelements` for some interpretations of curves as such. This should also include some workable methods for sending commands, so that it could be used in a command line fashion, and for API building. This should also permit some kinds of middle language api stuff send commands that are properly written in Lhymicro code. This might need to be subdivided between these different functions.

joerlane, 2020-01-16:
MeerK40t.py should take file input with a switch; and determine if asked to run in a headless fashion. Does it need a --headless switch? I have no idea what switch makes sense here for file input; but -o for open file or -i for input file. Could be more specific like -s for SVG, or -p for vector path (-v should be verbose?). What about a project file? I believe that would be the standard in; and users could install a symbolic link to the executable in the path specific location (linux mac and windows?) for the console functions as needed. 

I'm not 100% up to speed on the dos/windows way; but I know you can pass command line arguments in almost exactly the same way there. You should be able to run your exe from the command prompt, and send it a file with the same switches. Once implemented; Linux and Mac versions should accept the same by default from what I understand. python3 MeerK40t.py -whatever file.svg -P 1000 -S 200 -T>16.2 -T<18.2 --Home --Lock --??

Also needs a --help file?

tatarize, 2020-01-17:
It seems like there would still be limits within the running of the various operations. MeerK40t.py properly is a gui bit and if you're going to run without a gui you should depend on no GUI code.

The kernel has some solutions there. Since it would be possible to determine if a service is needed and start it. So if you need something in the background, processing a queue of Lhymico-gl data being sent to the controller, you could start up the controller, and then feed it data you load up, and shut it down when you're finished, and exit. No GUI needed.

You could also, launch the kernel, have it launch the controller and the spooler/writer bit and then give the spooler some commands, which it would turn into Lhymicro-gl code and add to the controller's queue. Which would then send that data to the device and perform those operations. No GUI needed.

You could also, it have the kernel launch the GUI and keep running even when the GUI is closed. Or however else you'd want that stuff to happen. Basically the idea is that the GUI should be a secondary consideration. Most of the stuff runs without it, so it wouldn't be too hard to run tkinter as a gui, or that webinterface gui thing, or both at the same time.

I previously wrote a CLI for K40Nano-Tools ( https://github.com/K40Nano/K40Tools ) This had a number of pretty basic commands that basically worked like a rudimentary version of spooler.  It had -i <file*> -o <file> writing, -p passes a couple different move and cut commands. A speed command, a wait command, home, lock and unlike rail. Basically it would stack up operations to do, with things like -p setting the pass would duplicate the stack. Etc. The idea was to do things like allow you run 25 copies or the same raster image, allowing you to add a wait time between each run. 

The goal of that project was to give Whisperer a good API that wasn't cryptic spaghetti. But since it depended on Whisperer code and Whisperer is GPL3 I mostly abandoned that project, since anybody using that API would be forced to be GPL3 themselves. 

I daresay this project is well beyond that stuff now. I'm not sure what people would want exactly for the CLI but basically anything is possible. I would think MeerK40t.py itself would simply take a filename and open it. Since that's the most basic and commonly used process for basically all OSes. However it's entirely possible to write other Python scripts that would take the eventual system and go to town. 

So in effect MeerK40t would be the wx.Python GUI version that would load things up and give everything a nice gloss but wouldn't be required. You could write your own script that would load the kernel, load the K40Controller and LhymicroWriter and add them to the kernel. Then you'd have access in the Kernel to either send Lhymicro-gl data directly to the board or to send higher level LhymicroWriter instructions to the writer. So with that, you could easily write a script to perform any operations you want. You'd load the controller and writer, attach your own thermostat code, then load an SVG and feed it to  LhymicroWriter for writing. When finished, you could call shutdown. But, if during the operation the thermostat code detected temperature problems it would call on the K40Controller to pause operations.

The GUI hooks and stuff would just be reading the information being stored on the kernel and the various signals being sent between processes. So given that, the CLI specifics are basically unneeded. You could write your suggestion there as a script, that accepts those arguments, and run it exactly as you want. And somebody else could write something different. You could adapt the code to run entirely as an addon to Inkscape, where you set speed values and various parameters in inkscape. And it calls a command line whatever thing that loads the SVG, and those properties and exports to the laser. Literally all of your GUI in that case would be Inkscape.

MeerK40t should be more seen as the shiny gui being put on a fairly reasonable backend set of code that currently doesn't have its own name. So MeerK40t.py properly should accept a command line of a file that should be loaded in MeerK40t, but more expansive CLI stuff could use the system created for its own ends, implementing everything properly as a script. So a more robust CLI would be its own file, and could literally do anything.

joerlane, 2020-01-17:
The reason I brought this up was purely technical / conceptual. 

When I compile and application for Mac or Linux I specify the MeerK40t.py file. The resulting application is the embodiment of that program. If it were to accept command line arguments; they would need to hook into MeerK40t.py. 

On the other hand; if it makes more sense to you to use a separate program (meerk40t_cli?) to interface K40Controller and LhymicroWriter; that would simply be a second compile for the console applications. This isn't significantly more effort at all; and I simply wanted to make that distinction. I reason that you could probably even start with the shell of Nano.py (all the same command, etc.), rename it; and just interface with the MeerK40t components?

As far as I'm concerned it's a choice to be made; and little more than a philosophical difference either way.

tatarize, 2020-01-17:
Hm. That makes sense. Since I compile the windows version into a single file, there's really nowhere else it could go except to that file. It does make a lot of sense for end users to have that work that way. I was thinking about somebody who hypothetically wanted to hack on the code. It's not actually that hard to take the code that currently exists and make something new like a CLI. And it would be possible to work some CLI functionalities into MeerK40t proper. So you really could run it without the GUI if you wanted. Something like:

MeerK40t --autostart --nogui myfiletoburn.svg

Would seem to be reasonable. That's clearly something I could see somebody wanting. Especially if did something basic like let the wx.Python import fail if you gave it a --nogui flag. So even without the rest of the code, you could burn an svg, even if you didn't install wx.Python.

joerlane, 2020-01-17:
This makes complete sense to me. And, you might be allowed to omit the --nogui flag. In that; if anyone sends the -i flag for input file (or any other command line switch); then it would automatically start with no GUI? 

Choices here... Some people may expect to be able to send flags and load the GUI too; so it's more of a philosophical issue there too. While I'm not currently coming up with any reason to send a flag if you already want to load the GUI; I could be missing something.

Being able to run without wx significantly lowers the bar of entry. But it still requires pyusb and Pillow? I can't think of any way to get around those. Again; with compiled binaries none of that matters on the unixy side (since all that's in the package).

tatarize, 2020-01-17:
I'd expect people in many OSes would drag and drop an SVG onto the icon for the program. That usually runs the program with the filepath. Without any flags. So `meerk40t <filepath>` and the typical behavior there absolutely must be be to load the file out of the gate with the gui.

The autostart thing might also need to be done with or without a gui. Maybe you want to just launch  that but you'd also like to control things, or watch them, etc. I am unsure what --nogui would be goodfor without --autostart. But If there was a second thing you could do, like issue a command as such then the --nogui and --autostart flags would be fine.

MeerK40t --nogui --home

To merely home the device. Or:

MeerK40t --nogui --unlock

To unlock the rail.

--

Also, Pillow is in the requirements but actually is only required if you use an image. If you're not loading an image directly or loading an SVG that references an image. Pillow never needs to get called. 

The SVG code in question does:
```python
    def load(self, directory=None):
        try:
            from PIL import Image
            if self.data is not None:
                self.load_data()
            elif self.url is not None:
                self.load_file(directory)
            self.set_values_by_image()
        except ImportError:
            pass
```

The only way to get around pyusb though would be to not send data to the K40. So:

MeerK40t --nogui --output compiled.egv filetoburn.svg

And assuming there is a writer that handles that, it'd be fine. Or:

MeerK40t --nogui --output mysvg.svg somelightburnfile.lbrn

And none of those svg files would be permitted to have images since no Pillow. -- Obviously no-requirement code would be a lot hard to do anything useful with. But, honestly wx.Python is sometimes larger and even getting the other requirements it might be nice to not require it. Take a Raspberry Pi for example, if you have the SVG to burn and you can install the other requirements quickly are you really going to spend all that time compiling wxPython if you could just run it without the gui and on the file you made and have it run.

joerlane, 2020-01-17:
Good clarifications on the lack of file input switch. That's totally fine of course; and I hadn't considered drag and drop at all. 

You are 100% correct about the wx issue; and the Pi as well. At this point, wx is the most annoying part of this which is what makes the compiled binary so compelling. It's a serious limitations for a lot of potential uses. 

What I was really poking at in terms of asking about pyusb; was a licensing or inclusion issue. Do you know off hand what would be required to include that with MeerK40t? As you mentioned; if you don't need Pillow (not doing raster work? or just images raw?); and can use a straight SVG, then all you need to include is the pyusb code for a command line app to drive a laser out of the box (assuming the OS has Python)? Again; once you start asking for pip installs, some people get queasy. Pretty sure pip is not included in a Mac out of the box; but Python always is. I'll have to double check Pi; but I would guess the same off hand.

Are we close to moving to IRC? I used to get on chat.freenode.net to talk open source back in the day... Just created #meerk40t.

tatarize, 2020-01-17:
Yeah, you can code something that only use pyusb and just sends the data you give it to the laser. There's a few protocol issues it needs to use but that was part of what K40Nano was doing. Giving channels to move data to the laser. It wasn't wildly successful but was a solid proof of concept of some stuff and worked as a great platform to figure out the code the laser used.

I'm thinking for flags:
--nogui / -z
--output / -o
--home / -h
--unlock / -u
--autostart / -a
--path  / -p

So "MeerK40t -zap M50,50h1000" would add a path at 50,50 going horizontal 1000 mils (1 inch) and would autostart that without a gui. That it spells out `zap` is just an added bonus. 

I always have mIRC up from my olden days, I used to write scripts for it, my Trivia bot was pretty popular for a while there. I should autojoin it.

tatarize, 2020-02-05:
This was added in 0.4.0 might need some more stuff but cursory work is there. The GUI was moved over to wxMeerK40t.py so that loading it would not be a requirement of any module. So the -z option is permitted without wxPython installed. In fact, with the windows driver, I should make it default to that if you lack pyusb. And fails out to pyusb with an import error you could actually run MeerK40t with no dependencies at all.

```
python MeerK40t.py -h
usage: MeerK40t.py [-h] [-z] [-a] [-c] [-p PATH] [-i INPUT] [-o OUTPUT] [-v]
                   [-m]

optional arguments:
  -h, --help            show this help message and exit
  -z, --no_gui          run without gui
  -a, --auto            start running laser
  -c, --convert         convert filetypes
  -p PATH, --path PATH  SVG Path command
  -i INPUT, --input INPUT
                        input file name
  -o OUTPUT, --output OUTPUT
                        output file name
  -v, --verbose         display verbose debugging
  -m, --mock            Use mock usb device.

```

Frankly there's little point to convert if there is an output. And I suppose the save command might work out of the gate. Since everything was already designed this way, everything works pretty much from the start.

tatarize, 2020-02-05:
So if you had a little computer that didn't have a screen, `python MeerK40t.py -zai <filename>` should run with no gui, and autostart the file you give it. And you could certainly execute that from far enough to burn down your house. 

You could install MeerK40t ssh into your system running inside the laser box from work over wifi, tell it to -zai the file, and come home to either a perfectly lasered project or the smoking wreck that was your home.

tatarize, 2020-02-05:
Actually already wrote the skip if no pyusb thing. So that's not actually a requirement. I forget what happens if you don't have Pillow installed and try to use images though.

closed 2020-02-09 by tatarize