meerk40t/meerk40t #104: SVG data not loaded correctly

szepnorbee, 2020-01-04:
Hy. I imported an SVG, but the text (with different outline color and, infill) are not loaded into workspace. Marked az NOT IMPLEMENTED.
Original image from Inkscape:
[image]
Imported Image to Meerk40t Workspace:
[image]

Am i doing something wrong?

tatarize, 2020-01-05:
Convert the text to paths in inkscape. So rather than text objects you get paths. I'm working on a conversion thing but for now the text, but for now the conversion would need to be made in inkscape.

szepnorbee, 2020-01-05:
Thanks @tatarize its working now, but with this solution i have a lot of Path element on the list. Unfortunately, this cannot be managed. This is not grouped or visualised on the preview when i select one item. It would be nice to have multiple elements selection and possibility change the settings in one shot for all the selected items. Congratulations! This is going to be very good software!

tatarize, 2020-01-05:
You can actually merge them together in inkscape. There's a method to combine the paths which is what would be needed to do that. If you wanted to change all the settings for them you could just change the settings on the group itself. Such as if you wanted a different speed. It will change the settings on all the elements in the group. 

The hope is that I can deal with the text objects directly. There's a need in lasering to permit text flowing. Like if you wanted to engrave 50 pencils. It would be much preferred if each pencil could get the next name in a list of text. It also might be appropriate to show a preview icon in the tree of the path. 

If you have any comments or issue, feel free to raise them here. Even if they are easy to deal with, it's good to know how people are using it and what they find to be confusing or cumbersome.

closed 2020-01-11 by tatarize