meerk40t/meerk40t #105: Alpha PNG may cause constant burn rather than raster.

tatarize, 2020-01-06:
The transparency loaded with the png images might cause issues that are not accounted for. Needs direct testing. See notes at #71

joerlane, 2020-01-08:
Mentioned in notes at #71; TIFF images with an alpha channel included (embedded in inkscape SVG) also constant burned for me. 

Will try a few other image converters to make sure it's not any other specific file issue. Not sure what's driving the problem; other than a transparent background may "appear" black to the translation.

tatarize, 2020-01-10:
Most transparents are transparent black. Which is a black and thus gets stuck on. Though I haven't actually checked the code itself, but that makes sense.

tatarize, 2020-01-14:
This issue is closable when #102 gets finished and integrated. I created a different image pixel parser for RGBA vs. RGB. https://github.com/meerk40t/meerk40t/pull/102/commits/4a4de182a8c6bbe90d1843394f07cd1fbc01576e

```python
       elif mode == "RGBA":
            def image_filter(pixel):
                return (1.0 - (pixel[0] + pixel[1] + pixel[2]) / 765.0) * pixel[3] / 255.0
```

When it has alpha it multiplies the entire thing by the alpha. So 255 is opaque so the color value is scaled with the sum of the pixels. When alpha is 0. The filter can only return 0 which is a do not burn value.

joerlane, 2020-01-16:
I've tested this; and it appears to be fixed for every file type I could conjure up.

closed 2020-01-19 by tatarize