meerk40t/meerk40t #107: Addon System API

tatarize, 2020-01-12:
From #84 

There are also some functions that aren't trivial and might require some other python dependencies that I wouldn't want to support out of the box. For example potrace ( http://potrace.sourceforge.net/ ) is some very nice software for converting images into rasters. But I dunno if I'd like to add it as a dependency to ally somebody to run that as a function. But, I could certainly see including that as an addon. But, it would need to show up as a function for images but only if the custom code existed.

Or perhaps my future text code sucks and you'd rather do what Whisperer does and simply use inkscape to do that action. You could save a temp svg, then call inkscape with --export-text-to-path add the paths to the scene. I wouldn't necessarily want that function in MeerK40t proper since it depends on having inkscape, but I would want that custom bit of code to be easily added to MeerK40t without too much trouble, or needing to hack it in directly. It would simply need to take 'text' elements and give them a custom operation. There's likely a bunch of custom packages and addons people might like to use, but would not need to clutter up MeerK40t's basecode since maybe that function or whatever is only available on Linux, or is proprietary, or GPL code (which couldn't be directly included into MeerK40t which is MIT code).

Inkscapes system has three main categories of methods:

* Input, providing translation from a file format to SVG
* Output, providing translation from SVG to a format
* Effect, taking in SVG, changing it, and then outputting SVG

Properly any MeerK40t addon system might want to modify:
* Menu function addition.
* Add specific operation for a particular type (image, path, text).
* Permit such operations to happen during Job preprocessing, or in the scene.
* Add a non-specific operation that could be added to the spooler. Beep, Send Home command, call some custom code commands. within the spooler based on addon-code.
* Define a different kind of operation for the laser to perform using the defined middle language.
* Create a threaded function to be repeatedly called while MeerK40t is running or while the laser is being used. Must be permitted to access most of the operating ecosystem, reading and writing signals.
* Add a dialog, to be called similarly to how all the dialogs currently work. Load the window, then call `set_project()`

The biggest most vague line in the sand is figuring out where MeerK40t code should cover and what is to be left to an addon. For example temperature display stuff might be a good part of MeerK40t as well as pausing laser operations, but the actual temperature retrieval code might be too hardware specific or workflow dependent and need to be excluded.

tatarize, 2020-01-15:
A lot of these methods were added though not fully fleshed out in #102. The LaserProject class was morphed into the Kernel class with most of these core functions. It's possible to have some things like Controller working on its own. It runs as a registered thread, setup by the GUI. So manipulations of the thread can be done through the Kernel and not through direct access.

The main way to load and save persistent and non-persistent data from a location is more consistent and will, with a Config object make the data automatically persist, and without that let the data only use the defaults. From this it's not too hard to add entire modules that can contain things like, loaders, savers, effects, control, and easy access to persistent data. Or a command line execution with a kernel not backed by persistent data.

This meets many of the expectations of the discussion in #84. Though, since changing the core framework touched on everything this version is going to need a top to bottom audit before publishing.

joerlane, 2020-01-15:
I spent my day on hardware; and I don't even want to make you guess where the smoke was. I put a volt meter on my power output POT and got a mere 0.10V deviance in quality output with this material. I can also say; that the temperature does indeed shift this rather linearly upwards. Higher water temperature required just a hundredth more in terms of knob power adjustment.

Still have a lot to review; but I will start working in this direction.

joerlane, 2020-01-15:
I pulled this down and played whack a mole for a bit. I couldn't get past an error at line 770 of MeerK40t.py. 

Traceback (most recent call last):
  File "MeerK40t.py", line 1874, in OnInit
    self.MeerK40t = MeerK40t(None, wx.ID_ANY, "")
  File "MeerK40t.py", line 335, in __init__
    self.on_size(None)
  File "MeerK40t.py", line 770, in on_size
    self._Buffer = wx.Bitmap(*self.ClientSize)
wx._core.wxAssertionError: C++ assertion ""w > 0 && h > 0"" failed at /Users/robind/projects/bb2/dist-osx-py37/build/ext/wxWidgets/src/osx/core/core/bitmap.cpp(1008) in Create(): invalid bitmap size

tatarize, 2020-01-15:
Its allocating the bitmap for the double buffering. That code really shouldn't have changed. I could see maybe having an invalid bitmap size though, if it says zero it will be invalid. And since it's saving the window size, you might in theory make the window too small to have any scene, then the scene would want a widthX0 bitmap which will cause an assertion error. Then since nothing changed it would remain broken. I'll doublecheck the invalid bitmap sizes and put in a check. Not sure why ClientSize() would give such a thing. Or maybe it just doesn't on Windows and resizing your MeerK40t height to be very small has always crashed on Linux/OSX.

tatarize, 2020-01-15:
Changing the code to:

```python
        width, height = self.ClientSize
        if width <= 0:
            width = 1
        if height <= 0:
            height = 1
        self._Buffer = wx.Bitmap(width, height)
```

Should make sure the bitmap buffer is always reasonable.

tatarize, 2020-01-15:
It certainly does go negative, but the bitmap itself never tries to build in Windows. ClientSize is spitting out negative numbers, and feeding that into bitmap, which doesn't care in my setup.

joerlane, 2020-01-15:
So I made the change as you noted (which is almost what I had changed it to playing whack-a-mole). The original problem was a hard error at launch. Now I get no window at all; but also no errors. Python application just opens up without anything else. 

I ran in verbose mode if that helps at all.

[MeerK40tErr1.txt](https://github.com/meerk40t/meerk40t/files/4067538/MeerK40tErr1.txt)

tatarize, 2020-01-15:
Hm. And I had just punched in the new commit ( https://github.com/meerk40t/meerk40t/pull/102/commits/65ea6e2006730d00fcea9fa6363521ae440b50e8 ) for that. Maybe it's an issue with the setting returning zero for the size and MeerK40t taking that literally. Which is a bit surprising but reasonable enough. I'll add in a min size for the window and push another commit.

tatarize, 2020-01-15:
Commit 
https://github.com/meerk40t/meerk40t/commit/5308356aadf74d704b5a759bbf3af688f0ea255e should force the minimum window size code shouldn't let it equal zero even if it's somehow getting those numbers out of persistent data.

joerlane, 2020-01-15:
A couple more details. 

It's actually opening the app; and maybe even the main window. I can then open the spooler, navigation, etc from the menu; but I can't find the main window anywhere. It shows up in the Window menu as "MeerK40t 0.2.4; but if I select and minimize it; it doesn't go to the dock (it should animate like the Keymap window does). Maybe the main window is being set to a negative size itself? Just grasping at straws here. It acts as if it does not exist (or it's not worth drawing); but it does exist in so much as it's an object the Window menu knows about...

Hope that helps in some way (or makes sense?).

tatarize, 2020-01-15:
Though it seems like maybe the default values for the persistent data isn't working there for linux. Which means all the settings will default to 0 or '' depending and would not actually default to the defaults the code gives.  Which is certainly a pretty important issue. But, the 
https://github.com/meerk40t/meerk40t/commit/5308356aadf74d704b5a759bbf3af688f0ea255e commit should fix the size and make sure it doesn't try to make that window not exist. Though even then resizing it should work.

I might have to load it up myself to test it. Since I did change some fundamental things for the kernel.

tatarize, 2020-01-15:
Okay, pushed the last known other bug fix.  I'll try this version in Linux and try then get with testing and finding other bugs. Though as far as I can tell (minus unfound bugs, this will kinda be it for 0.2.4. Hopefully squashing the bugs in due course.

joerlane, 2020-01-15:
Fixed the main window. That's pretty tiny; but it does remember window size the second time I open it. It does appear to center the position; which isn't a bad thing. 

Can't open any SVGs or PNGs; I get three errors in sequence but the app stays open (those are attached as it's wordy). 
Clicking on the Project tree generates an error.
Trying to jog around with the arrow keys generates an error, and doesn't jog the red circle position. 

Other panels open; but I can't test anything without being able to load a document.

[MeerK40tErr2.txt](https://github.com/meerk40t/meerk40t/files/4067649/MeerK40tErr2.txt)

[image]

joerlane, 2020-01-15:
When I try to drive around with the arrow keys:
Traceback (most recent call last):
  File "MeerK40t.py", line 1027, in on_key_press
    self.execute_string_action(*args)
  File "MeerK40t.py", line 984, in execute_string_action
    writer.send_job(self.execute_move_action(*args))
  File "MeerK40t.py", line 996, in execute_move_action
    amount = Length(amount).value(ppi=1000.0, relative_length=min(project.size))
AttributeError: 'Kernel' object has no attribute 'size'

When I RIGHT CLICK on the PROJECT (which I can't load any files into) - Left click works.: 
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/wx/core.py", line 3360, in <lambda>
    lambda event: event.callable(*event.args, **event.kw) )
  File "MeerK40t.py", line 821, in update_buffer_ui_thread
    self.on_draw_scene(dc)
  File "MeerK40t.py", line 1229, in on_draw_scene
    self.renderer.render(dc, self.project.draw_mode)
  File "/Users/jlane/Applications/meerk40t/LaserRender.py", line 31, in render
    element.draw(element, dc, draw_mode)
  File "/Users/jlane/Applications/meerk40t/LaserRender.py", line 142, in draw_path
    self.set_brush(gc, node.fill)
  File "/Users/jlane/Applications/meerk40t/LaserRender.py", line 127, in set_brush
    gc.SetBrush(self.brush)
wx._core.wxAssertionError: C++ assertion "retval != __null" failed at /Users/robind/projects/bb2/dist-osx-py37/build/ext/wxWidgets/src/osx/carbon/graphics.cpp(116) in wxMacCreateCGColor():

tatarize, 2020-01-15:
Yeah, if that's the bug you're getting some other code has to change. Because it means all your settings will default to zero just like the window size must have.  So you'll start with automatically lock rail stuck to False. And other issues. It's just hard enforcing that 300 300 is the absolute min permitted to be saved.

The other things seem like they are throwing an error with somehow the brush being set to something wrong.  The other crash should be easy. Kernel doesn't actually have a size property so that's a whack-a-mole error.

joerlane, 2020-01-15:
SVGs and PNGs load, and I can right click project. Still have the keyboard jog issue.

tatarize, 2020-01-15:
https://github.com/meerk40t/meerk40t/commit/1f8fe4bd6860eb961af55cf808b23d76979b5425

Should fix the keyboard jog, that was entirely me. The other one is kinda cryptic mac issue with the code returning null where it shouldn't but the code I have follows the API correctly.

tatarize, 2020-01-15:
Heading out for a couple hours. But, hopefully it'll be together enough for you to test some. There's likely a bunch of other quick bugs like the keyboard thing where I failed to account for some function and the code is still old-form. The documentation on Kernel is fairly well, but some of those functions don't exist yet.

joerlane, 2020-01-15:
Currently clearing a job with 6 SVG paths converted to raster, and a single SVG path cut. No issues with the conversions or queueing in the spooler. If I right click an item in the spooler this error comes up.

Traceback (most recent call last):
  File "/Users/jlane/Applications/meerk40t/JobSpooler.py", line 166, in on_item_rightclick
    element = self.project.queue[index]
AttributeError: 'Kernel' object has no attribute 'queue'

Interestingly; I was able to change the speed mid-job in the Project window; and the spooler fixed the speed of the task automatically. Neat!

tatarize, 2020-01-15:
166 should read:
element = self.project.spooler.queue[index]

tatarize, 2020-01-15:
https://github.com/meerk40t/meerk40t/commit/47a8b6ff323892b056207ea15b535516eb5940ea Corrects it.

joerlane, 2020-01-15:
Fixed. And I was able to edit the queue by changing the properties in the Project pane on the fly... That does pretty much the same thing as the context menu in the spooler. 

I've checked everything I could find; and that I've been using. The files I was running last night work; and having those preview icons is GREAT! Thanks; and enjoy your afternoon.

tatarize, 2020-01-16:
In theory in the future there is intended to be a break between the two, so that when you modify things in the jobs dialog, it's because you need to spool them it shouldn't go back and alter the original content. So if one of the required operations is something like convert this vector to an image so it can be rastered, it shouldn't lose the original image. 

For now it's kind of a low priority though, but it should become an important issue when I get more of the methods defined so people can run things like z-depth raster or a cross hatched raster.  But, even lesser things like converting the transformed image into one with pixels could be damaging if you needed to run that same image several times at different sizes.

I'll test things some more. And I need to do a burn test with some image with transparency. But, generally seems correct. And I'll double check that the method for persistence in Linux isn't defaulting to zero for everything. And how to fix that if that's what's going on.

joerlane, 2020-01-16:
I was surprised that the spooler did update based on the project settings. I might actually use that short term. It seems to update when it advances to the next job in the spooler. 

I set up a test of the alpha channel issue on some junk mail with 13 different files; including a new set exported from Photoshop. Anything that had an option for alpha channel was saved both with and without. Every single file type worked with transparent backgrounds. That included PNG, TIFF, and GIF from memory. 

I got the laser controller to entirely freeze up again. Separately I also ran into trouble trying to set the raster speeds on those images embedded in the SVG for testing. Looks like if I paste in the value of 200 it's not likely to stick (it did set one or two out of more than a dozen). If I type out the numbers: 2, 0, 0 it does work every time. That is a rather strange UI issue in my opinion. 

No matter since I can still fix the speed on the fly! I'm not sure how to recommend that be considered in the future. Maybe make the spooler queue editable instead? I've more than once forgotten to set a speed or power setting (or ran into an issue like this were it didn't take); it's great to be able to fix issues on the fly. Or even to modify as job as you begin to see results. I have in the past loaded a new; second or third SVG file while the spooler was executing a job without issue. Then when it finishes, I delete the first file and can already be prepared load up the next spooler queue.

Overall it's functioning for me given those known caveats.

joerlane, 2020-01-16:
I've gotten the error "processes are still running, issue an emergency stop and close?" when trying to quit. That is after a job has run to completion and the border is no longer red. The spooler is empty. The application hangs at this point. Doesn't happen every time; but it's a pretty hard crash.

The keyboard doesn't move the head; but the arrows buttons in navigation pane do.

Raster job stutters on greyish area due to buffer underflow?

Not sure who's bug this is; but an image loaded directly on the upper left hand corner of the stage will trigger my end stop in the X axis. It appears to keep retrying the same line and bounces off the end stop indefinitely (or the UI doesn't update right). I got annoyed and stopped it each time long before it finished. Is this a sign that maybe Shift Home Position is coming? All of my SVG file have a reference alignment that I carry over; such that I don't place objects out of bounds of my physical bed. 

That actually might be a bug though; is it trying to overscan past the region it can physically move? I don't think it started out hitting the end stop, it was a few lines down from the beginning of the raster job...

Still nothing I can't work around at the moment; and I've never tried to just load an image directly with upper left registration in the past. This is the file I used for reference (not important to engrave; just a random technical selection); it may well start banging your machine too after a dozen or two lines. I was running 70mm/s default. It's also the file that buffer under-runs mid-way if you move it out into the bed further.

That's what I've got for the time being. I've packaged it up; and been using it on my own. I'll do my best to take notes and isolate some of the more unusual glitches. 

[image]

joerlane, 2020-01-16:
Aaaand; one more thing. When I imported the file with all those images embedded in an SVG for testing; those with a black bitmap and randomly transparent backgrounds... I got all black on black in the tree. The icons showed up, but I could only read the text if I clicked on the element. Might pay to think about how to at least prevent that explicit problem (black text, in a black fill, with a black border). I think this is a stroke vs. fill default conundrum when considering embedded bitmaps, in an SVG, inside the tree structure display of MeerK40t. 

Is it also compounded by OS X's dark mode? That's also half broken for me; and I need to investigate further.

tatarize, 2020-01-16:
Yeah, I have seen that. Same color fill and stroke and it suddenly becomes invisible and a bit weird. It's certainly something worth considering.

tatarize, 2020-01-16:
Just finished testing against the machine. It does seem to get a bogged down bug, which I fixed with a new messaging system that broken some additional things. But, it ties things together a bit better. And seriously reduces the CPU usage. Like it takes ~9% CPU underload. Though I switched to the other method for Linux so it does make checking that stuff in Linux another large priority.

tatarize, 2020-01-16:
> I was surprised that the spooler did update based on the project settings. I might actually use that short term. It seems to update when it advances to the next job in the spooler.

There's a couple things that might be overly shocking. In Alignment Ally, the confirmation tests has a moveable selection for the power level. It ties, directly into the writer. And since the writer is writing on the fly, if you go in there and change that value, you can adjust the PPI power modulation while it's running. At least until the next spooled command forces a different power level.  Sort of needs some kind of direct modifying bit for the spooler, as it's running. It's entirely possible to tell it, while it's running to change the speed or power, to exit out of the current mode, and do some other things. There's a few easily possible function that might be pretty nicely helpful. It's built like a proper system so everything can modify anything on the fly.

> Looks like if I paste in the value of 200 it's not likely to stick (it did set one or two out of more than a dozen). If I type out the numbers: 2, 0, 0 it does work every time. That is a rather strange UI issue in my opinion.

That sounds like a weird encoding issue to me, like the pasted value is like <arial-font "200"> or however it wants to add the other info about that 200, rather than do it in plain text. You could zip up the offending file and post it to the thread. Seems like it's be easy to test that.

> I've more than once forgotten to set a speed or power setting (or ran into an issue like this were it didn't take);

Yeah, I ran into that issue too. Edited the settings for the group and it didn't save it in the group or trickle down and it's supposed to do both. Gotta remember to fix that one.

> Or even to modify as job as you begin to see results. I have in the past loaded a new; second or third SVG file while the spooler was executing a job without issue. 

Yeah, this is another reason why it sort of needs to break the connection between the spooling job and the elements in the scene. You could very well want to run the same image at with completely different settings, so you add the tweak settings, add the job, tweak settings, add the job, tweak settings, etc. But, the writer will only lock in the speed once it's running that particular element and is passed the speed command. And even then it's entirely possible to tell the writer to change the speed, while it's running. There's just no code that does that. Maybe something that would let you tweak the stuff going on right now in the writer. So you could adjust the speed or power, or otherwise see the current speed or power.

joerlane, 2020-01-16:
That's an interesting point in particular. I had switched over to Repetier-Host to control my 3D printer specifically for the pseudo-realtime control of speed and plastic flow rate. There were so many times I needed; or simply wished to nudge one setting up or down a hair... Sure; it won't take effect until the buffer strained through; but it really didn't matter with even reasonably large buffers in that situation. This laser buffer drains plenty fast for that kind of effect.

tatarize, 2020-01-16:
> I've gotten the error "processes are still running, issue an emergency stop and close?" when trying to quit. That is after a job has run to completion and the border is no longer red. The spooler is empty. The application hangs at this point. Doesn't happen every time; but it's a pretty hard crash.

The trigger for that is that the spooler thread state is started or the controller is started. Something didn't die properly.  One of the things I did in the last commit is make these things a bit more overt with forcing them to die and registering them as threads as such. It might need some tweaking. Currently the task bar should say whether your controller is Finished or your writer is Finished. One of those seems like it might not be the case.

>>The keyboard doesn't move the head; but the arrows buttons in navigation pane do.

I just used it with PC version, and the keyboard totally moved the head. Also did the press the Q button that sets location Q. Then when you press Q it goes back to that location automatically. Sort of needs the keymap fleshed out, there's some cool stuff in that. And it would be a breeze to save that stuff with the easier registration of persistent data. Also made the buffer_max in spooler properly save.

> Raster job stutters on greyish area due to buffer underflow?

Yes. The command power density drops too low. Basically the commands it's being given do not take long enough and it finishes all of them before the next packets get loaded so when the commands are like 1 pixel long, they go URaDRaURa over and over again, which gives us 9 characters to do 3 mils of commands. And when you're going really fast it's pretty easy to bottom that out.

>> That actually might be a bug though; is it trying to overscan past the region it can physically move?

It does and is. If you start exactly in the corner like that, it will hit the edge with the overscan. Since I think it default gives you overscan of like 20. It will actually start striking the side over and over again. I guess I could set overscan to 0 by default. Rather than break things with unstated features. That and let people actually modify the overscan outside of setting it in an svg. overscan="200" etc.

> That's what I've got for the time being. I've packaged it up; and been using it on my own. I'll do my best to take notes and isolate some of the more unusual glitches.

Thanks for that. You've been great with the feedback, bouncing things off somebody makes me less likely to spend a month writing a fantastic but utterly secondary SVG library. And the alternative OS testing was pretty sorely lacking. Now it's turning out pretty nicely, I really like the way the kernel setup is working. Should let you check temperature or whatever you want to hack into the code pretty easily, but it's also correcting a number of oversights I made for how the stuff should work.

It's a bit more broken than it was a bit ago. And I have a pressing number of things that are there but just turned off. And some of the stuff I add like fill and stroke colors have strange entirely foreseeable errors like becoming a solid bar, but it's really coming together quite nicely.

joerlane, 2020-01-16:
I can do linux testing anytime (os x is reasonably close). I have a few options; and don't mind loading another from scratch if you are missing something worthwhile.

joerlane, 2020-01-16:
I'll run it through a few boards when I get them hooked up. What linux distros are even being tested so far?

joerlane, 2020-01-16:
Might need to backup and remove settings too; in reference to the wx window issues.

joerlane, 2020-01-18:
Getting back to the API. While working through the design elements of a rotary adapter for the K40; I started questioning some simple things.

You recently mentioned the F8 Path Entry dialog. Does it make sense to allow that; or a similar tool to accept scripted commands? Do all API inputs talk to the controller, as opposed to the spooler; or is that a separate API Add-on scheduler? A pane to implement very simply logic on input would be ideal. Example: [IF, Sched, Always, etc.] variableTemperature [<,>,=] variableInput DO *(spoolable | immediate/urgent)* output action [switch GPIO (on | off), toggle GPIO, Emergency stop, pause, etc.]. This is something I can quickly draw up in wxGlade once we hash out details.

What about adding a right click option to the spooler window to add basic commands. This may require other developments first; like drag and drop re-ordering. Spooler queue additions like a standard beep, home, go to preset 1-5 numbered location, pause, unlock rail, and perform a user API action. In this case; you probably don't need a ton of these user actions; but more than one can't hurt. Some obvious use cases for user actions are: using GPIO, relays to switch power to air assist, water pump, interior lighting, aiming laser; also more complicated things like PWM to drive the hardware laser power.

Then there are input questions. So the software is polling some input hardware; like a temperature sensor, camera?, measured laser current, measured laser power setting (pwm?) voltage (as feedback), end stops, position sensors; or worse something demanding like, a rotary encoder that pulses upon movement... Position feedback is common to monitor stepper motors for absolute location. Water flow sensors can also output a pulse train. "Who" does this; and what provisions are there for priority or frequency? I also noticed some hard coded delay times while debugging; would it be a good idea to add a "Debug" panel somewhere to facilitate making adjustments?

I recognize some of these are outside the scope of the MeerK40t API. But I thought it would make sense to have a "discussion" about all the possibilities (and what would make an "IDEAL K40 rig"?); then begin the process of pruning, and dividing duties/responsibilities for the user and MeerK40t. 

From a physical standpoint; the GPIO header on a Pi has enough pins to drive all of this. In reality; there is a fair amount of additional hardware involved. Things like a ADC/DAC combo chip, relays, sensors, and so on. On Pi hardware; there would certainly be potential issues with scheduling, so it may not be possible to achieve some things without a co-processor. The simplest solution would probably be a single cheap micro connected to the Pi's hardware serial port. I would certainly be very wary of trying to attach the laser power hardware input (on the power supply) to a Raspberry Pi hardware PWM output pin...

I don't bring this up because I need any of these additions anytime soon. I simply believe it's wise to plan, and help avoid misguided efforts one way or another. At this point I think I can use this current version of code for all my upcoming tasks this month and beyond. So take your time, think about the implications; and again, thank you for all the effort you've put in to this. I'm sure you are aware; MeerK40t already goes above and beyond anything else available for the K40.

tatarize, 2020-01-19:
> You recently mentioned the F8 Path Entry dialog. Does it make sense to allow that; or a similar tool to accept scripted commands?

The way keymap half works currently is that they are parsed like a little script command. I had at some point intended to add scripting, but I figured python would be the best language for that and you'd already have that. But, keymap is doing some basic command parsing like adding 'window Preferences' as something that is keymapped to F4 or something would let F4 load up the preferences window. 

The main point here is that if addons don't really know what's in the system and don't assume they do know what the system, they just interact with the capabilities registered by the Kernel. They can send signals to each other, and listen for signals to react to events. And they can write variables on the kernel but they won't know if anything else is reading them or changing them but they are just a stored memory location and should be really quick.

> Do all API inputs talk to the controller, as opposed to the spooler; or is that a separate API Add-on scheduler?

All the addons could talk to the controller but they wouldn't have anything to say. The controller is just sending Lhymicro-gl code to the board through USB. If you want to do that, you can totally do that. But, you're much better having those commands plotted out by the spooler that knows how to convert more useful commands into Lhymicro-gl.

The scheduler is going to be in the kernel and always running until shutdown. So if you needed a function to be called every 2 seconds or so. You'd be able to put that job in the scheduler and it would execute your code. 

> A pane to implement very simply logic on input would be ideal. Example: [IF, Sched, Always, etc.] variableTemperature [<,>,=] variableInput DO (spoolable | immediate/urgent) output action [switch [...]

```python

    def temperature(self):
        kernel = self  # or self.project or whatever, everything has a kernel reference
        kernel.setting(bool, 'halted', False)
        kernel.setting(float, 'temperature', None)
        kernel.setting(float, 'max_temperature', 25.0)
        def check_temperature():
            if kernel.temperature is not None:
                if not kernel.halted and kernel.temperature > kernel.max_temperature:
                    kernel.execute("K40-Pause")
                    kernel.execute("My Other Operation or whatever.")
                    kernel.halted = True
                elif kernel.halted and kernel.temperature < kernel.max_temperature:
                    kernel.execute("K40-Resume")
                    kernel.halted = False
                
        kernel.cron.add_job(check_temperature, args=(), interval=5)
```

The idea behind this is the addon just registers some settings which it then can use. If sharing settings between different addons and it doesn't know the addon is there it still needs to initialize them. You can then just use them. They will either have persistent values or have values other addons have been messing with. That bit of code doesn't actually know the variable `temperature` exists. It registers it with a value of None. If it exists or existed and was saved. Then it has a temperature. And uses that but, it's completely agnostic as to where that value came from. That code, there (though I didn't test it), and K40-Pause and K40-Resume aren't operations I had the controller register (though I easily could). 

This would also be agnostic about the `max_temperature`.  That would need to have been set somewhere by something, but that's more a gui problem, etc.

> What about adding a right click option to the spooler window to add basic commands.

That's perfectly doable. The right click to the control window is just giving you a list of registered operations. Any addon could register one to perform some arbitrary code. 

> This may require other developments first; like drag and drop re-ordering. Spooler queue additions like a standard beep, home, go to preset 1-5 numbered location, pause, unlock rail, and perform a user API action.

Some of those are in Jobs and more apt for that. The problem is the spooler is running and gobbling up the events and the topmost event is often being used so it's not something you could change easily. You could delete the rest of the stuff and it would work fine. Or drop other spooled items. But, making stuff should be a JobInfo problem or a different dialog since anything could feed stuff into the spooler. Operations like you'd find in a print spooler are all perfectly reasonable.

> In this case; you probably don't need a ton of these user actions; but more than one can't hurt. Some obvious use cases for user actions are: using GPIO, relays to switch power to air assist, water [...]

The spooler will execute the code you give it. If you need a function spooled that can be done. And should be done in a blocking fashion. So if you want to spool something to change the height of your bed, that'd be doable. (well it wouldn't but I just added that for `yield COMMAND_FUNCTION function`.)

> Then there are input questions. So the software is polling some input hardware; like a temperature sensor, camera?, ... a pulse train. "Who" does this; and what provisions are there for priority or frequency? 

[... cut ...]  doesn't add the MeerK40tGui module or ask the main window to start. 

And I'm planning on moving the main spooler thread to the kernel. This way you could add spooler stuff in a much more agnostic fashion. And the LhymicroWriter.py would just be the registered command interpreter.

However big changes make fast versions harder, and I've pushed a lot of basic bells and whistles to the side for too long. So I'll hold off on those design shifts since they are more likely to break things than most.

tatarize, 2020-01-19:
https://github.com/meerk40t/meerk40t/wiki/Kernel-API-System

Is the current workup. There might be clear need to add specific function things that apply to other specific parts. Or some of the stuff in there might need to just go away. But, it currently would allow people to add a data saver or data loader, some control functions, signals, scheduler, and settings that get shared and preserved.

tatarize, 2020-02-06:
This was the key principle change in 0.3.0

closed 2020-02-06 by tatarize