meerk40t/meerk40t #109: Project save ability, duplicate tree item: enhancement requests

joerlane, 2020-01-16:
There are times when I want to duplicate an object in the tree; or one that is displayed on the stage. 

Also noticed there is a project import; but still no project Save/Export. There are situations where a base project that is MeerK40t specific, can trickle down into numerous tests without ever loading as second SVG. Saving a complete project would allow that same effort to continue across application quit sessions. Seems like a new button between Open and Laser might be needed; but a menu option is totally fine, or a right click context menu option for the Project is even less effort I'm sure.

Is that a standard EGV? The contents of the spooler's output over time? Does that already kind of work when you mock USB?

tatarize, 2020-01-16:
EGV is already compiled and basically converted into K40 code for the controller to use. It would make impossible things like changing speeds or editing various shapes etc, or changing the content of text. It isn't too hard to get a bunch of that data feed it straight into the controller, and it would certainly work. My EGV data is basically the same. Except I added two characters as controller commands. Namely `-` and '\n' which mean perform a finish and wait sequence (-) and (pad whatever packet this command is in). There's some clear reasons for needing this sort of thing with how the controller works. But, yeah, the EGV could be tossed into a file and throw into LaserDrw.

But, what it kind of needs is something like saving a specialty SVG, that contains all the data and settings. And wouldn't be too hard to do. I do know SVG quite well and some of the internal stuff certainly already spits out SVG path data etc. 

Lightburn's methodologies here would be more apropos. Basically saves as an xml with the application version, material height. Then saves the individual cut settings. Then saves various shapes with given types.  Path, Rect, Text, Bitmap, couple more. These all have an <XForm> property which is the affine transformation.  And these shapes always reference a CutIndex, which has all the settings. name, minPower, minPower2, maxPower2, speed, kerf, zOffset, enableLaser1, enableLaser2, startDelay, endDelay, throughPower, throughPower2, enableCutThrough, priorty, frequencie, overrideFrequence, PPI, enablePPI, doOutput, hide, runBlower, blowerSpeedOverride, blowerSpeedPercent, numPasses, zPerPass, perforate, perfLen, perfSkip, dotMode, dotTime, dotSpacing, scanOpt, bidir, crossHatch, overscan, overscanPercent, floodFille, interval, angle.

As nice as that stuff is, literally all of it has direct parallels in the SVG spec. I'm pretty sure, it might be easier to save everything in just an SVG.  And it would already be set to load for MeerK40t since it loads those settings from the SVG already.

The only possible exception would be the Cut settings, in lightburn those aren't directly attached to an object. They could be worked into a <foreignObject> and simply assigned by `id`. I'm sure there's some reason they did it like that. Apparently letting you reuse the same cut for many different objects or have cuts that aren't applied to any object. And loading .lbrn files is an eventual goal kinda worth having.

Not only that if projects were set entirely as SVGs with some caveats. You could make an inkscape addon that would simply let you directly set the speed and power of various cuts etc. Rather than doing the whole this color line = these settings things. Not that there's much wrong with that set of methods. Though actually with MeerK40t already loading that stuff, doing that is *already* possible. You'd just need to make an addon that lets you set a special attribute of `speed` for a particular svg object.

--- 

So yeah, projects would probably just be losslessly set svgs that have all the settings.  The nodes themselves already delegate to svgelements code nodes so the objects are get to use that well developed code there. It wouldn't be too hard to write. If you wanted I could even move it up the list to be the thing I mess with after 0.2.4 is finally finished up. Writing an SVG is much easier than reading them, and I already made code to read them very robustly, and store the stuff it loads as those kinds of objects. I don't see any issue with any of that. I could even write the images without any trouble.

joerlane, 2020-01-16:
Thanks for bringing up the Lightburn aspect here. I haven't had exposure to commercial software for a number of years. I started out with the old Corel Draw, and USB key way back in the day; many years ago now at a local Makerspace. Then they upgraded to a larger machine with more useable software, and laser power control. It still took a lot of file prep to bend it to my will at that time. I'll have to investigate available features again, and see what's out there; I'm a bit out of touch these days.

The idea of a modified SVG is sound in my mind. I don't know much about extending inkscape in that way; but it doesn't seem all that difficult. As you noted; there really aren't that many additional parameters. I don't mind hand edited files in the short term if there are things worth testing.

Does cut order come into the thought process at all here?

I don't know if this method supersedes color designated settings; but it does seem like a duplicated effort in some ways. The problem I see is that you wouldn't want to assign new colors to SVG objects when setting up a Project in MeerK40t; but adding the tags internally to the objects upon configuration would make sense. Then the saved project would just be that resulting SVG, with embedded configuration elements for the laser?

tatarize, 2020-01-17:
The SVG would be saved in the expected cut order. If a particular item is flagged as a cut operation that modification would be run in the Jobs panel. Basically it needs to be optimized so any operation flagged as a cut, is put last and is assembled inside to outside. Other than that the order would be saved the same way they are now, in the first come first serve manner. I'm pretty sure the projects would save 1:1 without losing any set data.

tatarize, 2020-01-19:
#118 will be addressing these requests. The basic work is thus far done.

Still needs to save out more information to be properly lossless and convert the images into png and then into base64 dataurl. 

Better Save Icon.

Use working directory so, save actually does something other than delegate always to `save as`.

But, converted the save and load stuff to use the kernel so any modifications from custom modules would be reflected in the GUI.

tatarize, 2020-01-19:
I do however love the fact that the save work area works perfectly. So you can just save a blank project and it'll give you an SVG that's sized to your device bed.

tatarize, 2020-01-21:
This was completed in 0.3.2. The save format was made SVG. And a namespace of ns:meerk40t was added to tell them from other formats. 

https://github.com/meerk40t/meerk40t/wiki/Namespace It already saved and loaded set properties from SVG files. So that was also completed.

Duplicating an item requires you set the number of passes to a value above 1 then call the "Split Passes" function.

closed 2020-01-21 by tatarize