meerk40t/meerk40t #112: Mouse navigation not working on OS X in current branch pull

joerlane, 2020-01-17:
This worked in 0.2.3; and early 0.2.4 pulls. It appears to be a focus issue from what I can tell. I can navigate around the tree with arrow keys if focused over there; but can not move the laser head/cursor with the arrows at any point. 

This does continue to work on the Pi/Linux; even with the latest pull.

tatarize, 2020-01-17:
To get the focus to work correctly it switches focus when you mouse over a region.  The arrow keys should have fixed a bit to actually move 1mm rather than 9.6 or whatever it was. If you work with the latest pull see if any hotkeys work. I added events for F6, F7, and F8. And save points for keys 1, 2, 3, 4, 5.

tatarize, 2020-01-17:
So for F8 you should be able to hit F8 paste in "m 867.403576567,83.3231257031 c -89.7778838604,-54.5106716602 -212.73836843,-83.3231257031 -355.592276567,-83.3231257031 s -265.814392706,28.812454043 -355.592276567,83.3231257031 c -102.198518353,62.0540443359 -156.219023433,154.348477787 -156.219023433,266.908516424 c 0,109.992523717 69.1583016849,308.634670846 172.090097848,494.296219184 c 52.0105170377,93.8077216699 107.241286465,173.527560486 159.724043904,230.534234971 c 64.7776193629,70.3632884883 125.338298051,106.040903719 179.997158247,106.040903719 s 115.219538884,-35.6776152305 179.997158247,-106.040903719 c 52.4827574397,-57.0066744844 107.715899934,-136.726513301 159.724043904,-230.534234971 c 102.931796163,-185.661548338 172.090097848,-384.306002309 172.090097848,-494.296219184 c 0,-112.560038637 -54.0205050803,-204.852165246 -156.219023433,-266.908516424 z" and draw a guitar pick.

joerlane, 2020-01-18:
Same with the very latest pull. The arrow keys work normally in Debian on a Pi (yes; I noticed they don't move nearly as far anymore). The number keys don't seem to do anything in either place, and the F-keys work on both Debian and OS X.

I checked the keymap settings on the Mac, and they look right; but still no functionality. Also not editable; but I'm guessing you already knew that.

tatarize, 2020-01-18:
The number keys don't seem to do anything, but they are actually saving the current location of the device head. So if you hit that number a second time, it will return that that same spot. So you can move, hit 1 then move some more and hit 2 then when you hit 1 again, it'll move the laser head to that location.

joerlane, 2020-01-18:
Keyboard is still not working at all in Mac OS X. Numbers are still not working at all in Mac OS X. F-Keys work.

Debian Linux on Pi does work with arrow key controls. The numbers kind of work; but not as they should. Messing with it too much breaks both keyboard and number control; but the Navigation panel still does work. I'm able to focus on the tree and keyboard select up and down; but back in the scene it's still dead as a doornail. Kind of like the way the Mac acts all the time.

joerlane, 2020-01-18:
Keyboard events appear to end up in the spooler...

[image]

Then I added a job to the spooler and they define themselves with a description...

[image]

tatarize, 2020-01-18:
Some keyboard events end up in the spooler. Others will happen right away etc. It depends on whether it needs to control the laser and the laser is busy or not.

joerlane, 2020-01-18:
There is a pretty strong difference between what happens in OS X here; and what happens in Debian. I'm holding on to a Mac OS binary that was only a few commits into the 0.2.4 version; and that does exactly what is expected on the Mac which is how Linux acts.

In the Linux app; I can open MeerK40t and immediately begin arrowing around. If there is no laser cutter connected; it just moves the read circle and begins to fill the buffer. When the buffer fills up; it begins to queue things in the spooler. If the laser cutter is connected to a USB port; it connects and begins driving the laser head around immediately; and also keeps track with the red circle as expected.

If I perform the same actions on the same code running on OS X; I get different results in a few ways. When I open the app and begin arrowing around; all commands end up in the spooler. It doesn't matter if the laser USB cables is connected; the controller is connected in software, whatever. The red circle doesn't move; and the buffer doesn't fill up. They simply queue up and have to be released en masse. The same applies to number commands. Those will also happily queue up in the spooler; but aren't filling the buffer.

Clearly, it's not actually a focus issue since the keys are recognized and issue command. But I can't at all explain why Linux is working as expected; and the same code is so much different on Mac OS X. I don't know what distinction there would be between sending the commands (and even automatically connecting to the laser); and spooling them in the queue. Does the red dot not moving help sort this out?

tatarize, 2020-01-18:
The thing you're describing is the spooler failing to start processing things. The stuff ends up in the buffer because they were spooled. The stuff always go to the spooler and anyway, it's how it knows what the code to write is, the difference here is that without the spooler running it will just stay there. In theory it should autostart. And, in fact, there is a setting for autostart. Which must actually be off and not exposed anywhere. I could just get rid of the setting and have it always autostart. The idea was there would be sometimes you might not want the spooler to just start buffering stuff. But, I think it was based on older concepts where there wasn't a jobs dialog to preprocess stuff before it was sent to the spooler. I'll get rid of the setting.

https://github.com/meerk40t/meerk40t/pull/113 

Should have it forced to True. Might need to just be removed if I can't think of some reason to have it. Pretty certain the dialog for the spooler also wouldn't need the giant green start button (if you push that it'll start running but die once everything is spooled and not autostart again).

tatarize, 2020-01-18:
So this was caused by a secret, undocumented feature, that breaks things and has no method to stop breaking things exposed to the user.

joerlane, 2020-01-18:
This appears to be resolved in 0.3.1. What kind of secret feature(s)?! 

I woke up dreaming of git bisect; so you saved me a bit of time there... I believe this can be closed when you see fit.

joerlane, 2020-01-18:
In a semi-unrelated (what do I know, it might be related) issue; crash on quit is back for the Mac. I haven't figured out exactly which actions cause it though. Straight open -> quit seems to work. But just jogging around and playing with the numbers can cause the hang crash; or sometimes I simply get a Processes are still running warning, and then a clean exit after.

tatarize, 2020-01-18:
Yeah, I think it's something to do with the threads loading up when you invoke the spooler or the controller. They should be dead by the time the quit goes through but, it might still be sending some data in memory that is being deallocated and thus stops having access and quits. But, it seems to only happen when you try to quit, and it's python itself crashing. And gives no data of use thus far.

tatarize, 2020-01-18:
There was a flag to turn off the autostart of the spooler queue. This was inteded to be so you could preprocess the data. But, the Jobs dialog with preprocessing fixed that. But, it got left in. Somehow you managed to get False for that value, which means it doesn't spool automatically and since I removed all switches to flip, it just stays there stuck off and causes that effect. Other than the menu inside Controller there shouldn't be any secret features. Though it's pretty common to code them up, because sometimes code is good enough to ask people about but not good enough to show everybody.

joerlane, 2020-01-18:
I fooled around more in PyCharm; and now I'm starting to see some of the fun toys. I can inspect variables with a quick mouse over. Haven't tried breakpoints yet because I'm not sure where to cast my net (still a little green on this whole package). 

This is the best I could gather from the hung state after jogging the head around for a moment.

[image]
[image]

tatarize, 2020-01-18:
State 1 there is started it means nothing changed anything.  and it's running around in a circle checking if the job are scheduled and if they are run them. It's just the scheduler thread. Find jobs that should be run since their set times are beyond the current time and execute them. 

Currently the only thing that uses that thread is the signals job that sends out the signals from the various modules to anybody listening to those signals. But, in theory, you could add a job for check my temperature code, and when it's above some value, execute a pause on the controller. So other custom modules won't need their own threads.

joerlane, 2020-01-22:
Controlling the navigation has been working reliably for the past few days of code revisions. Fixed.

closed 2020-01-22 by joerlane