meerk40t/meerk40t #119: Path and Image objects move upon Scale or Rotation changes

joerlane, 2020-01-19:
Not sure if there is an anchor point in the SVG file that's being referenced for these translations. Both scale and rotate can send things outside of the cutting area; and none of the operations left the objects in even remotely the same location. 

[image]

Not sure what it does; but I tried Reify User Data after and got two errors, then the tree disappears:
Traceback (most recent call last):
  File "wx/core.py", line 3360, in <lambda>
  File "Meerrrr/MeerK40t.py", line 448, in on_elements_update
  File "Meerrrr/MeerK40t.py", line 600, in refresh_tree_elements
  File "Meerrrr/MeerK40t.py", line 583, in add_element
  File "Meerrrr/LaserRender.py", line 55, in make_raster
  File "Meerrrr/LaserNode.py", line 456, in flat_elements
  File "Meerrrr/LaserNode.py", line 452, in flat_elements
TypeError: 'NoneType' object cannot be interpreted as an integer

Traceback (most recent call last):
  File "wx/core.py", line 3360, in <lambda>
  File "Meerrrr/MeerK40t.py", line 780, in update_buffer_ui_thread
  File "Meerrrr/MeerK40t.py", line 1195, in on_draw_scene
  File "Meerrrr/LaserRender.py", line 29, in render
  File "Meerrrr/LaserNode.py", line 456, in flat_elements
  File "Meerrrr/LaserNode.py", line 456, in flat_elements
  File "Meerrrr/LaserNode.py", line 452, in flat_elements
TypeError: 'NoneType' object cannot be interpreted as an integer

After that; the SVG disappears from the tree, and clicking on the location generates this error:
Traceback (most recent call last):
  File "wx/core.py", line 3360, in <lambda>
  File "Meerrrr/MeerK40t.py", line 780, in update_buffer_ui_thread
  File "Meerrrr/MeerK40t.py", line 1195, in on_draw_scene
  File "Meerrrr/LaserRender.py", line 29, in render
  File "Meerrrr/LaserNode.py", line 456, in flat_elements
  File "Meerrrr/LaserNode.py", line 456, in flat_elements
  File "Meerrrr/LaserNode.py", line 452, in flat_elements
TypeError: 'NoneType' object cannot be interpreted as an integer

tatarize, 2020-01-20:
Yeah, scale's screwy, I'm going to have to work on it. It's to do with the bounds for some objects that can't really produce bounds.

tatarize, 2020-02-19:
This particular problem is fixed, mostly because the offending code was scrapped.

closed 2020-02-19 by tatarize