meerk40t/meerk40t #123: CH341 EEPROM Dumps

joerlane, 2020-01-20:
[ch341.zip](https://github.com/meerk40t/meerk40t/files/4088488/ch341.zip)

tatarize, 2020-01-21:
Yeah, solid try, but I guess the chip doesn't have anything of use in there.
https://github.com/meerk40t/meerk40t/wiki/K40's-CH431A-EEPROM-dump

closed 2020-01-21 by tatarize

tatarize, 2020-01-21:
[dump.just_binary_fake.zip](https://github.com/meerk40t/meerk40t/files/4090493/dump.just_binary_fake.zip)

joerlane, 2020-01-21:
[image]

joerlane, 2020-01-22:
[image]

joerlane, 2020-01-22:
[CH341DLL.H.zip](https://github.com/meerk40t/meerk40t/files/4095410/CH341DLL.H.zip)