meerk40t/meerk40t #126: Deceleration/acceleration functionality.

tatarize, 2020-01-23:
There's some chance that the speedcode gearing value 1-4 is an accel/decel amount. If this is a thing the M2 K40 max effective speed could be upped by compact mode dropping at sharp angles. The accel/decel twitches occur when entering and leaving compact mode during speed changes. If we did not change angles on the fly, but rather broke the compact mode and then reentered drawing the new angle, the machine could perform at speeds above 35mm/s without sounding like it's going to break.

Even if the decel and accel twitches are not due to the speed code 'gear' value, they do happen only at compact mode breaking and testing code with breaking at hard angles would be a worthwhile path of research.

tatarize, 2020-01-23:
@joerlane's hypothesis turns out ot be correct. Those numbers *are* acceleration and deceleration. Apply some breakpoints at inflection points does trigger these well and resulted in better movements at speed. Many questions are unanswered though.

* You can trigger the decel with a laser turn off. What specifically is required to do that and can it just be smartly forced to happen?
* The decel goes in 1 of the 8 directiosn. What exactly ensure it goes in way rather than another? Is it last direction seen?
* There is a certain amount of acceleration, how much is there and how quickly does it get to full speed? 
* Is there a better algorithm to avoid hits that is as nice as the naive version?

tatarize, 2020-01-23:
The Marlin 3d Planner calculates the positions of the proper speed curve inflections and determines if that change in acceleration exceeds a maximum, and slows down the device for those curves which will exceed that amount. This problem turns out to be similar to how Android implements fling and a number of other solutions. Marlin's uses the Newton-Raphson to find the changes in the acceleration. We can see the naive algorithm will have inflections in position which are not actually major speed changes and false negatives where major speed inflections occur but are not broken.

[image]

In this case the red curve suffers no inflection under the naive algorithm and the black curve suffers one. Because it changes two directions in three points, though I attached it in the grouped section so it's looking for inflections over blocks of orthogonals and diagonal changes. So going NE to SE counts. But going NE to E to SE does not, regardless whether the speed changes in the latter are much more serious or the speed change in the former was very minor and would be well within the slack of the belt to move up by 1 then back down by one without issue.

joerlane, 2021-06-18:
Acceleration tick counts in cut/engrave: Acceleration 1-2=512, 3=1024, 4=1536. Four ticks will move 1 mil distance. But acceleration and deceleration (braking) happen in the same distance, so Accel 1-2 would be 256 mil total speed up and slow down distance. 

Acceleration tick counts in raster: Acceleration 1 Step=512, 2 Step=2048, 3 Step=3080, 4 Step=3584 ... d=0.896"

380µs is the default speed?

11µs=625mm/s, 12µs=600mm/s, 14µs=500mm/s, 18µs=400mm/s,  24µs=300mm/s,  35µs=200mm/s, 70µs=100mm/s

tatarize, 2021-06-18:
egv IBzzzzRzzzzS1P$  Period 91-92µs, Frequency 10.87-10.99kHz (for each stepper) since it’s diagonal. 

```
691µs/1.45kHz=10mm/s
345µs/2.90kHz=20mm/s
230µs/4.35kHz=30mm/s
173µs/5.78kHz=40mm/s
138µs/7.25kHz=50mm/s
115µs/8.70kHz=60mm/s
99µs/10.10kHz=70mm/s
86µs/11.63kHz=80mm/s
77µs/12.99kHz=90mm/s
69µs/14.49kHz=100mm/s
63µs/15.87kHz=110mm/s
57µs/17.54kHz=120mm/s
53µs/18.87kHz=130mm/s
50µs/20kHz=140mm/s
46µs/21.74kHz=150mm/s
```

tatarize, 2021-06-18:
45.6µs 21.93kHz for the rapid. Which charts to 150 but the chart is in unfixed speeds which are off by 8% or so and should be multiplied by 0.919 to get the real speed. This means our 150 is `150 * 0.919 = 137.85` which is the same as the rapid speed going by the frequency of the ticks. 4 ticks per rotation.

Sophist-UK, 2022-02-20:
I would like to propose a further enhancement of this concept...

We know that the X-steppers are capable of moving at up to 550mm/s. (Tests I undertook a day or two ago suggest that above this the steppers go completely screwy.) I have not tested raster speeds in the Y-direction, however they are likely to be several hundred mm/s. Yet, the K40 can vector at only c. 35 mm/s.

As usual, I am unable to follow the above discussion, but it appears to me to be about how the m2 nano changes direction internally and whether use of different codes could improve the way it does vectors (like the recently improved twitch code does vectors).

I would like to take an alternative perspective on this and suggest some ideas to experiment on...

### Proposal
We take a lesson from Formula One motor racing, where cars do not maintain a constant speed but instead go faster on the straights and slower around the turns, and...

For a defined Operation speed and PPI, rather than maintain a fixed speed and PPI, instead allow driver to vary these both upwards and downwards dynamically, varying PPI and speed simultaneously to keep the degree of burn the same. 

### Explanation

It seems to be that the reason that rastering can work at much faster speeds are:

1. It allows for acceleration and deceleration distances at the start and finish of each sweep during which the laser is off. Whilst the laser is operating, the speed is constant.
2. Possibly because it is unidirectional - specifically that it is in a constant direction.

The differences between this and vector burns is that vectors:
1. Still need to be at a constant speed, but able to vary direction as needed.
2. Need to maintain a continuous burn at this continuous speed except at sharp corners. 
3. I have seen vector burns at acute-angled corners go past the corner (and turning off) and then restarting again in the other direction - I am unclear whether this is m2-nano functionality or something in MK.

I suspect that the vector burn speed limit is actually a limit not on speed but on the accelerations necessary to do e.g. tight curves at high speeds which would require high accelerations to achieve big velocity changes in short distances. In other words we can do gentle curves at high speeds, but tight curves at low speeds.

If we are going to vector at high speeds, we would also need an acceleration/deceleration distance - and this would either need to be before / after the burn with the laser off (which would effectively require new twitches, or would need to accelerate / decelerate with the laser on using PPI to compensate for the laser remaining on a particular area of the material for longer.

So my proposal is that:
* subject to an upper limit of 1000PPI and (say) 500mm/s and a lower limit of 100PPI (but no lower speed limit except as needed to keep the burn energy constant), 
* we drive the head at a variable speed and use a lower PPI with lower speeds and a higher PPI with higher speeds. 

So for example, if you had a straight line with a tight curl at either end, and an engrave operation defined as 50mm/s @ 500PPI, the tight curl would likely require way too much acceleration to follow at 50mm/s, whereas on the straight you could actually go faster than 50mm/s without any issues. So in this example we would instead:
* burn the first curl at 25mm/s at 250PPI and then 
* accelerate to 100mm/s at 1000PPI during the straight line, before 
* decelerating back down to 25mm/s at 250PPI for the final curl.

The only practical concern is just how often people would actually want to burn at high speed at full power when IME on a K40 you are limited by laser power to vector speeds at 35mm/s or lower. That said, if you have an m2 nano and an 80W laser, the speed limit is much more of a constraint. Equally, it could be argued that 40W tubes are more standard because the m2 nano is limited on vector speeds and much higher power tubes cannot use the extra power because it is limited by vector speeds.

(I am not sure just how well I have explained myself here. Please ask if it is not clear.)

tatarize, 2022-07-20:
moved relevant data to wiki

closed 2022-07-20 by tatarize

Sophist-UK, 2022-07-20:
I am unclear whether my variable speed / variable PPI suggestion has been incorporated or rejected or moved to the wiki...