meerk40t/meerk40t #129: Crash logs

ursnation, 2020-01-26:
Hi,

I'm running this on windows without the laser connected to this machine as I just wanted to create a project to transfer to my linux box which is used to control the laser.

I've got a few crashes:
1. when drag and dropping some images into the plane
2. when doing Reify User Changes after a Convex Hull

Here are the dumps in no particular order:

[MeerK40t-2020-01-26_15_40_43.txt](https://github.com/meerk40t/meerk40t/files/4113528/MeerK40t-2020-01-26_15_40_43.txt)
[MeerK40t-2020-01-26_15_40_56.txt](https://github.com/meerk40t/meerk40t/files/4113529/MeerK40t-2020-01-26_15_40_56.txt)
[MeerK40t-2020-01-26_15_51_45.txt](https://github.com/meerk40t/meerk40t/files/4113530/MeerK40t-2020-01-26_15_51_45.txt)
[MeerK40t-2020-01-26_15_56_00.txt](https://github.com/meerk40t/meerk40t/files/4113531/MeerK40t-2020-01-26_15_56_00.txt)

Cheers!

joerlane, 2020-01-26:
I've run into that bug before; but it is my understanding that the actualization will happen automatically. I think in the very short term, you can ignore that feature; and simply send the job to the laser. It will fix anything that needs to be actualized on the fly. Even without a laser connected, you can fill up the spooler to make sure each of the paths and rasters are applied as desired. Saving the edited file in MeerK40t should save the the conversions and laser job specific settings as well.

tatarize, 2020-01-26:
Thank you for the report. Looks like it's some issues with strangely made objects not having properties that I assumed all objects would have. It seems like in both of those `self.speed` has a value of `None`. This is likely due to a conflict between the self.speed attribute and the self.speed property.

Still quite worth fixing.

tatarize, 2020-01-26:
Ah. The bug is in svgelements.py it wrongly makes a copy of the class object without making a copy of the arbitrary non svg informations in values which is where `speed` is located. It should copy the concrete values and move the dict of values that contain nothing in the original vanilla code.

tatarize, 2020-02-19:
This was fixed as of 0.5.0

closed 2020-02-19 by tatarize