meerk40t/meerk40t #134: Convert wxMeerK40t to wxAUI

tatarize, 2020-02-06:
I'm usually a much more vanilla and clunky looking interface with a lot of windows. This isn't really needed and should be transitioned to something more advanced like wxAUI. 

https://www.kirix.com/labs/wxaui/screenshots.html

frogmaster, 2021-01-15:
+1 on this issue, as I don't use window borders and title bars and use taskbar to close and minimize them, I'm left with bunch of windows in meerk40t i can't really close or move without some extra effort... Something dockable for nav, settings, etc... would be vastly superior...

tatarize, 2021-01-15:
Okay, I might have to run some test with how easy this might be to do. Obviously the old style windows are much less modern, and there's a lot of little windows that could be docked places and be really interesting to have. Like breaking up controller into a bunch of tiny window and let you use the parts of of it you care about independently.

tatarize, 2021-01-15:
Running a cursory experiment in 0.7.0 seems like conversion isn't too hard and I can still use most of my established widgets already. Might take some work making interesting and useful subelements out of controller and nav etc. But, it's not too hard implementation-wise.

tatarize, 2021-05-20:
This was made to be saving and restoring in 0.7.0. No access to panes.

closed 2021-08-12 by tatarize