meerk40t/meerk40t #136: customization issue

jhindes521, 2020-02-10:
having issues with the customization of axsis directions and setting home greyed out and none reponsive any ideas? 

wwin 10 opperating system installed via cmd / python

tatarize, 2020-02-10:
That's by design. I added the GUI elements, but I neglected to actually code the stuff up. I'll push the feature up after the current version gets done. Squeaky wheels get the grease. 

I still have the issue as #86 since it's not yet finished.  It needed some gui controls to be able to be completed, but so when I was redoing the gui, I included the axis stuff. But, since I didn't have the code done, I disabled it so people wouldn't turn it on and expect it to work.

I'm currently in a feature freeze for 0.5.0 but I'll move that to the top of the list for the next version.

jhindes521, 2020-02-10:
Ah brilliant. Was worried I’d missed something during Cmd install and had gone through it several times and come up blank. Thanks for everything you’ve done so far. V impressive ! 
J

tatarize, 2020-02-22:
This was corrected in 0.5.1 and is now closed.

closed 2020-02-22 by tatarize