meerk40t/meerk40t #150: Camera Interface

tatarize, 2020-02-22:
The camera uses OpenCV for the camera but also has built in camera calibration to correct for the fisheye distortion. 

The interface has a couple primary functions.
* Fisheye distortion correction, this is also provided by opencv
    * https://medium.com/@kennethjiang/calibrate-fisheye-lens-using-opencv-333b05afa0b0
    * https://stackoverflow.com/questions/50857278/raspicam-fisheye-calibration-with-opencv

* Send frame as background to scene. The frame to background works but without the fisheye correction this isn't wildly helpful. Also, this function will require the background be adjusted positionally. 

* Send frame to scene as element. This currently has a button, and the functionality isn't too difficult but isn't implemented.

1. OpenCV does not play nice with pyinstaller.
2. I don't have a camera for my laser. I can't really test or calibrate any of this.
3. Running before it walks. Most of the power-functions for a camera-laser require commands and code MeerK40t currently does not have. This was a request by @joerlane just to watch his laser. 
4. This is a non-core element with optional requirements. And should be given lower priority.

---

Feel free to have any discussions or comments about the Camera here. Or help it move up in priority by popular demand.

joerlane, 2020-02-23:
The background image would need to be distorted, scaled, and cropped to conform to a properly calibrated machine’s allotted bed size. Typically one chooses a camera with a slightly wider field of view than needed to ensure full coverage for this purpose. But even with simple distortion fixing; the camera is likely still at an angle not perpendicular with the bed; and vaguely different for each particular installation. Adjustable mounts here are commonplace.

The current interface is very promising. The image capture -> background is a pretty ideal situation once the image is modified. I marked off the edges of my bed by marking material at the extreme edges. All of the marks show up in the view of the camera; and are visible in the captured image. 

A calibration routine should be relatively simple to implement; and should only serve to fine-tune the image to bed transformation. Calibration itself could be a simple as drawing a few vectors with path input commands; marking on a test material, and then refreshing the background image to verify alignment. Background adjustment settings could be similar to navigation controls to make fine adjustments of scale, crop, shift, stretch, and angle compensation.... Sounds really complicated when you spell it out like that. 

The kickstand to hold up the cover is an important consideration on the K40. If supported on a single end; the lid flexes significantly. In theory that should also be possible to calibrate away in software; though, a sturdy, highly repeatable, center supporting implement is a lot closer to an ideal situation.

tatarize, 2020-03-01:
https://www.pyimagesearch.com/2014/05/05/building-pokedex-python-opencv-perspective-warping-step-5-6/

And step 6 of this guide should be added. Needs to try to capture the bed, Also, should permit fixing the perspective issues. And likely add in super-res functionality in there if it's easy to implement.

tatarize, 2020-03-03:
Implemented the zoom and pan stuff and moving the perspective widgets. So even from an angle it should get an overhead view.

It's going to try turning those four vertices into something the size of the bed. This should work effectively well enough for now. It might also be awesome or useful in other ways.

joerlane, 2020-03-04:
Having tested this pretty thoroughly; I have a few comments on the setup I used to achieve measurements indistinguishable from the object placement in software over the background image. Some of this was eluded to before; but I've been able to nail down some things after testing. 

First; I had previously measured out my bed by jogging the laser head around carefully. I set those limits as dimensions in the MeerK40t preferences. Then I created an inkscape file that was the same size as the bed dimensions; with a cut rectangle of those same dimensions. Simply setting the X and Y of that rectangle to zero in the file so it would align to the document edges, and ultimately mark the extreme's of the available bed.

I loaded up a piece of precisely cut cardboard (white cardboard is nice for this), and ran that job at ultra low power so it only left a mark in the cardboard. Then I marked the machine and the cardboard together with nail polish so that I could re-align them later to the exact same place. I then glued my checkerboard pattern to the cardboard, with registration in the upper right hand corner. At this point; some darker pen marks might help as guides to find those locations if the resolution of the webcam is low.

Then I loaded the cardboard/checkerboard back into the machine as precisely as it was the first time when it was cut (using the nail polish to assist alignment). I'm using a kickstand that holds my cover open which is 11-1/2"; between the bottom of the cover and the top of the machine where the cover lays. Individual setups will differ in terms of that kickstand length based on the angle the camera is mounted (or where it attaches. I chose this particular angle so the camera could also provided a reasonable view of the laser head and bed with the cover closed. That allows me to also watch jobs on the video feed as they run.

After doing the de-distortion setup, and then setting the perspective handles to map to the actual corners and lines on the cardboard; I was ready to setup a test job. I laid a piece of MDF over the top of the checkerboard, then loaded a file of rasters. I then moved those objects around to different locations I could easily map out based on the still visible checkerboard squares. Ran the job once, and compared the output to the screen. My first attempt was about 1-1.5mm off shifted to the right; but the vertical was fine. From that knowledge I re-adjusted my perspective points to further compensate. Then resent the adjusted image to the background. 

This process may sound involved, but it actually happens rather quickly. It did not take very many tries to end up with objects that were so well placed I couldn't perceive or measure any deviation. 

One thing I paid close attention to during that process, was how level my cover was when it was on the kickstand. I place the kickstand in the center under the cover, so that the cover doesn't bend left or right. I also align my cover far left; up against the machines edge to close that gap as much as possible (and add to the repeatability). Finally; I added a spirit level (bubble level) to the inside of the cover to allow me to more closely verify that for future repeatability. 

It does help a lot to zoom in extensively when setting each of the perspective handles. They can be adjusted at the sub-pixel level over the camera image, which is necessary if one wishes to have sub-millimeter alignment ability. But it does work; and that alignment resolution isn't all that difficult to achieve, even with a very low resolution webcam. I thought a lot about this setup process, but it only took me a few tries start to finish.

tatarize, 2020-03-05:
So are there any features that might be superior? I simply gave the perspective correction because it checked all the right boxes of "solves the problem". I know lightburn burns things that aren't in the corner then asks where those things actually are. In theory that could achieve the same result and is maybe better tested. I certainly wouldn't do them this version but future versions could certainly have such things.

Other than that. Proper functionality report. Laptops where you prefer it choose your second camera goes ahead and chooses your first. If you have more than one camera you're out of luck.

tatarize, 2020-05-15:
https://www.youtube.com/watch?v=HNfPbw-1e_w Calibration setup instructions for any future documentation.

tatarize, 2020-10-11:
Youtube review. Likely from a higher understanding user.

"Karel Kuna -- 10 hours ago simply WOW. -- this is amazing. I just install cheap camera from aliexpress to my 4060,  put carton bigger than work place, lasered 4 dots in workplace corners and calibrate camera with checker board + distortion correction to these 4 corners. It is spot on now and so easy to arrange jobs.  Thank you very much for this feature."

tatarize, 2020-10-11:
[image]

This checkerboard pattern likely needs to be more commonly tossed around since it's kinda needed to undo fisheye.

closed 2021-01-30 by tatarize