meerk40t/meerk40t #152: Interface suggestions

josemariaaraujo, 2020-02-23:
Hi, just found about this project in reddit, and I think you have done a great work!

I've tried it with great success, and have a couple of suggestions if you don't mind:

- When selecting a path in "Operations", it should highlight it in the graphic view, like it does if you select it in "Elements". It whould allow for easier reordering of cuts (right now I have to sort the elements before creating a new cut operation)
- Allow for placing the laser head anywhere on the table by clicking on the graphic view or something in the order: : right now it's hard to confirm that a cut fit's a tightly dimensioned material. In K40Whisperer you have a shortcut to go to the four corners of the cutting envelope.
- There's a small but noticeable delay between cutting each path in a single cut operation. When cutting a lot of small parts it seem's like a lot of time lost.

I hope you don't take this comments wrong, You have done ag reat work and looking forward for improvements to make it the best software for our K40s.

I was developping a ESP32 GRBL Shield for the K40, and now that I found  meerk40t don't know if it is really worth the effort!

joerlane, 2020-02-23:
I can add comments to a few of your suggestions. The selection issue is a legitimate concern; and there was a previous development snapshot where that actually worked. Unfortunately it was problematic in some other ways at the time... Presumably it will return again when those details are worked out.

The newest release does have an alternative home position available; which can be helpful when placing tightly dimensioned material. There are also numeric shortcut keys (numbers 1-5) available for custom positions. One can use the arrow keys to step around 1mm at a time, and also the navigation pane which has an adjustable step size. Then simply press a number 1-5 to save that location. Further key presses of that number will recall the location saved, and the shuttle will drive to that location.

What speed are you cutting the parts when you notice a delay? There is a slowdown feature in the M2 Nano controller which drives the head backwards (a sort of active braking); and that may be at issue in this case. I just now checked one of my testing files, and noticed it might be trying too hard to slow down after running at 10mm/s (and perhaps slower). I'll raise an issue if I can find constraints; preliminary testing shows it might be attempting to slow down less at a higher speed; which isn't intuitive or helpful in this case. 

I too looked at replacing my K40s stock controller early on; until I got more comfortable with the available open source software. 

Your comments are absolutely appreciated; and I'm quite sure you'll receive even more feedback in regards to those suggestions here soon.

josemariaaraujo, 2020-02-23:
Thank you for your prompt  answer.
Custom positions are cool and a good answer for what I was looking for, but found a couple of issues: can't reset them after saving the first time, and don't work if the cursor isn't over the graphic view, like when it's over the navigation window. These two issues combined makes them hard to use, when testing more than once set a position wrong and then that key was unusable anymore (until restart, of course).

I noticed the delay between paths when cutting 3mm plywood at 18mm/s. I can send the svg if needed. I think you're right about the driving the head backwards, I've just run the file again (with the laser off just for testing) and noticed it pulls back after each path, indeed at a slower speed than the cut itself!

Thank you everyone for the good work!

joerlane, 2020-02-23:
Thanks for reporting those issues. I was aware of the mouse over the graphic view being an concern; but while investigating the nuances I came across some other problems that can arise from trying to navigate this way. Clicking navigation arrow buttons (maybe with the nav dialog out of focus), and then using to the arrow keys may well change the step size. This could be bad if it goes unnoticed. 

It looks like key mappings are an incoming feature. Allowing the setting of custom positions manually (and resetting them), should be added to the list of possible options. 

That 18mm/s is on the faster end; but I watched my machine a bit closer and can't perceive a difference in deceleration between 10mm/s and 35mm/s off hand. At 10mm/s it shouldn't really need any at all; and at 35mm/s it's still exaggerated in an unnecessary way. Maybe that needs to be a setting as well. Banding for speed codes should be user adjustable; or at least closer to those of rasters in terms of speed. 

I do wonder if it has to do with the drawing order of the part though. If the component ends in the Y direction; is it faster or slower between paths? Is there another optimization happening on the fly? These are questions about MeerK40t, and the M2 Nano's reactions; and somewhat rhetorical in both cases...

What OS are you testing with?

josemariaaraujo, 2020-02-23:
Here is the SVG:
[key.zip](https://github.com/meerk40t/meerk40t/files/4242295/key.zip)
I just cut it without changes, after deleting the raster lines (leftovers of dxf imported from fusion).

I'm using Windows 10.

tatarize, 2020-02-24:
Seriously thank you. The interface's rough edges are the current highlight in 0.5.2 all suggestions along those lines are gold.

1) Select items from Operation Elements and have it highlight them like when selecting the Element Branch Node. This was added with the latest commit to 0.5.2 just now.
2) Do move to corners navigation. This was added earlier today. 
[image]

The other movement dialog does the more whisperer like functionality. It moves to the corner and the center as needed and lets you hit the side buttons there to drag the design and move the laser head.

3) "There's a small but noticeable delay between cutting each path in a single cut operation. When cutting a lot of small parts it seems like a lot of time lost."

This is certainly fixable. It'll help joerlane too since getting rid of the waits between elements means implementing `@` in micro-gl rather than just `F` for everything. OSX has some issues with those waits, at times. Which would heavily reduce but not correct #111.  Basically there's two ways to quit a speed setting, and I only really implemented one. And used it for everything. I think it forces the decel when you exit a speed like that.

--

My strategy here is often squeaky wheels get the grease. Since I imagine there's dozens more people not saying things for all the people who are, and quick results might encourage more feedback. Odds are quite good all three of those will be added before next version. Especially considering two of them area already in the 0.5.2 branch.

tatarize, 2020-02-24:
@josemariaaraujo Okay, and the final correction got added to 0.5.2. Implements the `@` command which should go markedly faster for those transitions.

Far from taking offense, I'm much more likely to move any suggestion offered to the top of the pile. Have any more?

tatarize, 2020-02-24:
For record keeping:

* Custom positions. can't reset them after saving the first time, 
* Don't work if the cursor isn't over the graphic view, like when it's over the navigation window. 
* The rearrange of items is a bit weird when multiples are selected. Sometimes it flags the item based on the object backing it, but that's actually unhelpful since there can be multiple copies of the same item in the same job.

josemariaaraujo, 2020-02-26:
Thank you all for prompt replies and accepting my suggestions.
I'm just a hobbyist without professional experience that uses the k40 from time to time , those issues where the ones that standed out the most to me. Surely when more proficient users appear they will have more and better suggestions.

tatarize, 2020-02-26:
Most K40 users are hobbyists and while I'd like to also help out professional folks too, there's no reason to devalue your contributions. And I don't see lots of more proficient users raising issues. I'll triage the grease when there's more people offering stuff but until then, the most recent squeaky wheel. Most of these things are on the agenda but the order those items are dealt with or their importance given to them isn't something I have firm grasp on. In fact, it undervalues those contributions more because really intuitive UI should be obvious to new users. Rather than just those who have seen such things before and face a less stark learning curve.

tatarize, 2020-03-11:
Most of these are live now. Still holding off off closing this until I fix the hotkeys thing. I pushed it to the next version.

tatarize, 2020-05-20:
The hotkeys things is fixed in 0.6.0 though the text movement thing might just not be feasible and I should maybe allow the arrow keys to default to arrow keys. Maybe using shift arrow keys to do jogging instead, so you can use them in the tree menu just as easily.

tatarize, 2020-06-23:
I moved the arrow keys to the WASD keys so they don't overlap. And added the hotkey things.

This issue was corrected in 0.6.0. Which is now in full release.

closed 2020-06-23 by tatarize