meerk40t/meerk40t #157: what's the problem?

tomislav17, 2020-03-04:
MeerK40t crash log. Version: 0.5.1
Please report to: https://github.com/meerk40t/meerk40t/issues

Traceback (most recent call last):
  File "CH341LibusbDriver.py", line 209, in CH341InitParallel
  File "site-packages\usb\core.py", line 1043, in ctrl_transfer
  File "site-packages\usb\backend\libusb0.py", line 593, in ctrl_transfer
  File "site-packages\usb\backend\libusb0.py", line 431, in _check
usb.core.USBError: [Errno None] b'libusb0-dll:err [control_msg] sending control message failed, win error: A device which does not exist was specified.\r\n\n'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "Controller.py", line 370, in on_button_start_usb
  File "Kernel.py", line 223, in execute
  File "K40Controller.py", line 224, in open
  File "CH341LibusbDriver.py", line 363, in open
  File "CH341LibusbDriver.py", line 212, in CH341InitParallel
ConnectionError

tatarize, 2020-03-05:
Well, it's welcome to not connect to the device, but it died trying to perform the control switch between the CH341 into EPP 1.9 mode to the Lhychip. It should not have thrown that error to you. It should have tried the alternative driver when the libusb driver did not find the device as one that existed, and simply logged that failure.

You might have libusb installed but for something else, or it could be a strange presentation of the device not being connected to correctly. In either case it should not have crashed that. I'll make sure that error forces it into a backup method. And logs the failure correctly. 

"A device which does not exist was specified."  --- But, that should be dealt with and logged rather than hard crashing.

tomislav17, 2020-03-05:
the older version of the program is working, the new one is not working

tatarize, 2020-03-05:
What's wrong with the new one? Something other than that section I assume, since that's a 0.5.1 crash and nothing in that section was changed.

tatarize, 2020-03-05:
Also, apparently you specified a USB Index rather than leaving it at -1. Judging from the position of the calling function. It's fine but mostly for multiple K40s and in this case isn't finding that index hence the crash. Though still shouldn't let it go all the way up to top level.

tomislav17, 2020-03-05:
reinstall driver and fix problem

closed 2020-03-05 by tomislav17

tatarize, 2020-03-06:
Still a solid bug report. I'll have the crash itself fixed in 0.5.2 and give you a better explanation of what failed and where.