meerk40t/meerk40t #158: WxPython 4.0.7 Mac, default matrix is inverted.

tatarize, 2020-03-09:
For unknown reasons things that use a matrix are flipped like they have a scale_y of -1 for most things in the scene. 4.1+ works fine. And it might be a specific bug for a specific version.

tatarize, 2020-06-02:
Since there was no way to figure out this error an option was added in 0.6.0 to add FlipXY and undo this inversion even while not knowing it exists.

tatarize, 2020-06-02:
https://github.com/meerk40t/meerk40t/commit/74d4455a058f064711c13f9ec9318b8ff4c64e7d

joerlane, 2020-06-23:
This went away when 4.1.0 went mainstream. Closable issue from what I can tell.

tatarize, 2020-06-23:
Also, the work around is the only thing I could do. I guess they have a pointless option to make it weirdly confusing. All good anyway.

This issue was corrected in 0.6.0. Which is now in full release.

closed 2020-06-23 by tatarize