meerk40t/meerk40t #161: Installation trojan warning

sbengali, 2020-03-30:
Installation failed with the following message from Windows Defender: 
Threat detected: Trojan:Win32/Wacatac.D!ml
Alert level: Severe
Category: Trojan
Details: This program is dangerous and executes commands from an attacker.

Affected items: ..\MeerK40t.exe
webfile: |https://github-production-release-asset-2e65be.s3.amazonaws.com/197991764/bab78b80-633c-11ea-9ef7-fd87b2500563?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20200330%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20200330T022844Z&X-Amz-Expires=300&X-Amz-Signature=608ff35e81f5b0f543aef346fc434401b05a62fd4925268904def76681a2d76e&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3DMeerK40t.exe&response-content-type=application%2Foctet-stream|pid:4652,ProcessStart:132292097584493200

Bftech, 2020-03-30:
I got the same problem a few days ago with the latest (0.5.2) ".exe" release on Windows 10.
This is the virus total analysis : https://www.virustotal.com/gui/file/7569e11c2911649a722dabd81880d7cd2a8868e68a5edad22eb95e23b506f8ba/detection

tatarize, 2020-03-30:
I would just recommend running from the python directly, if concerned and to always be concerned. It's a lot smaller anyway to do that. 

My guess is those programs look the same because they are also python based pyinstaller execs. So a pretty significant portion of the code will be similar.  Since of the 50 meg size something like 1 meg is the total actual source size.

https://stackoverflow.com/questions/43777106/program-made-with-pyinstaller-now-seen-as-a-trojan-horse-by-avg

Has some proposed solutions, apparently recompiling pyinstaller locally will make it less likely to look the same since the compiled stuff will be different from the viruses' compiled stuff.

tatarize, 2020-03-30:
Yeah, I'll recompile the bootloaders and update the wiki for how to build. Assuming it then passes the virustotal check thing, after making the changes.

https://github.com/meerk40t/meerk40t/wiki/How-to-generate-MeerK40t.exe

tatarize, 2020-04-01:
Properly recompiling the bootloaders required visual studio C++ but the install is a bit hinky. If this was *just* viraltotal giving a false positive I wouldn't worry much. But Windows Defender is built into Windows so at a minimum that needs to be avoided, and compiled in a way that doesn't object.

tatarize, 2020-04-06:
Ah, I see what VirusTotal is doing. I didn't grasp that before. It's running all the antiviral stuff on each product. What I need to do is make sure the Microsoft false positive doesn't exist. Since that's defender and the problematic item. I could likely coax that into correctness by pre-running a check on any new release. Even a second run of pyinstaller seems like it passed that. So I should be able to fiddle with the pyinstaller version or whatnot and get beyond this.

tatarize, 2020-06-23:
This is still an issue on 0.6.0 I'm going to try a different computer to compile on since mine doesn't like recompiling the bootloader.

tatarize, 2020-06-23:
Okay, recompiled and fixed. Microsoft no longer issues a false positive. And the only things that still issue false positives are pretty obscure and I've not heard of them.

https://www.virustotal.com/gui/file/f764814c3cae33b43d361d97e135b6dc0c033ddac95296a14fbdfb19c4eb78c0/detection

tatarize, 2020-06-23:
Going forward this will be a release check. And I can manage to solve it.

closed 2020-06-23 by tatarize