meerk40t/meerk40t #163: When opening navigation window, error thrown

stevebottrell, 2020-04-22:
MeerK40t crash log. Version: 0.5.2
Please report to: https://github.com/meerk40t/meerk40t/issues

Traceback (most recent call last):
  File "wxMeerK40t.py", line 285, in <lambda>
  File "Kernel.py", line 893, in open_window
  File "Navigation.py", line 350, in set_kernel
  File "Kernel.py", line 241, in setting
  File "Kernel.py", line 701, in __getitem__
  File "Kernel.py", line 794, in read_config
wx._core.wxAssertionError: C++ assertion "!IsNumericValue(szValue)" failed at ..\..\src\msw\registry.cpp(1041) in wxRegKey::QueryValue(): Type mismatch in wxRegKey::QueryValue().

tatarize, 2020-04-22:
Hm. Apparently some value in the windows version there changed its typing. I've seen errors like that at points. And sometimes the routine that saves the stuff to the registry, treats it as a different variable type. And it type matches to another. Like a float that gets saved the first time as an int because I wasn't careful with it, so the next time around when it tries to load the value, it can't do it correctly because it saved it as the wrong type.

There's some chance because this thing will have messed up the initial registry save that it could crash each time the window is opened. If that's the case you'd need to purge the registry entries.  These are at: HKEY_CURRENT_USER\Software\MeerK40t

You can also go into Controller and right click in there, it'll have a menu popup called "Force Kernel Events" one of which is "Delete Settings" at the top. If it's not a one off crash, that'll let you purge those settings, so they end up saving correctly.

Found it:
Repeating steps.
0) Using Microsoft Windows
1) Delete registry keys.
2) Load up MeerK40t.
3) Open Navigation.
4) Change the `Short Pulse` SpinControl value.
5) Close Navigation and MeerK40t.
6) Reload MeerK40t.
7) Open Navigation.

tatarize, 2020-04-22:
https://github.com/meerk40t/meerk40t/commit/f4196d9b910d953fca915c92341cf4a34502ecaa has it corrected. It'll be fixed next version.

Thank you for reporting that. You can mitigate the problem by deleting just that key or clearing the settings.

wilnero, 2020-05-01:
> [f4196d9](https://github.com/meerk40t/meerk40t/commit/f4196d9b910d953fca915c92341cf4a34502ecaa) has it corrected. It'll be fixed next version.
> 
> Thank you for reporting that. You can mitigate the problem by deleting just that key or clearing the settings.

I did it, but the problem still occurring. I tried both, deleting register key and cleaning the settings.

tatarize, 2020-05-01:
Hm. I quickly diagnosed it and duplicated it, but those corrections also fixed it for me.

Maybe I misdiagnosed it. Doesn't seem like it should persist if you kill the settings, then close the program. Open the program. Open navigation. Close Navigation. Close the program. Then Open the program again. After that point it should have the setting saved correctly as a float.

Ah, it is misdiagnosed. The bug is more serious than I previously gave it credit for. It can be recreated by merely modifying that value. It will load incorrectly and always give that annoying message when navigation is started.

Thus the mitigation requires clearing out that registry entry, and then not adjusting the pulse duration. And if you wanted to change it to do so in the registry rather than the program. That's annoying.

[image]

closed 2020-06-23 by tatarize