meerk40t/meerk40t #164: Error When first Launchin OSX

COWBOYSnANTHRAX, 2020-05-08:
Traceback (most recent call last):
  File "Meerrrr/MeerK40t.py", line 107, in <module>
  File "Meerrrr/svgelements.py", line 4528, in __init__
  File "Meerrrr/svgelements.py", line 4690, in parse
  File "Meerrrr/svgelements.py", line 247, in svg_parse
  File "Meerrrr/svgelements.py", line 208, in parse
  File "Meerrrr/svgelements.py", line 302, in smooth_cubic_to
  File "Meerrrr/svgelements.py", line 253, in get_pos
  File "Meerrrr/svgelements.py", line 180, in get
IndexError: pop from empty list

I get this error when opening for the first time. When I am able to start the application I can not change an operation to engrave or cut. Any file I open is raster only. Sometime when duplicating a element, a new cut or engrave operation will show up.

joerlane, 2020-05-08:
Just a couple of quick questions... What version of Mac OS X? Are you loading an SVG file; what does your file contain? Do you have pure red, and pure blue, vector paths for cut and engrave operations? Are you able to open the MeerK40t application without loading a file? Does it communicate with the laser cutter (the navigation panel should work)? Where/how was the file created? 

I will try to reproduce any and all of those problems with a bit more information.

COWBOYSnANTHRAX, 2020-05-08:
I am currently on 10.14.3. I am loading an svg file. I have tried a number of different files. Some with basic outlines and one with just two vector lines. I have tried both black and blue vectors. All have been pure RGB colors of both black and blue. Communication with the laser cutter seems to be fine. All navigation is working as expected.

joerlane, 2020-05-08:
Ah... What version of Illustrator? 

I'm running a similar version of OS X (10.14.5); and compile MeerK40t on the same. Seems like it should be reasonably compatible in that sense.

I've only really been testing with Inkscape generated SVG files though... I have an older version of Illustrator I can try generating files with; or converting my old .ai files from into SVGs. There may well be some bugs... SVG files are human readable with a text editor; so it shouldn't be too difficult to figure out where those issues are.

COWBOYSnANTHRAX, 2020-05-08:
I am using the CC version of illustrator which is 24.1.2. I will try using Inkscape and see if I encounter the same problems.

COWBOYSnANTHRAX, 2020-05-08:
It appears files generated in inkscape work by changing the color value to change engrave/cut/raster. If I could get this to work when generating with AI I would be happy. When I try to import an svg into inkscape that was created in AI and save it, it will not work. All AI generated files import as black fill objects regardless of color.

joerlane, 2020-05-08:
I don't have anything close to the Illustrator CC version in hand (I can try to find a way if needed); but you could try to open the .ai converted/saved SVG file; and look at the color value parameter. I'm not sure why it would be off; but it might shed some light on why it's not registering as a vector.

tatarize, 2020-05-08:
Any chance I can get a copy of the SVG that breaks? From the log there it looks like it's a genuine bug in what seems kinda like a malformed path but is likely perfectly fine. (You'd need to zip up the SVG file and drop it here). You could also delete most of the paths to find the offending one. Also you can email me it at tatarize at the gmail, if you don't want the file floating out there. I'd be happy to respect your IP.

COWBOYSnANTHRAX, 2020-05-08:
So I have imported an AI file into Inkscape and have changed all colors to RGB blue (they were all set to RGB blue from AI). I saved it in inkscape. I can now open in Meerk40t and it will show engrave but it will also show raster for the same file which is completely RGB blue. I can then delete all raster operations. I still have to change fill/stroke in meerk40t for it not to show as black filled boxes. Its workable for now but with a lot of extra steps.

tatarize, 2020-05-08:
Well, all those extra unneeded steps are issues. The failure to load the original SVG; that should not crash. (might need the original file to fix it). Some of the UI issues of having it harder than it needs to be, they are all clearly concerns.

I think the completely blue shows up as a raster-operation too is a legit as well. I ran a copy of Basic15 testfile it has a blue dragon that got rastered. I think the decision was to do that like Whisperer did that for the color -> operation classifications. I'll double check that one. Though if they are black-filled those are rastered operations.

COWBOYSnANTHRAX, 2020-05-08:
Here is the original file saved directly from AI. 
[maineoutlinerailscale.svg.zip](https://github.com/meerk40t/meerk40t/files/4596736/maineoutlinerailscale.svg.zip)

When opened in meerk40t it only shows raster operations. If I bring this file into inkscape and reapply the rgb blue values it will create both raster and engrave operations in meerk40t.

tatarize, 2020-05-08:
No crash on my end sadly. Is this the original that could properly crash?

The SVG in this case uses a class-based style. This is in the process of being corrected. It's initial reason for PR #160. In this case the paths have class="st0" and that is defined in a style-tag seen above it. This is valid integration of CSS and SVG it wasn't correctly mapped processed. It should be fixed in the next version. I decided to add that stuff after seeing that's how Illustrator did it a few months back.

If I can get the crashing to happen on my end I'll fix that too.

COWBOYSnANTHRAX, 2020-05-08:
Im sorry I may not have been clear. The crash occurs after opening the .dmg the first few times. Then it seems to go away. The crash is not tied to loading files within the application. I will try to play around and figure out why my AI SVGS are having issues.

techieStephy, 2020-05-09:
i get the same when trying to open the dmg. can't get into it at all 
[image]

joerlane, 2020-05-09:
Thanks for this OS X error report. What version of OS X are you using @techieStephy? I haven't been able to reproduce it myself; but since I compile on the same machine I'm using to test, it may be part of the reason.

I'll work on loading up a fresh OS install on another machine, but that could take some time to complete.

tatarize, 2020-05-10:
I don't understand what it could be parsing there.  It's apparently on start up trying to parse a thing with a malformed smooth_cubic cubic curve. I would recommend trying to just run it directly with python. It should be fine to install the requirements.txt and running it directly. I suppose I could also track the crash down in svg_parse and just tell it to not crash there, I am revisiting some of that code. Something invoking an svg parse with an S command and fewer than 4 operands. 

Assuming the OSX commandline operations work, you could launch it with a `-v` flag it'll launch and make a debug file. Which will be able to tell all the function calls made and what triggered them. At least then I might know what it's trying to parse and why it does that on launch. Though with the Meerrrr directiory it's likely something with Joe's doing.

techieStephy, 2020-05-10:
[image]
thats my machine, :)

tatarize, 2020-05-10:
The crash is being triggered by a set of flags namely something that says something like `-p M0,0S0,0` which will crash and fail to launch that way even in windows. I'm not sure why the OS would call anything like that, but at least I can fix part of it, on my end. When the CLI parses the path it should just catch any errors and throw the path away if it's broken.

tatarize, 2020-05-10:
[image]

Knowing a bit more about why, I can at least force it to happen in windows. Apparently pyinstaller won't ignore the crash on startup because likely happened *before* MeerK40t took over the crash handler.

joerlane, 2020-05-10:
For the debug on the Mac. Open terminal; browse to where the MeerK40t.app is, then use the command:  ./MeerK40t.app/Contents/MacOS/MeerK40t -v

tatarize, 2020-05-10:
It can crash rather minimalistically with -p S

Though if it's being launched from the terminal there's a good chance it'll just load. Since it won't force the command line stuff, which apparently is being inserted somewhere and has a -p flag and contains the letter S after that.

tatarize, 2020-05-10:
Okay, @joerlane added a hotfix. Should be able possible to pull that and replace the current version of 0.5.2 with the hotfixed version. Only older OSX versions are reporting it, so we shouldn't need to rebuild the suite.

joerlane, 2020-05-10:
There is a new .dmg uploaded to the release area for Mac OS X with the hotfix included.

tatarize, 2020-05-20:
I also added for the next version if it catches that error, it will send to the standard out a copy of the command line used to call the thing. That'll make debugging this easier if things come up.

closed 2020-05-30 by tatarize