meerk40t/meerk40t #168: Hidden Paths in SVG Shown in meerk40t

kshenneman, 2020-05-13:
Will meerk40t support hidden objects within the SVG file? I currently have SVG files created in Inkscape with hidden objects. K40Whisperer does not display or try to cut the hidden objects, but meerk40t does.

tatarize, 2020-05-14:
Yeah, I'll add it to the list for the svg update. It currently ignores that value. 
`visibility="hidden"` it also currently treats deals with the `<def>` values wrongly and shows them. That's also being fixed.

tatarize, 2020-05-20:
https://github.com/meerk40t/meerk40t/commit/2a24593b54155abadf1ad62eb4c44b45b7f9b73c

This is corrected in the next version branch.

tatarize, 2020-06-23:
This issue was corrected in 0.6.0. Which is now in full release.

closed 2020-06-23 by tatarize