meerk40t/meerk40t #169: No Version Number Listed

kshenneman, 2020-05-19:
Could you please include the version number in the About window.  After you download the software, there seems to be no way of telling what version you have, (Oh, I just found it...) except in the title bar of the application, which is very small and obscure.  Under the Meerkat logo on the About page would be a great location.

tatarize, 2020-05-20:
Okay, added. My answer to minor aesthetic changes is basically always yes.

https://github.com/meerk40t/meerk40t/pull/160/commits/49631f5e67c33f451ad040a19e6184da39254ab7

Next versions about page looks like:

[image]

tatarize, 2020-06-23:
This issue was corrected in 0.6.0. Which is now in full release.

closed 2020-06-23 by tatarize