meerk40t/meerk40t #170: Can't Save Cut/Engrave Settings

kshenneman, 2020-05-20:
If I create an SVG file and open it in Meerk40t, then change the default engrave/cut settings, like the speed and power for instance, then save the SVG file. When I open the SVG file again, the settings go back the default.  It would be nice to be able to save my settings for future use so I don't have to write them down and input them every time I want to engrave the same file.  It would also eliminate typos in future sessions.  I can just open the file and engrave.

tatarize, 2020-05-20:
Okay, added to next version branch.

https://github.com/meerk40t/meerk40t/commit/f8356c2933d5592587c3385e0b34b6e09dbfacfe

When I get 0.6.0 it'll be a part of that. Should use whatever setting as set last. Still allows override via elements property. I'll make a note on the todo list to test all the other save bits and to make sure dratio doesn't default to zero.

tatarize, 2020-05-21:
Hm. Upon re-reading this wants the save bit for the elements working again. Those were lost at the upgrade to 0.5.0 because of the switch to Operation primary object. I saved saved out the default classifications between sessions but not the particular elements to disk correctly.

I'll take that under advisement.

tatarize, 2020-05-30:
#153 is quite similar to this. Basically saving doesn't save the operations.

tatarize, 2020-07-16:
This is corrected in 0.6.1, now live.

The proper correction rather than the minor one I did with regard to this report initially.

closed 2020-07-16 by tatarize