meerk40t/meerk40t #171: DXF import: polylines are not imported correctly

dietervansteenwegen, 2020-05-29:
DXF created in Inventor 2018 (export face as...). Renamed to txt as uploading a dxf doesn't seem possible.

K40 Whisperer opens and burns this file correctly. Meerk40t does not, for example the outer fillets on the 4 panel corners are replaced by straight lines. What's worse: no error message is shown.

FWIW, Inkscape has issues importing these files as well (warning "`16 ENTITIES of type POLYLINE encountered and ignored. Please try to convert to Release 13 format using QCad.`"). The recommended solution to convert with QCad doesn't help. 
Furthermore, this DXF has been created by a genuine Autodesk product, so I don't see why it shouldn't work (… and it works in k40 Wisperer so every time  i need to laser a DXF I find myself going back to wisperer I'm afraid.)

[Top panel with DIN.txt](https://github.com/meerk40t/meerk40t/files/4703467/Top.panel.with.DIN.txt)

tatarize, 2020-05-29:
DXF uses the ezdfx code which does a great job interpreting everything in the dxf files. The problem is I haven't done the leg work on what the dxf objects actually are and mean and what to do with them. The edges of the bed should be rounded but are not rounded. The upper left hand circle is rendered wrongly as a square without rounding. The oval shaped outermost bits on the center left are also not rounded and should be. And a couple additional things are also not rounded when they should be.

Surprisingly it actually not too terrible. I mean, I put the dxf stuff together in a couple hours rather than giving it all the polish it actually needed. Mostly from a lack of test files.  Judging from the shapes there's some curve property of some of these elements that I just ignored. Lemme track that down for you.

tatarize, 2020-05-29:
Ah, the polyline objects have arcs. And ezdxf has been updated to allow `explode()` to read those more easily. Had that feature been there when I first wrote that code I'da used that. Okay, I'll check that and subparse those elements, rather than simply converting DXF Polyline to SVG Polyline on the assumption that it contains many-lines and does not contain arcs.

tatarize, 2020-05-30:
[image]

Okay, edited that stuff to get the DXF Bulges that the polylines have implemented (https://github.com/meerk40t/meerk40t/pull/160/commits/6bfb51c754d0eff883d5ac6e6e7b0956a0a6c915). It's fixed for 0.6.0  though Rastering is still broken in that version but you could run it a bit to see it generally works. Hopefully feature freezing and getting that version out. Got weirdly addicted to Ruida reverse engineering.

dietervansteenwegen, 2020-05-30:
Great. Looking forward to seeing this implemented!
Thanks, much appreciate the effort.

dietervansteenwegen, 2020-05-30:
Though, not sure whether

> though Rastering is still broken in that version

is an issue for me as I'm not sure when I would need it. For the last few months I've mostly reverted to k40wisperer because family life has been hectic and I couldn't find the time to figure out how everything worked in Meerk40t. For instance, Wisperer just asks for 3 colors to vector/raster/cut. Meerk40t trows all elements together and I couldn't find out how to easily get some things cut and some vector engraved. There are far fewer options compared to Meerk40t so it's much easier to work with, as long as you don't need anything it doesn't offer.
It's about time to put some effort into figuring out how to work with Meerk40t. What's the best place to ask questions? Seems there are some basic things that I can't figure out how you expect them to be done, so I'm struggling a bit with even basic tasks....

dietervansteenwegen, 2020-05-30:
Btw, if you need some more test DXF files, I'd be happy to generate them for you. Just let me know what you need.

tatarize, 2020-05-30:
How about instead of putting in effort figuring out how to work things with MeerK40t you just tell me how it *should* work. UI and UX is hard to get right. If you're confused other people are confused too and that means it's a problem with MeerK40t, not you. Even a steep learning curve is an issue that should be addressed.

I've made some design decisions which are notably a bit strange but make a lot of sense from a programming perspective. But, clearly are not well done or easy to use just yet. At a minimum though, there should be an easy intuitive workflow that works from the start and from there you can get fancier options as the need arises.

There are somethings about the classification system that just don't work. Like sometimes you'd need to recolor an object and reclassify things just so it'll make the object it needs.  https://www.youtube.com/watch?v=ikTlH5z2U-0 I think the UI there is just sort of in the way, for people. It's a good decision but poorly implemented, and I keep going off and making a killer terminal command and keybinds but no good way not a great way to make it cut things. 

I think after finishing up this current version I'm going to hyperfocus on UX. While the every object has a operation paradigm is wrong in particular ways, the workflow it gives rise to, of just having some colors which indicate an operation and assigning those to elements is actually pretty nice as far as user interfaces go.

It was a bit hyper-focused on the correct way to do things, without the functional chops to back it up what seems like a better paradigm.  It's a better method, but a much poorer implementation. 

Basically the operations are the important things and the elements are just the objects the operations work on. Most lasering software take a bunch of objects and assigns them some particular operation. But, then you're basically locked into a 1 object to 1 operation scheme, and you start making things like `scan then cut`, `dot layer`, etc. And you assign just that operation to that element, and that's all you can do. The better method is to have a bunch of operations that can do anything you might want to do and have them contain any of the elements you'd want. So you could cut 20 passes with an operation that does have explicitly says how many passes it has, or make 20 operation that each contain a 1 pass objects containing the same element. Or 1 operation with 20 references to the same object in it, or 5 operations each with 4 objects, with them all having some slight tweaks. You get smaller simpler and a more intuitive understanding of what's going to happen, and the order they will cut in when you send that job to the laser, and much more control over how that works.

The problem with this is I have very little code to make this part highly flexible methodology easy for the end user. And while the 1 color equals 1 operation bit works great, I kinda opted against it earlier because it felt a bit limiting since it basically works best when your elements and operations are 1:1 and really I can use the same element in every operation, as many times as is needed. To scan and cut that object, you add it to a scan and then to a cut. Which is much simpler. But the UI is much crapper. I figure the more typical 1 color = 1 operation, is a better starting point than I have.

You're not wrong, it is wrong, I'll work on it next version since I have a good idea about the problems and messed around with RDWorks a lot recently and understand the draw. Currently it does the Whisperer classification thing where red goes to cut, blue to engrave, and anything else to raster. And then it lets you move them around between the operations and elements at your leisure. 

But, that's not really that easy to use by any stretch, even though it's technically much more power.

---

This is a pretty good overview video of 0.5.2 (not by me)
https://www.youtube.com/watch?v=oESYORW7LBA

---

As for DXF anything simple that fails would be great. I know Hatches and Ellipses are not fully working, but there might be other things I don't know to look for, like polylines having arcs. I'm not an expert in DXF by any stretch so critical features might get left out since I didn't know they were there.

dietervansteenwegen, 2020-05-30:
In any case, I hope you realize there's people that really appreciate your work. I can do a little bit of Python, and can't imagine the amount of effort it takes to research, create, plan and maintain this. I can't imagine where you find the motivation (I suppose you thoroughly enjoy it), much less the time.

I don't really think I should tell you how it should behave/work. I use my k40 every few weeks lately since we have a toddler and a 4 week old baby, whereas there's people that use it daily. I'd think they have a better understanding of what is needed to make this an awesome tool, and what is required to change/added/removed.

What I could do is let you know where/when I need to search for certain actions or where I don't get the logic. It would be up to you (or others) to question whether my doubts/questions/stupidity is valid for a larger audience.

I'll take a look at the video's tonight, they would probably help me get started. The very early versions had less options and tools but were enough for me and how you did things was pretty obvious. I was actually tempted to go back to one of those versions until I remembered that those didn't work for the dxf files either. Since you spent much time typing replies I'm sure you realize how much time that takes and back then it was either write a issue report or getting the thing cut in wisperer. I chose the latter (it just had to get done).

Just one remark which really bothers me with _almost_ every open source/free software: documentation. I know how hard a good manual is. I know it's not the sexiest of all tasks and gets behind on software changes quickly. What could help (or could have helped me) is to have a message pop up on the first launch with some links to places with more info, even if they would be made by others (such as the videos you refer me to now). The information is there, why not point people towards it (ok, I could have done a little search myself, that is true. But since I couldn't manage to import my DXF even after installing Qcad, trying with inkscape to convert to SVG and several other half-assed options, I could only be bothered to document an issue, burned with Wisperer and went to bed. Very happy to have a reply waiting for me in the morning ;) )

dietervansteenwegen, 2020-05-30:
Oh, two more small things:

- English is not my first language. I usually manage to get by by referring to an online dictionary sometimes, but sometimes small nuances get lost in translation...
- I tried installing this on a Linux machine, but couldn't get it working. The issue was while compiling one of the dependencies (installed through pip). When ran as superuser, the dependency showed as installed but I think I couldn't launch Meerk40t for another reason. Don't know exactly anymore. In any case, I tried a couple of things but couldn't understand what was happening as it didn't seem to make sense. Since the issue was outside of Meerk40t but likely my system/python install I didn't feel like I should raise an issue.

tatarize, 2020-05-30:
* Yeah, people tend to do what's easiest for them, if my program isn't the easiest. That's on me. I have solid ideas for how to make that better but I'll put them off until I'm finished with the current version, since I've been working on that version since mid-March. 
* The program itself is built with multiple languages in mind. There are no translations currently available for it but it's built quite specifically to allow translations everywhere. So, if you'd like to use it in your own language, I'd gladly make that happen, explain the process, and include your translation(s) in the next version.
* There is in (https://github.com/meerk40t/meerk40t/releases) a natively pre-compiled Ubuntu Linux version. That might let you bypass the issues you have getting it to work. @joerlane does good work in this department.
* Even if something seems to fall outside the scope, it is a concern that I'd like to know about.  It would tell other users that this is a problem and maybe help produce solutions. I would rather error on the side of too many raised issues rather than too few.

tatarize, 2020-05-30:
If you want to provide a translation. I have (https://localise.biz/tatarize/meerk40t) setup and could send you an invite (an email to tatarize at my gmail.com account, since public sharing of email has spammer issues, if you wanted to go that way), or you could download things like some `poeditor` and the `messages.po` file (https://github.com/meerk40t/meerk40t/blob/0.5.3/locale/messages.po ).  It's highly specialized work, that I categorically can't do.

https://github.com/meerk40t/meerk40t/wiki/Providing-a-Translation.

dietervansteenwegen, 2020-06-01:
- I'll try the precompiled Linux version, and let you know how it goes.
- About the language, I'm sorry for the misunderstanding but I meant that I might express some feeling/idea in a manner that does not 100% reflect what I mean since I'm not native English. Even if I'd have plenty of time, I've never felt the need for translations of English software. There aren't that many menus etc to familiarize yourself with. Also, regardless whether you're using a translated version or not, you'll likely need support at some stage which would be in English anyway.

dietervansteenwegen, 2020-06-01:
Don't know where you want comments as this is likely not the best place. Until I know where to put them, I've got this one as a first.

As I import the DXF (apart from the polyline issue), I get the following:
[image]

That list goes on. Then there's the elements part, and the file foldout. All of them contain a list of about 50 items, with little to go by besides their text (such as "M -2314.442885722,20042.480248..." and what seems to be a crude representation of the element, although I don't think that's really working too well (I do realize most of my elements are about the same form, but that's how it will mostly be for me):
[image]

Now I know that if you're doing more artistic designs, these might be more representative and useful.

Also, 

-  why is that whole list repeated again in the file (because you can have multiple files as sources and you want to list which elements are from which file? In that case you could do a subdivision in your elements branch with a subsection for each file?)
- Why are all expanded by default? After importing the file, I get a tree with about 150 items.
- After reloading or changing some items in the tree, it scrolls back to the files subsection

dietervansteenwegen, 2020-06-01:
- at first, it's not clear how to add an operation. My initial reaction is to rightclick on the operations line, but that gives only "clear all" as options. Sure I'll find out.

dietervansteenwegen, 2020-06-01:
[image]
- I have been moving things around a bit, but now I'd like to select a few items using CTRL + click. That doesn't seem to work. Next idea: draw a bounding  box with the mouse to select multiple items. Doesn't work.
- The big "rectangle" seems to have precedence over the other items that are within its boundaries. Or it is on the top layer if you will. Only by moving the big outline of my panel (the big rectangle) aside, I can select other items and then only one by one. Difficult to rearrange items.
- I do realize I can select multiple items in the tree on the left, though its not very intuitive as the names/labels are not really related to the things they represent in my case. I realize that would be different if you're doing designs with more recognizable elements.

dietervansteenwegen, 2020-06-01:
- I try to drag one element to the operations to add a new operation. The operation list is long, but I expect the tree to scroll automatically when I am dragging to the top. It doesn't.
- after I release the mouse button, the list jumps back to the files folder (not sure what the correct names are), every time. This is particularly annoying since the list is long and unreprestative

dietervansteenwegen, 2020-06-01:
What would be great (for my specific case) is:
- select elements in the main window by clicking on them, selecting multiple by holding CTRL or drawing a box (ideally the Autocad way where drawing the box from left to right has a slightly different effect than right to left).
- right click on those selected/highlighted elements to show associated operations. Option to add an operation is there as well.

dietervansteenwegen, 2020-06-01:
[image]
Not all text fits in the box, that is fine. However, I can't seem to change the size of the panel to see everything. Since the text ends with three dots, it seems like there is more text ready to be shown on that line. So I tried to mouse over, but that didn't do anything either. Rightclick? Perhaps that will show me a properties panel with the full text? No.

I know this is nitpicking, but you wanted to have all the details...

dietervansteenwegen, 2020-06-01:
Selecting any element in the list highlights all other "instances" of that item. That's nice...

dietervansteenwegen, 2020-06-01:
Maybe a mark (in the tree) for those elements that currently fall outside the bed area could be useful?

tatarize, 2020-06-01:
Telling me stuff is generally fine. I'll likely start a new issue and just link to this one after it's closed since the original point was fixed. So sometime after I publish MeerK40t 0.6.0. I've worked on that version for too long so I'm pushing all feature changes to the next version that is going to focus on this User Interface stuff.

I'm raising #174 for the UI/UX corrections. Which I've already sort of decided should be the core of the next version. I hopefully rephrased transcribed your objections accurately.

tatarize, 2020-06-23:
This issue was corrected in 0.5.0. Which is now in full release.

tatarize, 2020-06-23:
0.6.0*

closed 2020-06-23 by tatarize