meerk40t/meerk40t #173: PIP exception at startup (because not admin?) but Meerk40t starts fine

dietervansteenwegen, 2020-06-01:
When I start Meerk40t (pull from GitHub) by using the install_run.bat script, I get a exception traceback from pip. After that Meerk40t does start fine. I expect it to be because I haven't started it with admin rights. 
Traceback:
```
`P:\Dieter\Projects\K40\meerk40t>pip install -r requirements.txt
ERROR: Exception:
Traceback (most recent call last):
  File "c:\program files (x86)\python36-32\lib\site-packages\pip\_internal\cli\base_command.py", line 186, in _main
    status = self.run(options, args)
  File "c:\program files (x86)\python36-32\lib\site-packages\pip\_internal\commands\install.py", line 258, in run
    isolated_mode=options.isolated_mode,
  File "c:\program files (x86)\python36-32\lib\site-packages\pip\_internal\commands\install.py", line 604, in decide_user_install
    if site_packages_writable(root=root_path, isolated=isolated_mode):
  File "c:\program files (x86)\python36-32\lib\site-packages\pip\_internal\commands\install.py", line 549, in site_packages_writable
    test_writable_dir(d) for d in set(get_lib_location_guesses(**kwargs))
  File "c:\program files (x86)\python36-32\lib\site-packages\pip\_internal\commands\install.py", line 549, in <genexpr>
    test_writable_dir(d) for d in set(get_lib_location_guesses(**kwargs))
  File "c:\program files (x86)\python36-32\lib\site-packages\pip\_internal\utils\filesystem.py", line 140, in test_writable_dir
    return _test_writable_dir_win(path)
  File "c:\program files (x86)\python36-32\lib\site-packages\pip\_internal\utils\filesystem.py", line 153, in _test_writable_dir_win
    fd = os.open(file, os.O_RDWR | os.O_CREAT | os.O_EXCL)
PermissionError: [Errno 13] Permission denied: 'c:\\program files (x86)\\python36-32\\Lib\\site-packages\\accesstest_deleteme_fishfingers_custard_nogimf'
WARNING: You are using pip version 20.0.2; however, version 20.1.1 is available.
You should consider upgrading via the 'c:\program files (x86)\python36-32\python.exe -m pip install --upgrade pip' command.`
```

When using "run as admin" in windows explorer I get the following (because of absolute filepaths?):
```
 `C:\WINDOWS\system32>pip install -r requirements.txt
ERROR: Could not open requirements file: [Errno 2] No such file or directory: 'requirements.txt'
WARNING: You are using pip version 20.0.2; however, version 20.1.1 is available.
You should consider upgrading via the 'c:\program files (x86)\python36-32\python.exe -m pip install --upgrade pip' command.

C:\WINDOWS\system32>python MeerK40t.py
python: can't open file 'MeerK40t.py': [Errno 2] No such file or directory

C:\WINDOWS\system32>pause
Press any key to continue . . .`
```

If I manually open a terminal with admin rights I get:
```
`P:\Dieter\Projects\K40\meerk40t>pip install -r requirements.txt
ERROR: Exception:
Traceback (most recent call last):
  File "c:\program files (x86)\python36-32\lib\site-packages\pip\_internal\cli\base_command.py", line 186, in _main
    status = self.run(options, args)
  File "c:\program files (x86)\python36-32\lib\site-packages\pip\_internal\commands\install.py", line 258, in run
    isolated_mode=options.isolated_mode,
  File "c:\program files (x86)\python36-32\lib\site-packages\pip\_internal\commands\install.py", line 604, in decide_user_install
    if site_packages_writable(root=root_path, isolated=isolated_mode):
  File "c:\program files (x86)\python36-32\lib\site-packages\pip\_internal\commands\install.py", line 549, in site_packages_writable
    test_writable_dir(d) for d in set(get_lib_location_guesses(**kwargs))
  File "c:\program files (x86)\python36-32\lib\site-packages\pip\_internal\commands\install.py", line 549, in <genexpr>
    test_writable_dir(d) for d in set(get_lib_location_guesses(**kwargs))
  File "c:\program files (x86)\python36-32\lib\site-packages\pip\_internal\utils\filesystem.py", line 140, in test_writable_dir
    return _test_writable_dir_win(path)
  File "c:\program files (x86)\python36-32\lib\site-packages\pip\_internal\utils\filesystem.py", line 153, in _test_writable_dir_win
    fd = os.open(file, os.O_RDWR | os.O_CREAT | os.O_EXCL)
PermissionError: [Errno 13] Permission denied: 'c:\\program files (x86)\\python36-32\\Lib\\site-packages\\accesstest_deleteme_fishfingers_custard_nogimf'
WARNING: You are using pip version 20.0.2; however, version 20.1.1 is available.
You should consider upgrading via the 'c:\program files (x86)\python36-32\python.exe -m pip install --upgrade pip' command.`
```

As stated, not an issue for me so far, but though I'd let you know.

tatarize, 2020-06-01:
* the batchfile instruction don't make it clear that you require python whereas the command line instructions do.
* The linux instructions might be better if they tell you to do sudo on permissions error.

Should be clarified for the 0.6.0 version: https://github.com/meerk40t/meerk40t/commit/193844d9ed41b807c40a5ea7b57ba4771de0919b

dietervansteenwegen, 2020-06-01:
Just to be clear: I do have python installed on my window machine, otherwise I would think pip wouldn't be there to trow an exception (and meerk40t wouldnt work), as far as I can see.

closed 2020-06-23 by tatarize