meerk40t/meerk40t #177: When changing application, some of the Meerk40t windows stay on top

dietervansteenwegen, 2020-06-06:
I am running Meerk40t on a single display (laptop) most of the time.
When I want to make some quick changes, I open Inkscape, but windows like "spooler", "navigation", are kept in the foreground, blocking my work area in Inkscape. I have to close them to continue working.

I'd expect them to move to the background if I change application.

tatarize, 2020-06-07:
Yeah, the problem here is failing to set those windows always on top causes secondary issues when they keep falling behind the main window.  I haven't given it enough thought or attempts but, the main window is the only window not flagged to be on-top. So all the other windows can't fall behind the main window. It's a solid problem and good to have an issue about. I'm not rightly sure how to fix it. There may be some way to keep the main window at the bottom of the stack while not flagging every child window as on top.  That or switch over to AUI #134 and have it fixed that way, since that upgrade would likely fix the stuff leading to this issue.

That or there might be other things that correct this problem in wx that I am unaware of.

tatarize, 2020-06-23:
I actually know how to fix this now, Gotta assign all windows a parent and have them "Float On Top" which should correct this oddity.

tatarize, 2020-07-02:
Fixed this in the current 0.6.1 branch. Seems to work fairly okay took a bit to correct the lifecycle issues giving all windows parents produced.

tatarize, 2020-07-16:
This is corrected in 0.6.1, now live.

closed 2020-07-16 by tatarize