meerk40t/meerk40t #178: Blue laser trace remains on screen

dietervansteenwegen, 2020-06-06:
[image]
Even after removing all files/elements/operations, the previous laser trace remains on screen.

tatarize, 2020-06-06:
#58

tatarize, 2020-06-07:
Yeah, I'm not sure what to do with that laser trace things. Pressing New will clear it in the next version. But, the overall utility is questionable and I'm not really sure what to do instead or whether to just clear that information a lot. It's just like a thousand lines between various position updates.

I'm not really sure what I should want done there or should do instead.

tatarize, 2020-07-24:
Changing the selection will now clear that data. There's also a control to perform this action which could be bound to a keypress

closed 2020-07-24 by tatarize